Koenkk/zigbee2mqtt #25908: Ikea E1766 OTA Update Fails

kajishun, 2025-01-20:
### What happened?

Ikea E1766 shows "update device firmware" available in Z2M OTA. Attempt to update device firmware (using proper wake technique) fails multiple times. Update starts, takes ~90 minutes, the device continues to show the old firmware 24.4.5. Devices do not update.
Previously this has worked properly in the past several months. Previously able to update several devices. Unable to say when exactly the ability to update changed.
Currently have 15 devices with 4 showing update available from 24.4.5 to 24.4.6, 2 show 24.4.5 without update available. Updating one or several at once fails.

### What did you expect to happen?

Device to update after ~90 minutes and show new firmware version.

### How to reproduce it (minimal and precise)

Navigate to Z2M, click on OTA, wake device while clicking "update device firmware," device to show new firmware after update.

### Zigbee2MQTT version

2.0.0 commit: unknown

### Adapter firmware version

7.4.5 [GA]

### Adapter

SLZB-06M

### Setup

via POE network connection

isibizi, 2025-01-20:
This ist not an z2m issue! The remote goes in the sleep mode, you must press a button durig the update.

kajishun, 2025-01-21:
incorrect. i know the proper wake process. the firmware update starts
properly. it never completes ~90 min later. or it completes and the device
reverts.

kajishun, 2025-01-21:
in addition, the firmware update used to work properly. now it no longer
does. something has changed in the few months i have been using HA and have
converted to Z2M.

kajishun, 2025-01-22:
> what is the LQI of this device?

There are several of these devices with a variety of LQI.

remote 10 - LQI 208
remote 5 - LQI 188
remote 4 - LQI 172
remote 1 - LQI 196

All have exhibited the same issue of saying they complete the update then
still showing the old update. Several have a very good LQI. I doubt the LQI
is at fault.

viliks, 2025-03-30:
Same issue for me if upgrading from 24.4.5 to 24.4.6. Upload completes to 100% without errors, but device stays (reverts) at 24.4.5

I had one with older version (i think it was 2.3.088) and this was updated correctly to 24.4.6

kajishun, 2025-03-30:
to resolve i went back to ikea dirigera hub. i don't like having multiple
zigbee networks but they just work better this way. in ikea home smart, the
remotes updated just fine.

viliks, 2025-03-31:
IKEA has two endpoints for OTA updates as stated here
https://github.com/Koenkk/zigbee2mqtt/discussions/24464
and that should be resolved here
https://github.com/Koenkk/zigbee-herdsman-converters/pull/8131

I downloaded both and used OTA index override file.

This one was rejected by device after upload. And this is the one that z2m uses: https://github.com/Koenkk/zigbee-OTA/tree/master/images/IKEA
`{"fw_image_type":4549,"fw_type":2,"fw_sha3_256":"499f08c42fdcd5019c8d48b3a17c2f3406cdc97b4bfb41db064a1487e695596c","fw_binary_url":"https://fw.ota.homesmart.ikea.com/files/tradfri-onoff-controller_release_prod_v604241926_3c2e5569-667c-49ed-a286-78a0e031935d.ota"}`

This one correctly updated multiple devices (E1743 and E1766) from 24.4.5 to 24.4.6, always on first try without any issues.
```
{
	"fw_binary_url": "http://fw.ota.homesmart.ikea.net/global/GW1.0/01.21.031/bin/10005778-tradfri_onoff_controller-24.4.6-prod.ota.ota.signed",
	"fw_file_version_LSB": 6,
	"fw_file_version_MSB": 9220,
	"fw_filesize": 205560,
	"fw_image_type": 4549,
	"fw_manufacturer_id": 4476,
	"fw_type": 2
}
```

So
`tradfri-onoff-controller_release_prod_v604241926_3c2e5569-667c-49ed-a286-78a0e031935d.ota`
I do not have logs for the one device updated from 2.3.079 to 24.4.6, but assume this was used as update file and it was updated correctly.

But to update from 24.4.5 to 24.4.6 only `10005778-tradfri_onoff_controller-24.4.6-prod.ota.ota.signed` is working.

kagstrom2100, 2025-05-18:
Having the same issue while trying to update the following devices:

* E1766 (Currently on 2.3.079)
* E2001/E2002 (Currently on 1.0.024)

Any fix for this coming?