Koenkk/zigbee2mqtt #25910: Aqara motion sensor. Differen behaviour with continous motion

boehser-enkel, 2025-01-20:
### What happened?

Hi,

i have many aqara motion sensors (RTCGQ11LM) which behave differently when there is continous motion:

1. motion - (occupancy timeout) - no motion - (directly) - motion again
2.  motion ---------- motion timeout

Second one does not get a new motion state so it stays the same until there is no motion again
This (2. behaviour) is only with some of my 10 motion sensors. All my automation restart on new motion to restart the included timer.
Configuration within z2m is the same for all sensors

Is there a way to change the behaviour?

greetings

### What did you expect to happen?

Change motion state change behaviour

### How to reproduce it (minimal and precise)

Initiate motion longer than the configured timeout

### Zigbee2MQTT version

2

### Adapter firmware version

20240710

### Adapter

Sonoff USB - P

### Setup

Docker

mrmaximas, 2025-01-21:
do you have specific settings configured?
Occupancy timeout 60
No occupancy since 60

boehser-enkel, 2025-01-22:
Sorry missed that: 
occupancy_timeout: 60
for each. Nothing else

closed 2025-03-31 by github-actions[bot]