Koenkk/zigbee2mqtt #25920: ECONNRESET error with ember and SLZB-06M

squito1, 2025-01-20:
### What happened?

Hello, I had to reinstall my Proxmox installation of Zigbee2MQTT and Mosquitto broker and had issues with them connecting using ember so I changed to ezsp and they connected. Today, I updated SLZM-06M to core firmware version: v2.7.0 and thought I would try ember again. Started getting the error below right after restart and now if I switch back to ezsp it will no longer work either giving the same message: Port Error: read ECONNRESET. Any ideas here? Thank you.

Zigbee2MQTT version: 2.0.0
Eclipse Mosquitto version: 2.0.20
SLZM-06M core version: 2.7.0
SLZM-06M Z2M version: 20241127

Starting Zigbee2MQTT without watchdog.
[2025-01-20 12:58:51] info: z2m: Logging to console, file (filename: log.log)
[2025-01-20 12:58:51] info: z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-20 12:58:51] info: z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-20 12:58:51] info: zh:ember: Using default stack config.
[2025-01-20 12:58:51] info: zh:ember: ======== Ember Adapter Starting ========
[2025-01-20 12:58:51] info: zh:ember:ezsp: ======== EZSP starting ========
[2025-01-20 12:58:51] info: zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-20 12:58:51] info: zh:ember:uart:ash: Socket ready
[2025-01-20 12:58:51] info: zh:ember:uart:ash: ======== ASH starting ========
[2025-01-20 12:58:51] error: zh:ember:uart:ash: Port Error: read ECONNRESET
[2025-01-20 12:58:51] info: zh:ember:uart:ash: Port closed.
[2025-01-20 12:58:53] info: zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-20 12:58:53] info: zh:ember:uart:ash: Socket ready
[2025-01-20 12:58:53] info: zh:ember:uart:ash: ======== ASH starting ========
[2025-01-20 12:58:54] error: zh:ember:uart:ash: Port Error: read ECONNRESET
[2025-01-20 12:58:54] info: zh:ember:uart:ash: Port closed.
[2025-01-20 12:58:56] info: zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-20 12:58:56] info: zh:ember:uart:ash: Socket ready
[2025-01-20 12:58:56] info: zh:ember:uart:ash: ======== ASH starting ========
[2025-01-20 12:58:57] error: zh:ember:uart:ash: Port Error: read ECONNRESET
[2025-01-20 12:58:57] info: zh:ember:uart:ash: Port closed.
[2025-01-20 12:58:58] info: zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-20 12:58:58] info: zh:ember:uart:ash: Socket ready
[2025-01-20 12:58:58] info: zh:ember:uart:ash: ======== ASH starting ========
[2025-01-20 12:58:59] error: zh:ember:uart:ash: Port Error: read ECONNRESET
[2025-01-20 12:58:59] info: zh:ember:uart:ash: Port closed.
[2025-01-20 12:59:01] info: zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-20 12:59:01] info: zh:ember:uart:ash: Socket ready
[2025-01-20 12:59:01] info: zh:ember:uart:ash: ======== ASH starting ========
[2025-01-20 12:59:02] error: zh:ember:uart:ash: Port Error: read ECONNRESET
[2025-01-20 12:59:02] info: zh:ember:uart:ash: Port closed.
[2025-01-20 12:59:03] error: z2m: Error while starting zigbee-herdsman
[2025-01-20 12:59:03] error: z2m: Failed to start zigbee-herdsman
[2025-01-20 12:59:03] error: z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-20 12:59:03] error: z2m: Exiting...
[2025-01-20 12:59:04] error: z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
at EmberAdapter.initEzsp (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:670:19)
at EmberAdapter.start (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1533:24)
at Controller.start (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:69:27)
at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:142:13)
at start (/opt/zigbee2mqtt/index.js:161:5)
root@zigbee2mqtt:/opt/zigbee2mqtt#

Config:
version: 4
homeassistant:
  enabled: true
permit_join: false
frontend:
  enabled: true
  port: 9442
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.1.61:1883
  user: xxxxxxx
  password: xxxxxxxx
  keepalive: 60
  reject_unauthorized: true
serial:
[... 223 lines cut ...]
[2025-01-20 13:28:24] debug:    zh:ember:uart:ash: Waiting for RSTACK... 1200/2500
[2025-01-20 13:28:24] debug:    zh:ember:uart:ash: Waiting for RSTACK... 1300/2500
[2025-01-20 13:28:24] debug:    zh:ember:uart:ash: Waiting for RSTACK... 1400/2500
[2025-01-20 13:28:24] debug:    zh:ember:uart:ash: Waiting for RSTACK... 1500/2500
[2025-01-20 13:28:24] debug:    zh:ember:uart:ash: Waiting for RSTACK... 1600/2500
[2025-01-20 13:28:24] debug:    zh:ember:uart:ash: Waiting for RSTACK... 1700/2500
[2025-01-20 13:28:24] debug:    zh:ember:uart:ash: Waiting for RSTACK... 1800/2500
[2025-01-20 13:28:24] debug:    zh:ember:uart:ash: Waiting for RSTACK... 1900/2500
[2025-01-20 13:28:24] debug:    zh:ember:uart:ash: Waiting for RSTACK... 2000/2500
[2025-01-20 13:28:24] debug:    zh:ember:uart:ash: Waiting for RSTACK... 2100/2500
[2025-01-20 13:28:25] debug:    zh:ember:uart:ash: Waiting for RSTACK... 2200/2500
[2025-01-20 13:28:25] debug:    zh:ember:uart:ash: Waiting for RSTACK... 2300/2500
[2025-01-20 13:28:25] debug:    zh:ember:uart:ash: Waiting for RSTACK... 2400/2500
[2025-01-20 13:28:25] error:    z2m: Error while starting zigbee-herdsman
[2025-01-20 13:28:25] error:    z2m: Failed to start zigbee-herdsman
[2025-01-20 13:28:25] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-20 13:28:25] error:    z2m: Exiting...
[2025-01-20 13:28:25] error:    z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
    at EmberAdapter.initEzsp (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:670:19)
    at EmberAdapter.start (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1533:24)
    at Controller.start (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:69:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:142:13)
    at start (/opt/zigbee2mqtt/index.js:161:5)
root@zigbee2mqtt:/opt/zigbee2mqtt#

Koenkk, 2025-01-21:
Likely your network connection is not stable, https://github.com/Koenkk/zigbee2mqtt.io/commit/3033f2b068c1f992d61a048eecdd3eced54e1f1d may help

squito1, 2025-01-21:
Then it sounds like an SMLIGHT SLZB-06M firmware problem because I switched back to core firmware version: v2.3.6 with **EZSP**, because EMBER does not work, and I was able to have it connect again. I hope someone figures how to get EMBER implemented correctly so I stop getting the 'EZSP is deprecated" message.

Tarik2142, 2025-01-22:
Are you trying to run 2 Z2M instances for one coordinator?

MikeDeltaHH, 2025-01-22:
Same here with a fresh Z2M lcx installation... 

@squito1 is downgrade the slzb-06m firm to 2.3.6 a working solution? I have the new 2.7.0 firm installed but with the same error...

squito1, 2025-01-22:
@MikeDeltaHH it has worked for me but I had to use EZSP and not Ember for the adapter.

MikeDeltaHH, 2025-01-27:
I changed the SLZB-06N to firmware 2.5.6 and ezsp adapter, after that everything worked. After I then updated again to firmware 2.7.1 and changed the adapter back to ember, the problem did not occur again, the stick now works perfectly for me and I am gradually switching my Zigbee devices...

Boui3D, 2025-01-27:
Upgrade to firmware 2.7.1 was helpful.

mxcxpx, 2025-01-28:
have same problem but already on 2.7.1 Firmware

Xites, 2025-01-29:
> have same problem but already on 2.7.1 Firmware

I had the same on 2.7.1, rebooting the SLZB-06M's ESP32 resolved the problem.
But I'm afraid this will happen again. At least I know a workaround now.

cairnsie13, 2025-02-12:
I was having major issues with my device and the ECONNRESET error and instability since the z2m update (Crashing in less than 12 hours). After lots of changing firmware etc. I’m now running 2.7.1 and the latest silabs firmware and have switched from ethernet connection to serial and now have no issues. Device has been running completely stable for over a week, the silabs firmware has also improved the connection to one of my devices that had poor signal.

jonisdorf1, 2025-02-13:
I have the same problem. I’ve tried both upgrading and downgrading the firmware, but it didn’t help. So, it must be on the Home Assistant side. @cairnsie13  already more or less confirmed that it’s not the firmware. When using the stick over LAN, it doesn’t work. However, I need to use the stick via LAN.

martinbjeldbak, 2025-02-13:
Found this thread by googling this error - a reboot of my SLZB-06M followed by a restart of the zigbee2mqtt container reconnected the two. Will keep an eye on any instability

kevinohara80, 2025-03-03:
Same issue here. Very frustrating!

Config:
```yaml
version: 4
permit_join: true
homeassistant:
  enabled: true
serial:
  port: tcp://192.168.1.137:6638
  baudrate: 115200
  adapter: ember
  rtscts: true
  disable_led: false
advanced:
  network_key:
    - 235
    - 241
    - 183
    - 228
    - 46
    - 166
    - 208
    - 112
    - 235
    - 31
    - 195
    - 73
    - 191
    - 192
    - 46
    - 108
  pan_id: 44700
  ext_pan_id:
    - 97
    - 229
    - 211
    - 56
    - 208
    - 106
    - 171
    - 161
#   transmit_power: 20
frontend:
  enabled: true
  port: 8099
mqtt:
  server: mqtt://core-mosquitto:1883
  user: addons
  password: ********
```

Logs:
```
[21:15:20] INFO: Preparing to start...
[21:15:20] INFO: Socat not enabled
[21:15:20] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-03-02 21:15:21] info: 	z2m: Logging to console, file (filename: log.log)
[2025-03-02 21:15:21] info: 	z2m: Starting Zigbee2MQTT version 2.1.3 (commit #unknown)
[2025-03-02 21:15:21] info: 	z2m: Starting zigbee-herdsman (3.2.7)
[2025-03-02 21:15:21] info: 	zh:ember: Using default stack config.
[2025-03-02 21:15:21] info: 	zh:ember: ======== Ember Adapter Starting ========
[2025-03-02 21:15:21] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2025-03-02 21:15:21] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-03-02 21:15:21] info: 	zh:ember:uart:ash: Socket ready
[2025-03-02 21:15:21] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-03-02 21:15:21] error: 	zh:ember:uart:ash: Port Error: read ECONNRESET
[2025-03-02 21:15:21] info: 	zh:ember:uart:ash: Port closed.
[2025-03-02 21:15:23] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-03-02 21:15:23] info: 	zh:ember:uart:ash: Socket ready
[2025-03-02 21:15:23] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-03-02 21:15:24] error: 	zh:ember:uart:ash: Port Error: read ECONNRESET
[2025-03-02 21:15:24] info: 	zh:ember:uart:ash: Port closed.
[2025-03-02 21:15:26] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-03-02 21:15:26] info: 	zh:ember:uart:ash: Socket ready
[2025-03-02 21:15:26] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-03-02 21:15:27] error: 	zh:ember:uart:ash: Port Error: read ECONNRESET
[2025-03-02 21:15:27] info: 	zh:ember:uart:ash: Port closed.
[2025-03-02 21:15:28] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-03-02 21:15:28] info: 	zh:ember:uart:ash: Socket ready
[2025-03-02 21:15:28] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-03-02 21:15:28] error: 	zh:ember:uart:ash: Port Error: read ECONNRESET
[2025-03-02 21:15:28] info: 	zh:ember:uart:ash: Port closed.
[2025-03-02 21:15:31] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-03-02 21:15:31] info: 	zh:ember:uart:ash: Socket ready
[2025-03-02 21:15:31] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-03-02 21:15:31] error: 	zh:ember:uart:ash: Port Error: read ECONNRESET
[2025-03-02 21:15:31] info: 	zh:ember:uart:ash: Port closed.
[2025-03-02 21:15:33] error: 	z2m: Error while starting zigbee-herdsman
[2025-03-02 21:15:33] error: 	z2m: Failed to start zigbee-herdsman
[2025-03-02 21:15:33] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-03-02 21:15:33] error: 	z2m: Exiting...
[2025-03-02 21:15:33] error: 	z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
    at EmberAdapter.initEzsp (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:670:19)
    at EmberAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1536:24)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)
```

Hakker, 2025-03-03:
Another one with the same error. Only 2.3.6 on ezsp worked. 2.6.8dev26 and up gave errors although dev26 worked somewhat with ezsp but not on ember. The rest doesn't work with ezsp and ember

cyayon, 2025-03-04:
Hi, same issue here append a few times (2.7.1 version too and latest Z2M 2.1.x).
Is it a good practice to enable "Allow multi-threaded socket connection" to workaround the issue (and have to restart manually Z2M sometimes) ?

thanks.

cyayon, 2025-03-04:
I tried to enable mutlti-threaded socket and the Z2M daemon crash every hour. 
Disabled the option, and no more crash of Z2M. 
Currently, Z2M is crashing about every month...

modestos, 2025-03-06:
> I tried to enable mutlti-threaded socket and the Z2M daemon crash every hour. Disabled the option, and no more crash of Z2M. Currently, Z2M is crashing about every month...

Same here! Mine is crashing every minute!
I think tried everything and am running out of ideas.

modestos, 2025-03-06:
I think I found a way to fix this issue. It works for about 20+hours.

The issue comes down to Home Assistant’s autodiscovery mechanism, which detects two integrations for your SLZB-06M:
SLZB-06M Zigbee Home Automation & SLZB-06M by SMLIGHT

Here’s the catch: you should only enable the second one, "SLZB-06M by SMLIGHT." If you have both enabled, they conflict, leading to the ECONNRESET error and making your Ember adapter unstable.

[image]

1. Disable the "SLZB-06M Zigbee Home Automation" integration in Home Assistant.
2. Add "SLZB-06M Zigbee Home Automation" to the ignored list to stop it from being detected again.
3. Make sure Zigbee2MQTT is set up correctly with the right firmware and adapter settings.
4. Experiment with "Allow multi-threaded socket connection" (worked both with and without)

The rest I used were pretty standard:
Zigbee2MQTT Configuration in Home Assistant (Current version: 2.1.3-1)
port: tcp://xxx.xxx.xxx.xxx:6638
baudrate: 115200
adapter: ember
disable_led: false
advanced:
  transmit_power: 20

SLZB-06M Firmware Versions:
- Coordinator firmware version: v2.8.2.dev3
- ZigBee firmware revision: 20241127

xprezz, 2025-03-14:
Had the same issue. Updated to v2.8.2.dev3 firmware and it seems to be working now. Using Ember.

maistrovyi, 2025-03-16:
> Had the same issue. Updated to v2.8.2.dev3 firmware and it seems to be working now. Using Ember.

Unfortunately not, there are falls with the same error, though rarely, once every 2-3 days

BlockHeid, 2025-03-17:
Hi first time post and a Home Assistant newbie (including Zigbee2MQTT!).

Purchased the SMLIGHT SLZB-06M and like all the posts above was able to setup the SMLIGHT device and ran without a hiccup for a day or two and then it drops out like all the posts above.  The Z2M device is on the latest Core and Zigbee released firmware and also have the configuration file edited to read "ember" for the serial connection.

Now I cannot reconnect due presumably due to ECONNRESET, as per snipped below:

[2025-03-17 10:01:26] info: 	z2m: Logging to console, file (filename: log.log)
[2025-03-17 10:01:26] info: 	z2m: Starting Zigbee2MQTT version 2.1.3 (commit #unknown)
[2025-03-17 10:01:26] info: 	z2m: Starting zigbee-herdsman (3.2.7)
[2025-03-17 10:01:26] info: 	zh:ember: Using default stack config.
[2025-03-17 10:01:26] info: 	zh:ember: ======== Ember Adapter Starting ========
[2025-03-17 10:01:26] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2025-03-17 10:01:26] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-03-17 10:01:27] info: 	zh:ember:uart:ash: Socket ready
[2025-03-17 10:01:27] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-03-17 10:01:27] error: 	zh:ember:uart:ash: Port Error: read ECONNRESET

Upon some digging within the Home Assistant Studio Code and not sure how relevant the coordinator_backup.json is for starting up the Z2M server however noted that it is using the EZSP configuration and also the coordinator_ieee does not correspond to the ieee on the SMLIGHT control panel (webUI).  Should there be a coordinator.json file as well as a backup file? 

Current Coordinator_Backup.json read:

{
  "metadata": {
    "format": "zigpy/open-coordinator-backup",
    "version": 1,
    "source": "zigbee-herdsman@3.2.7",
    "internal": {
      "date": "2025-03-16T10:15:46.233Z",
      "ezspVersion": 14
    }
  },
  "stack_specific": {
    "ezsp": {
      "hashed_tclk": "e2c3e4cc58e7d0fab757676ab81f590c"
    }
  },
  "coordinator_ieee": "de06ccfeffdbb484",
  "pan_id": "264b",
  "extended_pan_id": "f55773515cbadb4e",
  "nwk_update_id": 0,
  "security_level": 5,
  "channel": 11,
  "channel_mask": [

Is there a way to reconfigure the coordinator.json file to read the current SMLIGHT coordinator IEEE etc. or am I overlooking something much simpler?

modestos, 2025-03-17:
I do not know what I am doing wrong but, mine works withought any problems so far. 
I also have to say that after the firmware upgrade, I also upgraded my power supply to a solid 2.5A.
I will keep you updated if I notice anything.

maistrovyi, 2025-03-17:
> I do not know what I am doing wrong but, mine works withought any problems so far. I also have to say that after the firmware upgrade, I also upgraded my power supply to a solid 2.5A. I will keep you updated if I notice anything.

how many SLZB devices you have in your infra?

jjscholtes, 2025-04-01:
I had this error as well drove me nuts tried different firmwares etc, last resort I tried a new cat cable also bypassing the keystone. Solved everything not a single error since then.

akira215, 2025-04-03:
I also faced this issue, and my investigation is that it happen when something happen on the connection, which could be:
 - disconnection of the network (network equipment reboot, cable temporaly disconnected, ....)
 - it happen also when home assistant supervisor auto update ([https://github.com/Koenkk/zigbee2mqtt/issues/26008#issuecomment-2766877756](https://github.com/Koenkk/zigbee2mqtt/issues/26008#issuecomment-2766877756)), to avoid annoying so much with that, I leaved dev branch for HA supervisor an stay on the stable channel (as update are much more frequent on dev branch).

I think one behavior which could be implemented is
 - when ECONNRESET is detected (only with network zigbee gateway ???), set a reboot counter trial to an arbitrary value to avoid infinite retries
 - disconnect as it is already implemented
 - try to reconnect until reboot counter reach 0

I wanted to help but actually I don't have setup an environnement to work on the code and propose a PR. I will try to because this repo is awesome in a next future, but not in the following weeks.

Tarik2142, 2025-04-03:
> I also faced this issue, and my investigation is that it happen when something happen on the connection, which could be:
> 
> * disconnection of the network (network equipment reboot, cable temporaly disconnected, ....)
> * it happen also when home assistant supervisor auto update ([ECONNRESET error with SLZB-06M #26008 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/26008#issuecomment-2766877756)), to avoid annoying so much with that, I leaved dev branch for HA supervisor an stay on the stable channel (as update are much more frequent on dev branch).
> 
> I think one behavior which could be implemented is
> 
> * when ECONNRESET is detected (only with network zigbee gateway ???), set a reboot counter trial to an arbitrary value to avoid infinite retries
> * disconnect as it is already implemented
> * try to reconnect until reboot counter reach 0
> 
> I wanted to help but actually I don't have setup an environnement to work on the code and propose a PR. I will try to because this repo is awesome in a next future, but not in the following weeks.

If the network connection was not closed properly on the host side then you will receive this error from the coordinator until the coordinator closes the connection due to a timeout. To work around simply enable the "Allow multi-threaded socket connection" option in the "Z2M and ZHA" menu.

ulikasper, 2025-04-08:
In my case, I think the issue suddenly came up, since I connected my SLZB-06M to a cheap 5-port poe switch, where another poe device (cam) is also connected to. 
I tested a few cable and port changes, but nothing made the ECONNRESET disappear.
Now I switched to the old constellation without this switch, to check if this device is causing the issue. 
So, the SLZB-06M is only connected to a normal router (FritzBox) and powered with a standalone poe power-supply.   

I will report, if this fixed it for me. It has been stable now for 12 hours, which is not really a proof yet.

BlockHeid, 2025-04-09:
Hi all,

Stumbled across a YouTube video by SMLIGHT and noted this in the comment section that the "M" series is a relevantly new chip and has not be heavily tested as the "06" series.

However, using Home Assistant updating Z2M add-on to v2.1.2-1, Z2M has been pretty solid thus far.

[image]

Boui3D, 2025-04-10:
I powered my SLZB-06M using an external PoE injector (48V, 0.5A), connected to a SODOLA 9-Port 2.5G Managed Switch. This switch is uplinked to a SODOLA 24-Port 2.5G Smart Switch, forming the core of my network. My Zigbee network has been stable for months.

albertjh, 2025-05-17:
Same issue as described before, freshly new installation, upgraded the SLZB-06M to the latest formware and using mqtt2zigbee didnt work with same issue. I've downgraded to the 2.3.6 version and started working. Definetly something going on with the current version.