Koenkk/zigbee2mqtt #25923: update script broken?

0xdefec71f, 2025-01-21:
### What happened?

Hi, 
maybe the update skript is broken. it aborts at Installing dependencies with an npm error: 
`The `npm ci` command can only install with an existing package-lock.json`

But several lines before the lockfile checkout was successfull
`Checking out changes to package-lock.json...`

I also noticed, that there were changes to the update script not long ago:
[https://github.com/Koenkk/zigbee2mqtt/commit/b6186623f0bcf9d88de1772aebd8a9eaead21a4c#diff-e14594f6bb9b29ac0146c04e78ae33e8e92f2f00709fefab97359dea31b94d11](https://github.com/Koenkk/zigbee2mqtt/commit/b6186623f0bcf9d88de1772aebd8a9eaead21a4c#diff-e14594f6bb9b29ac0146c04e78ae33e8e92f2f00709fefab97359dea31b94d11)

I am no git pro, so i downloaded the new update skript manually and gave it a try. But the error messages stay the same.
z2m is running in a proxmox container, so I have tried several things and load a backup if failed.

Also every time the skript hits the command
`git pull`
or
`git pull --no-rebase`
it stops with
```
Changes not staged for commit:
  (use "git add/rm <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
        modified:    data/configuration.yaml
```

Again, I am no git pro, but of course my configuration is modified and of course I want to keep my changes.
Am I doing something wrong or is there a promblem with the skript? I would apprechiate, If the update skript would be more failsafe for common problems.

See full log attached. Let me know, if you need more information or tries.
[update.log](https://github.com/user-attachments/files/18490129/update.log)

Best regards!

### What did you expect to happen?

successfull update

### How to reproduce it (minimal and precise)

simply run ./update.sh

### Zigbee2MQTT version

1.41.0 commit: 69f72829

### Adapter firmware version

20220219

### Adapter

Zigstar

### Setup

Proxmox

Koenkk, 2025-01-21:
This is one of the 2.0.0 breaking changes, see "Bare-metal installations (no Docker/Home Assistant add-on)" from https://github.com/Koenkk/zigbee2mqtt/discussions/24198

0xdefec71f, 2025-01-23:
> This is one of the 2.0.0 breaking changes, see "Bare-metal installations (no Docker/Home Assistant add-on)" from [#24198](https://github.com/Koenkk/zigbee2mqtt/discussions/24198)

oh, sorry, I missed that. Thank you for the link. 
`pnpm run clean` did the trick for me.

kind regards!

closed 2025-01-23 by 0xdefec71f