Koenkk/zigbee2mqtt #25925: Brightness not saving - goes to 1 when switched off

kilian-f-za, 2025-01-21:
### What happened?

I have activated transition globally in the configuration.

Light is on
[image]

When switching off lights (Philips Hue downlight in my case), the brightness via mqtt or zigbee2mqtt frontend goes to the value 1.

[image]

If I switch off the light with a directly bound switch, you can see how the brightness decreases and finally goes back to the original brightness. 

[image]

As a result, if I switch off the light with Home Assistant and then switch it on with a switch that is directly bindend, the lamp is at a brightness of 1%. I think this is a very big bug, so direct binding is not possible.

### What did you expect to happen?

The lights should always have the last brightness so that they can be switched from any platform and the last brightness is used.

### How to reproduce it (minimal and precise)

- Switching on the light and adjusting the brightness 
- Switch off light via zigbee2mqtt or mqtt - brightness at 1%
- Switch light on again and set brightness again 
- Switch off light with direct bind switch - brightness remains at old value.

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

7.4.5 [GA]

### Adapter

Smlight slzb-06m woth ember

### Setup

proxmox lxc

kilian-f-za, 2025-01-26:
Does nobody have an idea or has the error too? I think the error must be in zigbee2mqtt and the transition. I tried it with deconz and it works without problems, but I would prefer to stay with z2m.

chpego, 2025-01-30:
I'm interested in the subject, have you also observed that when you switch a light on or off with a transition from HA, the brightness values are like frozen, they go from their current value to their target value without any real transition?

kilian-f-za, 2025-01-31:
I have also tried a lot with Home Assistant and service.light. The transition worked, but the brightness was always set, so it didn't work when switching the light on with the binding.

Currently, I have a light template that adds a transition when using set_level, but not for turn_on and turn_off. This way, I can control the light with both HA and the binding.

Now, I need a light template for all lights that I want to control with the binding. I find the binding function important, especially for lighting.

closed 2025-04-09 by github-actions[bot]