Koenkk/zigbee2mqtt #25928: zigbee2mqtt failure

kamaldeepdhiman, 2025-01-21:
### What happened?

hi there, i update my zigbee2mqtt to version 2.0.0 and after this my adapter fail to load when i shutdown and restart my server and restart gain then zigbee2mqtt service not start again the i have to recreate zigbee2mqtt container again to start the service. i add in my configuration.yaml file the adapter: ezsp( its skyconnect).

what could be the solution my above issue please.

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

 7.1.1.0 build 273

### Adapter

Sky Connect

### Setup

docker container on pi5

flix4f, 2025-01-21:
Hi, 
There are a couple braking changes in version 2.0 as explained in here: https://github.com/Koenkk/zigbee2mqtt/discussions/24198
Did you go through all of the changes?
Have a look at this thread as well as it might be helpful: https://community.home-assistant.io/t/zigbee2mqtt-not-starting-after-update-to-2-0-0/822514/4

kamaldeepdhiman, 2025-01-21:
Hi there mate, I try all setting like mentioned below 

advanced:
  homeassistant_legacy_entity_attributes: false
  homeassistant_legacy_triggers: false
  legacy_api: false
  legacy_availability_payload: false
device_options:
  legacy: false

But issues arise only when I shutdown and then try to restart the service then only zigbee2mqtt container do not start, rest all containers starts fine.

manuelkue, 2025-01-22:
Have you tried to switch the adapter from `ezsp` to `ember` which is the default driver since 1.37.0 as `ezsp` is deprecated?
See https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.37.0 and https://github.com/Koenkk/zigbee2mqtt/discussions/21462

kamaldeepdhiman, 2025-01-22:
[_zigbee2mqtt_logs (1).txt](https://github.com/user-attachments/files/18508520/_zigbee2mqtt_logs.1.txt)
please see the attached log file. i changed my adapter from ezsp to ember and it shows me erros in log file.
my service is running fine with ezsp adapter setting but fails with ember adapter settings in configuration file. any further solution.

manuelkue, 2025-01-22:
How does your zigbee2mqtt configuration.yaml look like? Do you also have set up some configuration in the addon's ui config?

Have you entered the correct baudrate under `advanced`? See https://www.zigbee2mqtt.io/guide/configuration/adapter-settings.html

kamaldeepdhiman, 2025-01-22:
thanks mate. issue solved, my adapter needs software update. after update my skyconnect to latest sofware and with ember adapter settings in configuration file. now all start fine. working now. thanks

manuelkue, 2025-01-22:
Nice @kamaldeepdhiman, then you may close the issue :)

closed 2025-03-31 by github-actions[bot]