Koenkk/zigbee2mqtt #25929: Power usage spikes Ikea INSPELNING plugs

AbbyBos, 2025-01-21:
### What happened?

I use 10 Ikea inspelning plugs to monitor power. Once every 3/4 days a random plug responds with a big energy usage jump (40kwh -> 60kwh) in an hour. It only happens with the INSPELNING plugs, I have a assortment of other energy meters.

[image]

The data in homeassistants database does match the graph as shown below

[image]

But the powerusage does not match the energy usage graph.

[image]

### What did you expect to happen?

I expect the graph to be flat, and not jump by 15kwh every 3 days.

### How to reproduce it (minimal and precise)

I'm not able to trigger a failure manually. It happens every 3/4 days. It does not seems to be related to anything.

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20240710

### Adapter

zStack3x0 slzb-06

### Setup

Docker container, docker compose file with Z2M/HA/Mosquitto automaticly updated by watchtower

### Debug log

No debug logs yet, will activate it and share when it's availble.

mrmaximas, 2025-01-21:
Are the reporting settings the same as mine?

[image]

AbbyBos, 2025-01-21:
It seems to be identical.

[image]

mrmaximas, 2025-01-21:
What do you see in the STATE of device section now? Do you have the z2m info logs enabled and can you find the message when the power consumption increases dramatically?

AbbyBos, 2025-01-24:
Hi!

STATE is also simular:

{
    "current": 0.18,
    "energy": 79.46,
    "linkquality": 116,
    "power": 33.8,
    "power_on_behavior": "on",
    "state": "ON",
    "update": {
        "installed_version": 33816645,
        "latest_version": 33816645,
        "state": "idle"
    },
    "voltage": 235.9
}

I have the debug logs enabled, yesterday it happend again.:
[Debuglogs Z2M.txt](https://github.com/user-attachments/files/18532627/Debuglogs.Z2M.txt)

Event at 8:30:57 with 57.98 energy
[2025-01-23 08:30:57] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Server schakelaar Stroom Meting', payload '{"current":0.17,"energy":57.98,"identify":null,"linkquality":43,"power":33.7,"power_on_behavior":"on","state":"ON","update":{"installed_version":33816645,"latest_version":33816645,"state":"idle"},"voltage":235.9}'

Messages received with 78763 Wh
[2025-01-23 08:30:58] debug: 	z2m: Received Zigbee message from 'Server schakelaar Stroom Meting', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":78763}' from endpoint 1 with groupID 0

Event at 8:30:58 with 78.76 energy, +- 15 kwh more.
[2025-01-23 08:30:58] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Server schakelaar Stroom Meting', payload '{"current":0.17,"energy":78.76,"identify":null,"linkquality":47,"power":33.7,"power_on_behavior":"on","state":"ON","update":{"installed_version":33816645,"latest_version":33816645,"state":"idle"},"voltage":235.9}'

mrmaximas, 2025-01-24:
@AbbyBos IMHO it's IKEA's FW issue ;(

AbbyBos, 2025-01-24:
I'm afraid so. 
All the ikea zigbee devices seems to be great on the surface, but I have issues with multiple of them after a while. 
Let's hope for a firmware patch.

mrmaximas, 2025-01-24:
@AbbyBos I don't know a working way to let them know about the problem, but I see they update the firmwares regularly.

AbbyBos, 2025-01-24:
I didn't have this problem before I updated them with the new FW to fix the 23.4 volt -> 234 volt. I'm looking into a way to rollback the FW.

fleskalebas, 2025-04-07:
I think I have the same issue, although it's not a power spike, but more a 'build up' of data which gets released periodically.

With other words: let's say a device draw 20W of power. The IKEA socket doesn't report 20W per hour, but 100W after 5 hours.

Since that's not the real usage, my Energy dashboard shows it as a 'spike' resulting in x W of untracked (negative) power at the time it reported 100W.

If I check the history of the INSPELNING in Home Assitant, I see indeed only a few data points:

22:59:67: 15,6W
07:01:42: 22,2W
13:00:35: 15,4W

which results in a weird graph (many flat lines) like in the screenshots of the first post in the OP becasue of the few data points.

While my other TuYa smart socket reports every x seconds, thus has a nice graph showing every change, and also plays nice with the energy dashboard.

Maybe we need to play with the Reporting settings in Z2M, i.e. decreasing the values?
Current settings show a Min rep interval of 10, max 65000, min rep change 50.
A TuYa socket for example doesn't have any settings set in Reporting.

theoneandonlytaz, 2025-04-09:
Came here to say i am experiencing the same ghost kwh usage reporting issues as @AbbyBos with multiple Inspelning plugs. I am running latest ikea fw and latest zigbee2mqtt version.  I hope this can somehow be fixed. Has anyone had any luck with different settings? Or a workaround?

cpuks, 2025-04-18:
Same here - Voron 0.2 has spikes when it uses >5600 W, plug is on latest firmware.

eduardkirkosa, 2025-04-24:
Same here:

[image]

[image]

mikkelke, 2025-04-27:
I have the same spikes on my plug monitoring my dryer. 
[image]

[image]

No problem at my washer

[image]

[image]

aornt, 2025-06-23:
I also think this energy value is reported straight from the plug. I encountered the same problem with 2 plugs recently. A sudden jump from 2.09 kWh to 4.22 kWh. An other plug from 31.96 to 46.67 kWh in 2 hour interval, where normally the increments are 0.1 kWh in 3 hour interval. It is behind a Synology nas, so it is really an error.

AbbyBos, 2025-06-23:
Hi everyone,

I wasn't able to figure out what was happening with these sensors, but they kept ruining the dataset. That made the energy overview unusable.

My current workaround:
I've made a integral helper that calculates the energy use from the power, it was important to pick the "left" option of the integral settings to make it accurate for fast changing sensors. I've ran them parallel for a while and the deviation was acceptable for my usecase and better then the current situation with the big KWh spikes.

I've now replaced the onboard energy sensors in the entity with the integral.

kiwizznz, 2025-07-06:
Just replaced a bunch of older units with these and <1 day in and also have these spikes:

[image]

[image]

Have it on some lower power use devices and don't see spikes in the same way:

[image]

Some 'spikes' but in a sane range:

[image]

Is it possible the spikes are caused by some sort of inrush current? Maybe there is a capacitor or something and when a device is switched on it spikes or something?

moonfall84, 2025-08-24:
I have same issues with my 3 pieces of IKEA Inspelling. 
I have no idea how to deal with that. 2 pcs are on 2.4.45 firmware, 1 pcs is on 2.4.34. All of them have same issue...

Except that issue the plug works good, for sure better than Girier plugs which are freezing sometimes on X kW even for days.

Koenkk, 2025-09-24:
@marcgeerlings7 could you try reading the acPowerDivisor in both the normal and abnormal situation? The value will be logged in the log.

[image]

michtr1988, 2025-11-19:
Same issue here on both plugs I use, the energy consumed spikes up randomly by a random amount. Usually when that happens both plugs have a glitch at the same time:

[image]

Here you can see that it occurred once on October 19th, a smaller blip on October 22nd and then again on November 1st, The amount added is random, and it occurs on random days.

This is how it shows on the energy dashboard:

[image]

Using zigbee2mqtt ver 2.6.2. One plug is for my work PC and the other runs only a Synology DS923+ server, so both are fairly low loads with a daily consumption of ~1.3 kWh.

floriankisser, 2025-11-22:
I'm also on ZHA and have these jumps. I have 7 Inspelnings since Oct 2024, I had a few jumps back then directly after firmware updates. Then nothing until this month, where I had 4 larger jumps on 3 devices. The jumps are proportional to the actual sums, but far from doubling.

Patje1986, 2026-01-07:
I have this problem too. Any ideas on how we could fix this? One measurement works fine, the other never does, and the other has phantom energy jumps.

[image]
In my case, it's exactly jumps of 46kWh.

[image]

james-fry, 2026-03-08:
Sadly I am seeing this too today for the 1st time.

Before:

[image]

After:

[image]

No power draw at the time:

[image]

Is this a problem with the plug FW implementation, or with the z2m/zha code?

MRobi1, 2026-03-22:
Just adding another data point. I've got about 20 of these plugs scattered throughout the house. Today marks the 5th one where I've had a random sudden spike.

1,101kWh spike

[image]

HuismAndre, 2026-04-05:
FWIW: I have 2 Zigbee networks, one at home (120 devices) and one in the warehouse (45 devices). The 5 Inspelnings at home have all generated spikes (ranging from 7 to 1 times per device over de last 2 months). The 8 Inspelnings in the warehouse have never generated spikes. This leads me to believe that maybe it's an issue when they have to perfrorm a lot or router tasks. This makes sense (to me at least) since the one at the edge of the Zigbee network (the one behind the dishwasher) only has produced 1 spike.

All Inspelnings were bought at the same time, all have the same firmware. ZIgbee network at home is on channel 15 (with good to very good LQI), Zigbee network in the warehouse is on channel 25 with slightly higher average LQI.

[image]

andrei-lazarov, 2026-04-06:
So for energy spikes, does the value go up, and never back down to normal?

Try to update the divisor and multiplier from device -> dev console -> read attributes -> seMetering.
See if this gets the value back to normal.

[image]

HuismAndre, 2026-04-07:
[7-4-2026, 16:47:10] zhc:tz: Read result of 'seMetering': {"divisor":1000,"multiplier":1}

Since the acPowerDivisor is 10 (or 1, 100 or 1000), you would get a factor of 10, 100, 1000 error. In this case I even see jumps when there's nothing to report (and whatever times 0 is almost always still 0).

vandenberghev, 2026-10-02:
Not stale, unfortunately it's still happening...

REUW, 2026-10-09:
Same issue, but much bigger jumps.

[image]

HuismAndre, 2026-10-09:
Naah. No way near a winner ;-)

https://github.com/Koenkk/zigbee2mqtt/issues/25929#issuecomment-4106144425