Koenkk/zigbee2mqtt #25936: Hive SLR2 legacy attribute “running_state_heat” removal problem

fenty17, 2025-01-21:
### What happened?

Upgraded to 2.0 having thoroughly read release notes/breaking changes and applied config changes. All went smoothly. Great work to the team. 

Having run for a day or so I notice my heating_state template sensor isn’t working which is due to the removal of the “running_state_heat” attribute. I assume this is one if the deprecated legacy entity attributes that have now been removed. I also assume they can’t be added back in via config change. 

The problem is, this is the only way for me to accurately tell when there’s demand on my boiler so it is taking away that feature which, for me, I’ve relied on for recording history stats and comparing against my oil usage. 

The attribute is still shown in the z2m logs (see below), so I’m raising this issue to check if the removal is indeed intentional, request reinstating the attribute as an optional sensor, or at the least hopefully suggest another route for me to obtain my heating time. 

The same actually applies to the ‘system_mode_water’ attribute of the hot water.  Hopefully not too much of a long shot. 

`2025-01-21 21:16:58z2m:mqtt: MQTT publish: topic 'zigbee2mqtt2/Heating', payload '{"linkquality":18,"local_temperature_heat":19.23,"local_temperature_water":21,"occupied_heating_setpoint_heat":24.5,"occupied_heating_setpoint_water":22,"running_state_heat":"heat","running_state_water":"heat","system_mode_heat":"heat","system_mode_water":"off","temperature_setpoint_hold_duration_heat":65535,"temperature_setpoint_hold_duration_water":0,"temperature_setpoint_hold_heat":true,"temperature_setpoint_hold_water":false}'`

### What did you expect to happen?

Hoped to still achieve a working sensor. 

### How to reproduce it (minimal and precise)

Upgrade to 2.0, check entity attributes

### Zigbee2MQTT version

2.0.0 dev branch

### Adapter firmware version

20210708

### Adapter

zStack3x0

### Setup

Docker on sff pc separate to home assistant on a pi

chris-1243, 2025-01-21:
I would guess HA forum might more helpful.
Anyway, you may try an [`mqtt` trigger](https://www.home-assistant.io/docs/automation/trigger/#mqtt-trigger) and [templating](https://www.home-assistant.io/docs/configuration/templating/) the value.json or to create an mqtt [sensor](https://www.home-assistant.io/integrations/sensor.mqtt/)

fenty17, 2025-01-22:
Of course that would be a workaround. Don’t know why I didn’t think of that. I’ll work out a template for a manual mqtt sensor based on the topic’s json payload.

closed 2025-01-22 by fenty17