Koenkk/zigbee2mqtt #25938: Phillips Hue Twilight Lamp ( cannot save color settings in Home Assistant Scenes

wardwolfram, 2025-01-22:
### Is your feature request related to a problem? Please describe

I set the lamp front and back lights to various colors and brightness.  Using HA, I save the scene for the lamp's front and back lights. When I apply the different scenes for the twilight lamp in HA, the scene changes do not take effect. Nothing happens as the light continues to display its current setting. No saved scene changes are applied to the lamp

### Describe the solution you'd like

The ability to save multiple scene settings for the lamp.

Scene #1 - front lamp with Warm White & brightness = 35% + back lamp with color = RED & brightness = 15%

Scene #2 front and back lamp with Warm White & brightness = 25%

The ability for HA to successfully switch between Scene #1 and Scene #2

### Describe alternatives you've considered

Since I am looking to automate using the 2 scenes, I cannot think of an alternate.  Manually applying the front and back lamp settings is not acceptable.

### Additional context

No additional context. Thanks!

mundschenk-at, 2025-01-23:
Are you talking about dynamic scenes in HA (created using `scene.create`)? There are some issues with multi-endpoint devices in the Z2M frontend and Zigbee groups including multi-endpoint devices show the state of the last-changed constituent device, but I haven't seen any problems regarding HA control so far.

wardwolfram, 2025-01-25:
@mundschenk - Thanks for your input. I rest my scenes and automatons. The result was successful.  I was able to save the Twilight Lamp front/back light settings with scenes and switch between them.  This does work. Thank for your reply!

closed 2025-01-25 by wardwolfram