Koenkk/zigbee2mqtt #25940: Wrong log level "warn" can be selected, z2m refuses to start

tbowmo, 2025-01-22:
### What happened?

In the UI, you can select the following log levels

[image]

If you select "warn" z2m continues to run, but if it gets restarted somehow (either manual, power failure etc.) it refuses to start due to invalid log level, as seen in the logs in home assistant.

```
[07:32:44] INFO: Preparing to start...
[07:32:44] INFO: Socat not enabled
[07:32:45] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
AssertionError [ERR_ASSERTION]: 'warn' is not valid log_level, use one of 'error, warning, info, debug'
    at Logger.init (/app/lib/util/logger.ts:41:15)
    at new Controller (/app/lib/controller.ts:79:16)
    at start (/app/index.js:159:18)
[07:32:50] INFO: Preparing to start...
[07:32:50] INFO: Socat not enabled
[07:32:51] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
AssertionError [ERR_ASSERTION]: 'warn' is not valid log_level, use one of 'error, warning, info, debug'
    at Logger.init (/app/lib/util/logger.ts:41:15)
    at new Controller (/app/lib/controller.ts:79:16)
    at start (/app/index.js:159:18)
```

### What did you expect to happen?

Zigbee2Mqtt should be able to start, regardles of what loglevel you choose in the UI

### How to reproduce it (minimal and precise)

Go to logs, change `log level`to `warn`, observe that z2m is continuing normal operation. Now restart z2m, and observe that it refuses to start due to wrong log level

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

 7.4.3 [GA]

### Adapter

EmberZNet

### Setup

home assistant add-on (2.0.0-2)

manuelkue, 2025-01-22:
At least the option has been removed in https://github.com/Koenkk/zigbee2mqtt/commit/1667ba64ac2b8272cb0dea7ce30e459450e1c847 as far as I see. So the issue should be solved in an upcoming release / on the current dev branch.

Koenkk, 2025-01-22:
This has been fixed in the dev branch already indeed (will be included in the 1 February release)

Adastreia, 2025-01-23:
I think that this just happened to me, is there a way to fix this manually somewhere? I cannot use my lamps as it is right now >.<

tbowmo, 2025-01-23:
You need to manually edit the configuration.yaml file for z2m.

closed 2025-04-01 by github-actions[bot]