Koenkk/zigbee2mqtt #25941: Sonoff Presence Sensor SNZB-06P "Failed to configure"

Elderas42, 2025-01-22:
### What happened?

On startup the following error message is shown for all SNZB-06P presence sensors multiple times. The devices and automations seem to work anyhow, therefore I am just ignoring it - but of course would like to know if there is a way to fix it. Thanks

z2m: Failed to configure 'DEVICE_NAME', attempt 1 (Error: ZCL command 0x7cc6b6fffe0b427c/1 msOccupancySensing.configReport([{"minimumReportInterval":10,"maximumReportInterval":60,"reportableChange":0,"attribute":"occupancy"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 54027 - 1 - 12 - 1030 - 7 after 10000ms)
    at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:59:23)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:640:47)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:672:28)
    at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:422:20
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZStackAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:420:16)
    at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20)
    at Endpoint.sendRequest (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:325:20)
    at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:951:28)
    at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:688:9))

### What did you expect to happen?

No error messages in the log

### How to reproduce it (minimal and precise)

Restart and check Z2M logs

### Zigbee2MQTT version

 1.42.0

### Adapter firmware version

20210708

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

Pi5

mrmaximas, 2025-01-22:
the best choice would be to upgrade the stick fw with  https://github.com/Koenkk/Z-Stack-firmware

closed 2025-03-31 by github-actions[bot]

Elderas42, 2025-06-30:
I did the update and since then the error is not occurring anymore.