Koenkk/zigbee2mqtt #25945: z2m with CC2652P - ZCL error timeouts, network slowed down, unusual behaviour

Gecko3600, 2025-01-22:
### What happened?

Hi, 

I've been having problems with my Zigbee network for about 3 months now. It all started with the update to 1.39 z2m.

About my setup:
HomeAssistant runs on a Proxmox VM, 4 CPU cores, 16GB Ram, SSD as boot disk and z2m as container in HA. (Should be sufficiently dimensioned so far)
My network has approx. 80+ routers and 30+ end devices, the coordinator is an SLZB06-P connected centrally in the house via Ethernet. The nearest access point is two rooms away (approx. 8m as the crow flies). The 2.4GHz WLAN runs on channel 1 and Zigbee on 15.
There are therefore several routers in each room.
I have already tried all firmware versions of the coordinator (2022 - 2025) as well as a downgrade and several new installations of z2m. I am currently using the latest firmwares.

Coordinator: 2.7.0
Zigbee: 20240710
HA Core 2025.1.3
HA Supervisor 2024.12.3
HA Operating System 14.1
HA Frontend 20250109.0
z2m: 2.0.0.2

My network was very stable before, despite even more APs and parallel Zigbee networks in the same house (I have eliminated all of these and bundled them into one large network due to the house expansion stage. I have also flashed and integrated two other coordinators (SLZB06 and Sonoff Dongle-P) as routers.

My error picture looks like this:
- slow conversion of sent commands
- timeouts and no implementation
- massive problems pairing new devices
- multiple ZCL error timeouts

### What did you expect to happen?

I expect the network to run quickly and reliably. It feels like 20-30% of all commands to and from HA via z2m fail.

### How to reproduce it (minimal and precise)

I notice that the errors are not reproducible and occur randomly. Deactivating the APs did not help either.

### Zigbee2MQTT version

2.0.0.2

### Adapter firmware version

20240710

### Adapter

SLZB06-P

### Setup

HomeAssistant runs on a Proxmox VM, 4 CPU cores, 16GB Ram, SSD as boot disk and z2m as container in HA. (Should be sufficiently dimensioned so far) My network has approx. 80+ routers and 30+ end devices, the coordinator is an SLZB06-P connected centrally in the house via Ethernet. The nearest access point is two rooms away (approx. 8m as the crow flies). The 2.4GHz WLAN runs on channel 1 and Zigbee on 15.

### Debug log

[2025-01-22 16:54:01] debug: 	zh:controller:endpoint: Error: ZCL command 0xa4c1389452e77ffa/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 25508 - 1 - 50 - 6 - 11 after 10000ms)
    at Object.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/utils/waitress.ts:59:23)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:663:47)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:695:28)
    at /app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:443:20
    at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZStackAdapter.sendZclFrameToEndpoint (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:441:16)
    at Request.send (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20)
    at Endpoint.sendRequest (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:324:20)
    at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:949:28)
[2025-01-22 16:54:01] error: 	z2m: Publish 'set' 'state' to 'Birne Büro Ecklampe' failed: 'Error: ZCL command 0xa4c1389452e77ffa/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 25508 - 1 - 50 - 6 - 11 after 10000ms)'
[2025-01-22 16:54:01] debug: 	z2m: Error: ZCL command 0xa4c1389452e77ffa/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 25508 - 1 - 50 - 6 - 11 after 10000ms)
    at Object.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/utils/waitress.ts:59:23)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:663:47)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:695:28)
    at /app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:443:20
    at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZStackAdapter.sendZclFrameToEndpoint (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:441:16)
    at Request.send (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20)
    at Endpoint.sendRequest (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:324:20)
    at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:949:28)

(Its not only this exact buld, it includes every device...  switches, spotlights, bulbs...)

r0m1CH, 2025-01-23:
hi everyone,

I am in the exact same issue/behaviour.
I waited a bit to open an issue but it seems we are in the same boat as Gecko3600.

I'm running Z2M 2.0.0 since a week now.
I made my others updates (home assistant container) a week before to be sure I can identify the issue.

Since I migrated to 2.0.0 my zigbee network is slow and having a lot of time ZCL error timeouts/failed to ping where before everything was very stable. 
I found the LQI way lower than before but I'm not sure about it. Maybe it's just a feeling.

I have the a Sonoff ZBDongle coordinator (CC2652P) and latest firmware version for months. 

Coordinator type
zStack3x0
Coordinator revision
20240710

[image]

I remain available if you need more info/test from me.

Thanks.

vulpes2, 2025-01-24:
Had the same problem, latest fw has been perfectly fine for months, and devices have started dropping off immediately after upgrading Z2M. Ended up replacing the coordinator because I couldn't even get anything to join my network anymore. Here are some relevant issues, root cause is unknown so far, there are some mitigations in the comments that you can try, such as downgrading the firmware.

https://github.com/Koenkk/zigbee2mqtt/issues/23329
https://github.com/Koenkk/zigbee2mqtt/issues/25734
https://github.com/Koenkk/Z-Stack-firmware/issues/518

r0m1CH, 2025-01-24:
Based on the feedback from your threads, It seems that the old firmware (20221226) is working better then ? 
I will take time to downgrade to it and see if it's better/less clunky: 
[https://github.com/Koenkk/Z-Stack-firmware/raw/refs/tags/Z-Stack_3.x.0_coordinator_20221226/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20221226.zip](https://github.com/Koenkk/Z-Stack-firmware/issues/url)

@vulpes2  Thanks for pointing this out

Gecko3600, 2025-01-26:
Hi folks,
I also tested the 20221226 firmware and unfortunately there was no improvement... also with 20230507. (zigbee coordinator)
I have also tested all core firmware versions down to 0.9.9 with no noticeable improvement.

Today I installed the 2.7.1 core firmware and will report back in a few days whether the network has stabilized. But I don't have much hope as the release notes only list “Fixed Zigbee OTA updates for SLZB-06/p7/p10” or am i missing something?

r0m1CH, 2025-01-27:
Hi,

Same for me. it's  not better with an old version. I made some test with the 20221226 (48H) but  I went back to the last one.
I will wait on the newer version of z2m i guess.

Let me know if you find anything else.

kafisc1, 2025-01-28:
The problem has been going on for nine or more months now. And unfortunately it is independent of the controller and the firmware. I previously had a Sonoff Dongle-P and a SLZB-06.
Has anyone tried ZHA to see if the problem exists here too?

Gecko3600, 2025-02-06:
Hi, in the meantime I have removed all the bulbs from the power so that the routes build up more “stably” via the continuous current switches... I also tried to set the routes manually but nothing helps. I've debugged and rebuilt the entire 110devices+ network 3 times in the last 8 weeks. I currently have absolutely no desire to change everything to ZHA... I was actually hoping for a quick solution to the problem. I'm living with all the errors and dropouts and be frustrated to the max. I've also noticed that more and more people are getting in touch with this error.

arthurperezbessa, 2025-02-10:
It seems no one has figured out what might be causing the issue.

At my home, I occasionally see GenOnOff errors on some devices, but they still function as expected.

However, on the Home Assistant setup I installed for a client, the errors are much more frequent, and sometimes one of the switches doesn't receive commands.

I’ll try adding another router to strengthen the mesh network and see if that helps.

arthurperezbessa, 2025-02-11:
Hello Folks!

Unfortunately I couldn't find any solution. I'll migrate my Zigbee2MQTT to Tuya Zigbee and use the integration in Home Assistant.

My client can't wait for a solution

Gecko3600, 2025-02-24:
Hi, I have been testing back and forth for the last few weeks and currently it is almost stable and running at the old speed again. every now and then there are a few performance drops (lights react too slowly but still come on after a certain time and transmit their status, etc.). At that time I had already tested the old Zigbee version, but not with the freshly released SLZB firmware...

I have 20221226 as Zigbee firmware and 2.8.1. dev as coordinator firmware. 

I am currently leaving it as it is, have already made several backups of the configuration and am wary of changing anything in the configuration at the moment. :D

Translated with DeepL.com (free version)

closed 2025-05-04 by github-actions[bot]