Koenkk/zigbee2mqtt #25954: [Wrong device]: eWeLink SNZB-05 / SNZB-03

fat-freematilde, 2025-01-23:
### Link

https://www.aliexpress.com/item/1005007053606743.html?spm=a2g0o.productlist.main.21.d8d9362aajbjst&algo_pvid=cf9ad02c-9146-432e-9c7b-2f82aa759ceb&algo_exp_id=cf9ad02c-9146-432e-9c7b-2f82aa759ceb-10&pdp_npi=4%40dis%21USD%219.15%218.70%21%21%2166.28%2163.00%21%40211b807017376113412025228e2ffa%2112000039243169080%21sea%21DK%213180256429%21X&curPageLogUid=hklZlDhk4hhp&utparam-url=scene%3Asearch%7Cquery_from%3A

### Model

SNZB-05

### Description

Water Leak Sensor

### Vendor

eWeLink

### Picture (link)

https://www.zigbee2mqtt.io/images/devices/SNZB-05.png

### Database entry

{"id":45,"type":"EndDevice","ieeeAddr":"0xa4c138dcefea8d23","nwkAddr":30011,"manufId":4742,"manufName":"eWeLink","powerSource":"Battery","modelId":"SNZB-03","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1,1280,32],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"SNZB-03","manufacturerName":"eWeLink","powerSource":3,"zclVersion":3,"appVersion":113,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b002e1df877","zoneState":1}},"genPollCtrl":{"attributes":{"checkinInterval":14400}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":0,"batteryVoltage":22}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b002e1df877","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b002e1df877","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":7200,"repChange":10},{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":7200,"repChange":10}],"meta":{}}},"appVersion":113,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1737608392945,"checkinInterval":3600}

### Notes

Water leak sensor is detected as a motion sensor.
There seems to be an issue on the "Supported Devices" list too where the white label model is listed as a motion sensor.

[image]

fat-freematilde, 2025-01-23:
I make a rather nasty workaround, but it works for me:

In /app/node_modules/zigbee-herdsman-converters/devices/sonoff.js

I have renamed all instances of SNZB-03 to SNZB-03_disabled

And in /app/node_modules/zigbee-herdsman-converters/devices/ewelink.js 

I have added SNZB-03 to the SNZB-05 definition. 
```
    {                                                                   
        zigbeeModel: ['SNZB-05', 'SNZB-03'],
        model: 'SNZB-05',
        vendor: 'eWeLink',
        description: 'Zigbee water sensor',                     
        extend: [(0, modernExtend_1.battery)(), (0, modernExtend_1.iasZoneAlarm)({ zoneType: 'water_leak', zoneAttributes: ['alarm_1', 'batt
    },    
```

I know it's a gigantic hack, and changes will be lost if I update anything - But It works :-S

closed 2025-01-23 by fat-freematilde

RobertoCravallo, 2025-01-24:
I guess this is not something I can do with STUDIO CODE SERVER...

> I made a rather nasty workaround, but it works for me:
> 
> In /app/node_modules/zigbee-herdsman-converters/devices/sonoff.js
> 
> I have renamed all instances of SNZB-03 to SNZB-03_disabled
> 
> And in /app/node_modules/zigbee-herdsman-converters/devices/ewelink.js
> 
> I have added SNZB-03 to the SNZB-05 definition.
> 
> ```
>     {                                                                   
>         zigbeeModel: ['SNZB-05', 'SNZB-03'],
>         model: 'SNZB-05',
>         vendor: 'eWeLink',
>         description: 'Zigbee water sensor',                     
>         extend: [(0, modernExtend_1.battery)(), (0, modernExtend_1.iasZoneAlarm)({ zoneType: 'water_leak', zoneAttributes: ['alarm_1', 'battery_low'] })],
>     },    
> ```
> 
> I know it's a gigantic hack, and changes will be lost if I update anything - But It works :-S

I guess this is not something I can do with STUDIO CODE SERVER...

fat-freematilde, 2025-01-25:
I'm unsure what you mean. 
How do you run your z2m instance?

RobertoCravallo, 2025-01-25:
I meant, how d I make the changes? Can't find "app/node_modules/..." via STUDIO CODE SERVER. Maybe I missed it.

I actually have 3 instances running. https://www.youtube.com/watch?v=QOx733CU1p8

fat-freematilde, 2025-01-25:
It's inside your z2m container you need to find the files - So you need to connect to your docker instance and edit the files using vi

Take a look here: https://kodekloud.com/blog/docker-exec/ on how to connect to your container.

RobertoCravallo, 2025-01-25:
> It's inside your z2m container you need to find the files - So you need to connect to your docker instance and edit the files using vi
> 
> Take a look here: https://kodekloud.com/blog/docker-exec/ on how to connect to your container.

Thanks for the reply, but I'm going to live with the current condition. That is above my expertise.  :-(