Koenkk/zigbee2mqtt #25958: [Wrong device]:  GIDEALED ZC05M-20 reported as  GLEDOPTO GL-C-008P

Navino16, 2025-01-23:
### Link

https://www.amazon.fr/dp/B0CQC49N76

### Model

ZC05M-20

### Description

Zigbee LED Controller RGBCCT (pro+)

### Vendor

GIDEALED

### Picture (link)

https://m.media-amazon.com/images/I/71BQDrW3sfL._AC_SL1500_.jpg

### Database entry

{"id":11,"type":"Router","ieeeAddr":"0xa4c13840e88a8e4d","nwkAddr":50471,"manufId":4687,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-C-008P","epList":[11],"endpoints":{"11":{"profId":260,"epId":11,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"GL-C-008P","manufacturerName":"GLEDOPTO","powerSource":1,"zclVersion":3,"appVersion":1,"stackVersion":2,"hwVersion":2,"dateCode":"20240904","swBuildId":"20851203"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorTempPhysicalMin":158,"colorTempPhysicalMax":500}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":2,"dateCode":"20240904","swBuildId":"20851203","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1737640286880}

### Notes

Controls like State, Brightness, Colors seems to work, other controls seems broken

kpanchen, 2025-01-24:
This is correct, device presents itself as five different devices depending on the mode selected, documentation reflects this. Not an issue.

kpanchen, 2025-02-16:
> [@kpanchen](https://github.com/kpanchen) Okay but some controls are broken. For example color temperature is inverted (selecting warm color, output cold color)

This is a different issue which has nothing to do with not an issue raised in original post. I don't use this device as RGB+CCT but you need to make sure that connection is right or just check signal with multimeter. Warm White must be connected to the bottom terminal and the Cold White is the second from the bottom. 

I've checked couple of strips (for you) and they have connections in reverse, make sure you are not connecting them in order but according to designation, here's one example:

https://ledstrip.ca/led-strips-without-accessories/24v-5m-ip20-rgbw-cct-5050-led-strip-60-leds-m-strip-only.html

closed 2025-02-19 by Navino16