Koenkk/zigbee2mqtt #25962: [New device support]: OTH3600-GA-ZB Thermostat

mbdrlp, 2025-01-23:
### Link

https://www.ouellet.com/en-ca/products/thermostats-and-controls/oth3600-ga-zb.aspx

### Database entry

{"id":5,"type":"Router","ieeeAddr":"0x500b9140000204af","nwkAddr":3583,"manufId":4508,"manufName":"Sinope Technologies\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":"Mains (single phase)","modelId":"OTH3600-GA-ZB\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":769,"inClusterList":[0,3,4,5,513,516,1026,1794,2820,2821,65281],"outClusterList":[10,65281,25],"clusters":{"genBasic":{"attributes":{"modelId":"OTH3600-GA-ZB\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","manufacturerName":"Sinope Technologies\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":1,"zclVersion":2,"appVersion":2,"stackVersion":2,"hwVersion":0,"dateCode":"20191021\u0000\u0000\u0000\u0000\u0000\u0000\u0000","swBuildId":"746\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000"}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2031}},"haElectricalMeasurement":{"attributes":{}}},"binds":[{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0xe406bffffe21ad0c","endpointID":1}],"configuredReportings":[{"cluster":1026,"attrId":0,"minRepIntval":10,"maxRepIntval":3600,"repChange":100}],"meta":{}}},"appVersion":2,"stackVersion":2,"hwVersion":0,"dateCode":"20191021\u0000\u0000\u0000\u0000\u0000\u0000\u0000","swBuildId":"746\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1737659651445}

### Zigbee2MQTT version

2.0.0-2

### Comments

This thermostat is exactly the same as TH1300ZB From Sinopé Technologies (Same hardware, same features, only the PN# changes). TH1300ZB is already supported by Zigbee2MQTT so I guess it's easy to support the OTH3600.

### External definition

```shell
const {temperature, electricityMeter} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['OTH3600-GA-ZB'],
	model: 'OTH3600-GA-ZB'],
    vendor: 'Sinope Technologies',
	description: 'Automatically generated definition',
    extend: [temperature(), electricityMeter()],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

I was not able to use the external definition

mbdrlp, 2025-02-18:
Any update? I would be happy to help in any way.

mbdrlp, 2025-02-19:
with this external converter my thermostat is seen as supported
```
const m = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['OTH3600-GA-ZB'],
    model: 'OTH3600-GA-ZB',
    vendor: 'Sinope Technologies',
    description: 'Automatically generated definition',
    extend: [m.temperature(), m.electricityMeter()],
    meta: {},
};

module.exports = definition;
```

But I can't do anything else. What's the next step?

mbdrlp, 2025-02-21:
so far, the following external converter works:

```const m = require('zigbee-herdsman-converters/lib/modernExtend');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['OTH3600-GA-ZB'],
    model: 'OTH3600-GA-ZB',
    vendor: 'Sinope Technologies',
    description: 'Zigbee smart floor heating thermostat',
    extend: [m.temperature(), m.electricityMeter()],
	fromZigbee: [fz.hvac_user_interface, fz.ignore_temperature_report],
	toZigbee: [
            tz.thermostat_local_temperature,
            tz.thermostat_occupied_heating_setpoint,
            tz.thermostat_unoccupied_heating_setpoint,
            tz.thermostat_temperature_display_mode,
            tz.thermostat_keypad_lockout,
            tz.thermostat_system_mode,            
        ],
		exposes: [
            e
                .climate()
                .withSetpoint('occupied_heating_setpoint', 5, 36, 0.5)
                .withSetpoint('unoccupied_heating_setpoint', 5, 36, 0.5)
                .withLocalTemperature()
                .withSystemMode(['off', 'heat'], ea.ALL, 'Mode of the thermostat')
                .withPiHeatingDemand()
                .withRunningState(['idle', 'heat'], ea.STATE),
			e.enum('thermostat_occupancy', ea.ALL, ['unoccupied', 'occupied']).withDescription('Occupancy state of the thermostat'),
            e
                .enum('second_display_mode', ea.ALL, ['auto', 'setpoint', 'outdoor temp'])
                .withDescription(
                    'Displays the outdoor temperature and then returns to the set point in "auto" mode, or clears ' +
                        'in "outdoor temp" mode when expired.',
                ),
            e
                .numeric('thermostat_outdoor_temperature', ea.ALL)
                .withUnit('°C')
                .withValueMin(-99.5)
                .withValueMax(99.5)
                .withValueStep(0.5)
                .withDescription('Outdoor temperature for the secondary display'),
            e
                .numeric('outdoor_temperature_timeout', ea.ALL)
                .withUnit('s')
                .withValueMin(30)
                .withValueMax(64800)
                .withPreset('15 min', 900, '15 minutes')
                .withPreset('30 min', 1800, '30 minutes')
                .withPreset('1 hour', 3600, '1 hour')
                .withDescription('Time in seconds after which the outdoor temperature is considered to have expired'),
            e
                .binary('enable_outdoor_temperature', ea.ALL, 'ON', 'OFF')
                .withDescription('DEPRECATED: Use second_display_mode or control via outdoor_temperature_timeout'),
            e
                .enum('temperature_display_mode', ea.ALL, ['celsius', 'fahrenheit'])
                .withDescription('The temperature format displayed on the thermostat screen'),
            e.enum('time_format', ea.ALL, ['24h', '12h']).withDescription('The time format featured on the thermostat display'),
            e.enum('backlight_auto_dim', ea.ALL, ['on_demand', 'sensing']).withDescription('Control backlight dimming behavior'),
            e.enum('keypad_lockout', ea.ALL, ['unlock', 'lock1']).withDescription('Enables or disables the device’s buttons'),
		],		
    meta: {},
};

module.exports = definition;```

I tried to add some other functionality from other Sinope converter, but without success. I have errors in my logs and my external convert doesn't work.
With this one I am able to control my thermostat. But I still have an error at startup:
```2025-02-21 09:18:37z2m: Failed to configure 'Bathroom 2 - Thermostat', attempt 3 (Error: ZCL command 0x500b9140000204af/1 haElectricalMeasurement.read(["acPowerDivisor","acPowerMultiplier"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:349:28) at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:954:26) at Endpoint.read (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:445:29) at result.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/lib/modernExtend.ts:1944:37) at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/index.ts:272:21) at Configure.configure (/app/lib/extension/configure.ts:124:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17))```

mbdrlp, 2025-08-02:
To keep open....

Dalto1, 2026-01-11:
@mbdrlp i got this converter that is working well on my side

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const m = require('zigbee-herdsman-converters/lib/modernExtend');

const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    thermostat_basic: {
        cluster: 'hvacThermostat',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            if (msg.data.hasOwnProperty('localTemp')) {
                result.local_temperature = msg.data.localTemp / 100;
            }
            if (msg.data.hasOwnProperty('occupiedHeatingSetpoint')) {
                result.occupied_heating_setpoint = msg.data.occupiedHeatingSetpoint / 100;
            }
            if (msg.data.hasOwnProperty('unoccupiedHeatingSetpoint')) {
                result.unoccupied_heating_setpoint = msg.data.unoccupiedHeatingSetpoint / 100;
            }
            if (msg.data.hasOwnProperty('systemMode')) {
                result.system_mode = {0: 'off', 4: 'heat'}[msg.data.systemMode];
            }
            return result;
        },
    },
};

const definition = {
    zigbeeModel: ['OTH3600-GA-ZB'],
    model: 'OTH3600-GA-ZB',
    vendor: 'Sinope Technologies',
    description: 'Zigbee smart floor heating thermostat with power metering',
    extend: [m.electricityMeter()],
    fromZigbee: [
        fzLocal.thermostat_basic,
        fz.hvac_user_interface,
    ],
    toZigbee: [
        tz.thermostat_occupied_heating_setpoint,
        tz.thermostat_unoccupied_heating_setpoint,
        tz.thermostat_temperature_display_mode,
        tz.thermostat_keypad_lockout,
        tz.thermostat_system_mode,
    ],
    exposes: [
        e.climate()
            .withSetpoint('occupied_heating_setpoint', 5, 36, 0.5)
            .withSetpoint('unoccupied_heating_setpoint', 5, 36, 0.5)
            .withLocalTemperature()
            .withSystemMode(['off', 'heat'], ea.ALL, 'Mode of the thermostat'),
        e.enum('temperature_display_mode', ea.ALL, ['celsius', 'fahrenheit'])
            .withDescription('The temperature format displayed on the thermostat screen'),
        e.enum('keypad_lockout', ea.ALL, ['unlock', 'lock1'])
            .withDescription('Enables or disables the device buttons (lock1 = locked)'),
    ],
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        const binds = [
            'genBasic',
            'genIdentify',
            'hvacThermostat',
            'hvacUserInterfaceCfg',
            'haElectricalMeasurement',
            'seMetering',
        ];
        await reporting.bind(endpoint, coordinatorEndpoint, binds);
        await reporting.thermostatTemperature(endpoint);
        await reporting.thermostatOccupiedHeatingSetpoint(endpoint);
        await reporting.thermostatSystemMode(endpoint);
        await reporting.readEletricalMeasurementMultiplierDivisors(endpoint);
        await reporting.activePower(endpoint);
        await reporting.readMeteringMultiplierDivisor(endpoint);
        await reporting.currentSummDelivered(endpoint);
    },
};

module.exports = definition;
```

mbdrlp, 2026-01-12:
@Dalto1 yes but the outdoor temperature is missing and would like to display it on my thermostat. I have exactly the same thermostat from Sinope (and it's the same hardware, Sinope written on the PCB), and it works well, including the outdoor temperature

mbdrlp, 2026-01-13:
wow it works! thank you for your help!