Koenkk/zigbee2mqtt #25964: Namron 4512768 reports 10x Voltage use and 1000x Amps

Aeliassen, 2025-01-23:
### What happened?

Both L1 and L2 reports 2300V and 64A.
For V this is 10x what it should be, for A this is 1000x what it should be

### What did you expect to happen?

I expect it to report 230V and 0,064A. There is a 15W lightbulb connected to each of the L1 and L2 

### How to reproduce it (minimal and precise)

It seems to affect all users that I have talked with that use this device

### Zigbee2MQTT version

2.0.0.2

### Adapter firmware version

7.3.1.0 build 176

### Adapter

?

### Setup

Add-On on Homeassistant on raspberry pi

Koenkk, 2025-01-25:
Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/3e055027efc32723af07e2dcf8b6ec6a): 

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, 
- in the frontend, check if the description of this device now ends with `CUSTOM` (this indicates the external converter has been loaded correctly)
- reconfigure the device using the yellow refresh button.
- check if issue is fixed

Aeliassen, 2025-01-26:
For some reason I can't get the external converter to load. 

I made the ext_converter.js file inside my zigbee2mqtt folder. I stopped the z2m addon, and added the external_converters to the configuration.yaml, and started z2m addon again.

When I look up the switches they still have "Zigbee 2 channel switch" as description, and not "Zigbee 2 channel switch CUSTOM" as the ext_converter.js says. 

Does it matter where in the configuration.yaml file I add the code?

Koenkk, 2025-01-27:
Whoops the instructions are outdated, new ones:

Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/3e055027efc32723af07e2dcf8b6ec6a): 

- save this as file next to `configuration.yaml` in the `external_converters` directory (create if it doesn't exist yet) as `ext_converter.js` (so `external_converters/ext_converter.js`)
- start z2m, 
- in the frontend, check if the description of this device now ends with `CUSTOM` (this indicates the external converter has been loaded correctly)
- check if issue is fixed

Aeliassen, 2025-01-27:
I have tried to save `ext_converter.js` at `external_converters/ext_converter.js` and at `data/external_converters/ext_converter.js` as it says at z2m website. 

After both these saves I started z2m, but still no CUSTOM in description. 

Am I correct in understanding that there should be no changes to `configuration.yaml`? 

Language problems is making me unsure if you ment to tell me that i should save the `external_converters` folder in the same folder as configuration.yaml (in my case with HA on RPI inside the zigbee2mqtt folder)? Or should there be a `configuration.yaml` inside the `external_converters` folder? 

[image]

Koenkk, 2025-01-28:
The `external_converters` folders should be next to the **Zigbee2MQTT** (not the HA one!) `configuration.yaml` Your z2m `configuration.yaml` should contain something like this:

```
version: 4
mqtt:
....
```

Aeliassen, 2025-01-29:
I got a friend that have the same switch to test this in his z2m. He have the same experience. After following the solution described above his device does not get CUSTOM in the description (the same as I experience) 

It seems like the external converter thing just is not working on both of our devices. Is there any logging I can turn on to see if z2m tries to use the external converts?

Koenkk, 2025-01-29:
@Aeliassen does your logging show `Loaded external converter` when starting up z2m? If not, the external converter is not in the correct location.

Aeliassen, 2025-01-29:
I get this in my log file
```
[2025-01-29 21:19:01] error: 	z2m: Failed to call 'ExternalConverters' 'start' (node:internal/modules/cjs/loader:1252
  throw err;
  ^
Error: Cannot find module 'zigbee-herdsman-converters/lib/fromZigbee'
Require stack:
- /app/dist/extension/externalJS.js
- /app/dist/extension/externalConverters.js
- /app/dist/controller.js
- /app/index.js
    at Function._resolveFilename (node:internal/modules/cjs/loader:1249:15)
    at Function._load (node:internal/modules/cjs/loader:1075:27)
    at TracingChannel.traceSync (node:diagnostics_channel:315:14)
    at wrapModuleLoad (node:internal/modules/cjs/loader:218:24)
    at Module.require (node:internal/modules/cjs/loader:1340:12)
    at require (node:internal/modules/helpers:141:16)
    at /app/data/extension/ext_converter.js:2:12
    at Script.runInContext (node:vm:149:12)
    at Script.runInNewContext (node:vm:154:17)
    at runInNewContext (node:vm:310:38))
```

I see a lot of people have had this problem, and solved it with creating the file with Terminal, but thats not working for me. Any idea why this is happening?

Koenkk, 2025-01-30:
Made a mistake, fixed it now: https://gist.github.com/Koenkk/3e055027efc32723af07e2dcf8b6ec6a

Aeliassen, 2025-01-30:
It workes now. It changes some of the order of everything under Exposes, but the V and A are still wrong. Here is a screenshot of the exposes with the above convertor

[image]

and here is a screenshot before I used the new convertor

[image]

Koenkk, 2025-02-01:
Could you provide the data/database.db entry of this device?

Aeliassen, 2025-02-02:
I tried to find information about database.db in the documentation, but could not find anythin. Is this what you are asking for? 

```
{"id":52,"type":"Router","ieeeAddr":"0xdc8e95fffe357b52","nwkAddr":2813,"manufId":4098,"manufName":"Namron AS","powerSource":"Mains (single phase)","modelId":"4512768","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,1794,2820,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"32768":0,"modelId":"4512768","manufacturerName":"Namron AS","powerSource":1,"zclVersion":8,"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"20230410","swBuildId":"1.09"}},"genOnOff":{"attributes":{"onOff":0,"startUpOnOff":0}},"haElectricalMeasurement":{"attributes":{"rmsCurrent":0,"rmsVoltage":2326,"activePower":0}},"seMetering":{"attributes":{"currentSummDelivered":20401}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe9f2ef7","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe9f2ef7","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe9f2ef7","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0,"manufacturerCode":null},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1],"manufacturerCode":null},{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":257}],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[0,3,4,5,6,1794,2820,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0,"hwVersion":0}},"genOnOff":{"attributes":{"onOff":0}},"haElectricalMeasurement":{"attributes":{"rmsVoltage":2323,"rmsCurrent":0,"activePower":0}},"seMetering":{"attributes":{"currentSummDelivered":8044}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe9f2ef7","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0,"manufacturerCode":null},{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"20230410","swBuildId":"1.09","zclVersion":8,"interviewCompleted":true,"meta":{"configured":-1251565648},"lastSeen":1738494855290}
```

Koenkk, 2025-02-02:
Interesting, there is no divisor, could you provide the debug log when re-configuring the device (yellow refresh) button?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

Aeliassen, 2025-02-02:
I must have done something wrong the first time I tried the external converter. This time it worked. I now see Power, Voltage and Amps like expected.

Koenkk, 2025-02-03:
Great, integrated the fix!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-02-03 by Koenkk