Koenkk/zigbee2mqtt #25968: Fail to pair SNZB-04P.

apoorv569, 2025-01-23:
### What happened?

I have a Sonoff contact sensor, SNZB-04P, trying to pair it fails. Attached the logs below.

The website says it is supported, https://www.zigbee2mqtt.io/supported-devices/#s=snzb-04p

I am trying to connect this via `Sonoff Zigbee 3.0 USB Dongle E Plus V2`

I have a Temperature and Humidity sensor SNZB-02, which works fine.

### What did you expect to happen?

It should have paired fine.

### How to reproduce it (minimal and precise)

Try to pair SNZB-04P.

### Zigbee2MQTT version

Zigbee2MQTT version     2.0.0 commit: unknown

### Adapter firmware version

Coordinator revision     8.0.2 [GA]

### Adapter

Sonoff Zigbee 3.0 USB Dongle E Plus V2 with ember driver

### Setup

Add-on on HomeAssistant

### Debug log

```
info 2025-01-24 03:05:32z2m: Zigbee: allowing new devices to join.
info 2025-01-24 03:05:32zh:ember: [STACK STATUS] Network opened.
info 2025-01-24 03:05:32z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254},"status":"ok","transaction":"j2smv-2"}'
info 2025-01-24 03:05:40zh:controller: Interview for '0x0ceff6fffe3f0f18' started
info 2025-01-24 03:05:40z2m: Device 'Door Sensor' joined
info 2025-01-24 03:05:40z2m: Starting interview of 'Door Sensor'
info 2025-01-24 03:05:40z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Door Sensor","ieee_address":"0x0ceff6fffe3f0f18"},"type":"device_joined"}'
info 2025-01-24 03:05:40z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Door Sensor","ieee_address":"0x0ceff6fffe3f0f18","status":"started"},"type":"device_interview"}'
info 2025-01-24 03:05:40z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Door Sensor","ieee_address":"0x0ceff6fffe3f0f18"},"type":"device_announce"}'
error 2025-01-24 03:05:49zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
error 2025-01-24 03:05:49zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
error 2025-01-24 03:05:49zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
error 2025-01-24 03:05:49zh:ember: Adapter fatal error: HOST_FATAL_ERROR
info 2025-01-24 03:05:49zh:ember:uart:ash: ASH COUNTERS since last clear:
info 2025-01-24 03:05:49zh:ember:uart:ash: Total frames: RX=79, TX=122
info 2025-01-24 03:05:49zh:ember:uart:ash: Cancelled : RX=0, TX=0
info 2025-01-24 03:05:49zh:ember:uart:ash: DATA frames : RX=64, TX=41
info 2025-01-24 03:05:49zh:ember:uart:ash: DATA bytes : RX=1173, TX=495
info 2025-01-24 03:05:49zh:ember:uart:ash: Retry frames: RX=13, TX=3
info 2025-01-24 03:05:49zh:ember:uart:ash: ACK frames : RX=0, TX=77
info 2025-01-24 03:05:49zh:ember:uart:ash: NAK frames : RX=0, TX=0
info 2025-01-24 03:05:49zh:ember:uart:ash: nRdy frames : RX=0, TX=0
info 2025-01-24 03:05:49zh:ember:uart:ash: CRC errors : RX=0
info 2025-01-24 03:05:49zh:ember:uart:ash: Comm errors : RX=0
info 2025-01-24 03:05:49zh:ember:uart:ash: Length < minimum: RX=0
info 2025-01-24 03:05:49zh:ember:uart:ash: Length > maximum: RX=0
info 2025-01-24 03:05:49zh:ember:uart:ash: Bad controls : RX=0
info 2025-01-24 03:05:49zh:ember:uart:ash: Bad lengths : RX=0
info 2025-01-24 03:05:49zh:ember:uart:ash: Bad ACK numbers : RX=0
info 2025-01-24 03:05:49zh:ember:uart:ash: Out of buffers : RX=0
info 2025-01-24 03:05:49zh:ember:uart:ash: Retry dupes : RX=13
info 2025-01-24 03:05:49zh:ember:uart:ash: Out of sequence : RX=0
info 2025-01-24 03:05:49zh:ember:uart:ash: ACK timeouts : RX=3
error 2025-01-24 03:05:54zh:ember:ezsp: ERROR Transaction failure; status=ASH_ERROR_TIMEOUTS. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=39 Len=10].
error 2025-01-24 03:05:56zh:controller: Interview failed for '0x0ceff6fffe3f0f18 with error 'Error: ASH_ERROR_TIMEOUTS'
error 2025-01-24 03:05:56z2m: Failed to interview 'Door Sensor', device has not successfully been paired
info 2025-01-24 03:05:56z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Door Sensor","ieee_address":"0x0ceff6fffe3f0f18","status":"failed"},"type":"device_interview"}'
```

bobo-cher, 2025-01-24:
You flashed SLZB07MG24 firmware in Sonoff ZBDongle-E and it works?

> error 2025-01-24 03:05:49zh:ember:uart:ash: Received ERROR from adapter, with code=**ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT**.
> error 2025-01-24 03:05:49zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR

Looks like related to #23861.

Try this firmware: https://github.com/Nerivec/silabs-firmware-builder/releases/tag/v2024.6.2-update5. It should fix the error.

apoorv569, 2025-01-24:
> You flashed SLZB07MG24 firmware in Sonoff ZBDongle-E and it works?
>
>
> Looks like related to [#23861](https://github.com/Koenkk/zigbee2mqtt/issues/23861).
>
> Try this firmware: [...]

Oops, I copied the wrong link here by mistake, I updated the original post.

bobo-cher, 2025-01-24:
Mine was disconnecting during OTA updates. The new firmware fixed the issue. Chances are high it will fix the issue for you too.

bobo-cher, 2025-01-24:
Same firmware but compiled with other options and that makes the difference.

apoorv569, 2025-01-24:
> Same firmware but compiled with other options and that makes the difference.

Could you share a direct link for it?

bobo-cher, 2025-01-24:
https://github.com/Nerivec/silabs-firmware-builder/releases/download/v2024.6.2-update5/sonoff_zbdonglee_zigbee_ncp_8.0.2.0_115200_sw_flow.gbl

apoorv569, 2025-01-25:
> https://github.com/Nerivec/silabs-firmware-builder/releases/download/v2024.6.2-update5/sonoff_zbdonglee_zigbee_ncp_8.0.2.0_115200_sw_flow.gbl

Just flashed this one, the device paired almost instant now and seems to be working. However OTA update failed for me. I waited for sometime after clicking `Check all` then sometime after it said this,

```
[2025-01-25 07:44:20] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{},"error":"Failed to check if update available for 'Door Sensor' (Device didn't respond to OTA request)","status":"error","transaction":"selse-5"}'
[2025-01-25 07:44:20] error: 	z2m: Failed to check if update available for 'Door Sensor' (Device didn't respond to OTA request)
```

bobo-cher, 2025-01-25:
This is normal for battery powered sensors as their radio sleep most of the time and do not respond to requests.
Try waking it up just before you check for updates and then immediately wake it up again after the check is initiated. That worked for me, it reported "no updates available".

apoorv569, 2025-01-25:
> This is normal for battery powered sensors as their radio sleep most of the time and do not respond to requests. Try waking it up just before you check for updates and then immediately wake it up again after the check is initiated. That worked for me, it reported "no updates available".

How to wake them up? like press the pair button or use them?

I tried both BTW and I got error the dongle disconnected like before, here are the logs,

```
[2025-01-25 21:27:06] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Door Sensor', payload '{"battery":100,"battery_low":false,"contact":false,"linkquality":176,"tamper":true,"update":{"installed_version":8704,"latest_version":8704,"state":"idle"},"voltage":3000}'
[2025-01-25 21:27:12] info: 	z2m: Checking if update available for 'Door Sensor'
[2025-01-25 21:27:30] error: 	zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2025-01-25 21:27:30] error: 	zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
[2025-01-25 21:27:30] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2025-01-25 21:27:30] error: 	zh:ember: Adapter fatal error: HOST_FATAL_ERROR
[2025-01-25 21:27:30] info: 	zh:ember:uart:ash: ASH COUNTERS since last clear:
[2025-01-25 21:27:30] info: 	zh:ember:uart:ash:   Total frames: RX=52, TX=59
[2025-01-25 21:27:30] info: 	zh:ember:uart:ash:   Cancelled   : RX=0, TX=0
[2025-01-25 21:27:30] info: 	zh:ember:uart:ash:   DATA frames : RX=47, TX=8
[2025-01-25 21:27:30] info: 	zh:ember:uart:ash:   DATA bytes  : RX=1646, TX=217
[2025-01-25 21:27:30] info: 	zh:ember:uart:ash:   Retry frames: RX=4, TX=0
[2025-01-25 21:27:30] info: 	zh:ember:uart:ash:   ACK frames  : RX=0, TX=51
[2025-01-25 21:27:30] info: 	zh:ember:uart:ash:   NAK frames  : RX=0, TX=0
[2025-01-25 21:27:30] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2025-01-25 21:27:30] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2025-01-25 21:27:30] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2025-01-25 21:27:30] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2025-01-25 21:27:30] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2025-01-25 21:27:30] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2025-01-25 21:27:30] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2025-01-25 21:27:30] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2025-01-25 21:27:30] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2025-01-25 21:27:30] info: 	zh:ember:uart:ash:   Retry dupes     : RX=4
[2025-01-25 21:27:30] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2025-01-25 21:27:30] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=0
[2025-01-25 21:27:30] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Door Sensor', payload '{"battery":100,"battery_low":false,"contact":true,"linkquality":192,"tamper":true,"update":{"installed_version":8704,"latest_version":8704,"state":"idle"},"voltage":3000}'
[2025-01-25 21:27:30] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Door Sensor', payload '{"battery":100,"battery_low":false,"contact":false,"linkquality":204,"tamper":true,"update":{"installed_version":8704,"latest_version":8704,"state":"idle"},"voltage":3000}'
[2025-01-25 21:27:30] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Door Sensor', payload '{"battery":100,"battery_low":false,"contact":true,"linkquality":196,"tamper":true,"update":{"installed_version":8704,"latest_version":8704,"state":"idle"},"voltage":3000}'
[2025-01-25 21:27:30] info: 	zh:ember:uart:ash: Port closed.
[2025-01-25 21:27:30] info: 	zh:ember:uart:ash: ======== ASH stopped ========
[2025-01-25 21:27:30] info: 	zh:ember:ezsp: ======== EZSP stopped ========
[2025-01-25 21:27:30] info: 	zh:ember: ======== Ember Adapter Stopped ========
[2025-01-25 21:27:30] error: 	z2m: Adapter disconnected, stopping
[2025-01-25 21:27:30] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2025-01-25 21:27:30] info: 	z2m: Disconnecting from MQTT server
[2025-01-25 21:27:30] info: 	z2m: Stopping zigbee-herdsman...
[2025-01-25 21:27:31] info: 	z2m: Stopped zigbee-herdsman
[2025-01-25 21:27:31] info: 	z2m: Stopped Zigbee2MQTT
```

closed 2025-04-03 by github-actions[bot]