Koenkk/zigbee2mqtt #25971: Bosch BTH-RA pi_heating_demand RW in z2m but RO in HA

tbrasser, 2025-01-24:
### What happened?

In z2m I can directly control the valve but in HA it shows as a sensor instead of a number entity. I can open a PR but couldn't find where exactly this is mapped.

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20240710

### Adapter

zStack3x0

### Setup

Addon

tbrasser, 2025-02-04:
[image]

[image]

closed 2025-04-15 by github-actions[bot]

Koenkk, 2025-04-15:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.