Koenkk/zigbee2mqtt #25976: Group state inconsistent with the state of included devices

rccoleman, 2025-01-24:
### What happened?

Revisiting the closed issue https://github.com/Koenkk/zigbee2mqtt/issues/25522

This inconsistent state issue is happening to me pretty regularly now and I'm wondering if it's a race or ordering issue between device state reporting and evaluation of the group state. Here's an example of where I think it went wrong, with some of my own debug inserted. Here's the full log:

[2025-01-23 19:47:40] debug:    zh:zstack:unpi:parser: --- parseNext []
[2025-01-23 19:47:40] debug:    zh:zstack:unpi:parser: <-- [254,27,68,129,0,0,6,0,116,150,1,1,0,54,0,77,162,96,0,0,7,24,76,10,0,0,16,0,116,150,29,53]
[2025-01-23 19:47:40] debug:    zh:zstack:unpi:parser: --- parseNext [254,27,68,129,0,0,6,0,116,150,1,1,0,54,0,77,162,96,0,0,7,24,76,10,0,0,16,0,116,150,29,53]
[2025-01-23 19:47:40] debug:    zh:zstack:unpi:parser: --> parsed 27 - 2 - 4 - 129 - [0,0,6,0,116,150,1,1,0,54,0,77,162,96,0,0,7,24,76,10,0,0,16,0,116,150,29] - 53
[2025-01-23 19:47:40] debug:    zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":38516,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":54,"securityuse":0,"timestamp":6333005,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[24,76,10,0,0,16,0]}}
[2025-01-23 19:47:40] debug:    zh:controller: Received payload: clusterID=6, address=38516, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=54, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":76,"commandIdentifier":10},"payload":[{"attrId":0,"dataType":16,"attrData":0}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2025-01-23 19:47:40] debug:    zh:zstack:unpi:parser: --- parseNext []
[2025-01-23 19:47:40] debug:    z2m: Received Zigbee message from 'Kit Sink', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
[2025-01-23 19:47:40] debug:    z2m: Group: Kit Light Group, Device: Kit Sink, Current State: ON
[2025-01-23 19:47:40] debug:    z2m: Kit Sink is ON
[2025-01-23 19:47:40] debug:    z2m: Group: Morning Group, Device: Kit (DR), Current State: ON
[2025-01-23 19:47:40] debug:    z2m: Kit (DR) is ON
[2025-01-23 19:47:40] debug:    z2m: Group: Garage Door Open Light Group, Device: Garage Workshop Lights, Current State: OFF
[2025-01-23 19:47:40] debug:    z2m: Group: Garage Door Open Light Group, Device: DNH (Bar), Current State: OFF
[2025-01-23 19:47:40] debug:    z2m: Group: Garage Door Open Light Group, Device: Kit (DR), Current State: ON
[2025-01-23 19:47:40] debug:    z2m: Kit (DR) is ON
[2025-01-23 19:47:40] debug:    z2m: Group: Kit All Lights Group, Device: Kit (DR), Current State: ON
[2025-01-23 19:47:40] debug:    z2m: Kit (DR) is ON
[2025-01-23 19:47:40] debug:    z2m: Group: Kit All Lights Group New, Device: Kit Desk, Current State: OFF
[2025-01-23 19:47:40] debug:    z2m: Group: Kit All Lights Group New, Device: Kit Pendants, Current State: ON
[2025-01-23 19:47:40] debug:    z2m: Kit Pendants is ON
[2025-01-23 19:47:40] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqttp/Kit Sink', payload '{"action":null,"activeEnergyReports":0,"activePowerReports":0,"autoTimerOff":null,"auxSwitchUniqueScenes":null,"bindingOffToOnSyncLevel":null,"brightness":254,"brightnessLevelForDoubleTapDown":null,"brightnessLevelForDoubleTapUp":null,"buttonDelay":"0ms","defaultLed1ColorWhenOff":null,"defaultLed1ColorWhenOn":null,"defaultLed1IntensityWhenOff":null,"defaultLed1IntensityWhenOn":null,"defaultLed2ColorWhenOff":null,"defaultLed2ColorWhenOn":null,"defaultLed2IntensityWhenOff":null,"defaultLed2IntensityWhenOn":null,"defaultLed3ColorWhenOff":null,"defaultLed3ColorWhenOn":null,"defaultLed3IntensityWhenOff":null,"defaultLed3IntensityWhenOn":null,"defaultLed4ColorWhenOff":null,"defaultLed4ColorWhenOn":null,"defaultLed4IntensityWhenOff":null,"defaultLed4IntensityWhenOn":null,"defaultLed5ColorWhenOff":null,"defaultLed5ColorWhenOn":null,"defaultLed5IntensityWhenOff":null,"defaultLed5IntensityWhenOn":null,"defaultLed6ColorWhenOff":null,"defaultLed6ColorWhenOn":null,"defaultLed6IntensityWhenOff":null,"defaultLed6IntensityWhenOn":null,"defaultLed7ColorWhenOff":null,"defaultLed7ColorWhenOn":null,"defaultLed7IntensityWhenOff":null,"defaultLed7IntensityWhenOn":null,"defaultLevelLocal":null,"defaultLevelRemote":null,"deviceBindNumber":null,"dimmingSpeedDownLocal":null,"dimmingSpeedDownRemote":null,"dimmingSpeedUpLocal":null,"dimmingSpeedUpRemote":0,"doubleTapClearNotifications":null,"doubleTapDownToParam56":null,"doubleTapUpToParam55":null,"energy":8.01,"fanControlMode":null,"fanLedLevelType":null,"firmwareUpdateInProgressIndicator":null,"highLevelForFanControlMode":null,"higherOutputInNonNeutral":null,"identify":null,"individual_led_effect":{"color":""},"internalTemperature":null,"invertSwitch":null,"ledBarScaling":null,"ledColorForFanControlMode":null,"ledColorWhenOff":null,"ledColorWhenOn":null,"ledIntensityWhenOff":null,"ledIntensityWhenOn":null,"led_effect":null,"linkquality":54,"loadLevelIndicatorTimeout":null,"localProtection":null,"lowLevelForFanControlMode":null,"maximumLevel":null,"mediumLevelForFanControlMode":null,"minimumLevel":null,"onOffLedMode":null,"outputMode":"On/Off","overheat":null,"periodicPowerAndEnergyReports":null,"power":0,"powerType":null,"quickStartLevel":null,"quickStartTime":null,"rampRateOffToOnLocal":null,"rampRateOffToOnRemote":null,"rampRateOnToOffLocal":null,"rampRateOnToOffRemote":null,"relayClick":"Enabled (Click Sound Off)","remoteProtection":null,"singleTapBehavior":null,"smartBulbMode":null,"state":"OFF","stateAfterPowerRestored":null,"state_1":"ON","state_2":"OFF","switchType":null,"update":{"installed_version":16908818,"latest_version":16908818,"state":"idle"},"update_available":false}'
Here, a message is received that the Kit Sink device turned off:

[2025-01-23 19:47:40] debug:    z2m: Received Zigbee message from 'Kit Sink', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
That kicks off an evalution of the Kit Light Group group, where it sees that Kit Sink is still ON:

[2025-01-23 19:47:40] debug:    z2m: Group: Kit Light Group, Device: Kit Sink, Current State: ON
[2025-01-23 19:47:40] debug:    z2m: Kit Sink is ON
And later we see that the new state of Kit Sink of OFF is finally reported to MQTT with
"state":"OFF",

Z2M never reevaluates the state of that group, even after a restart of Z2M, presumably until there's another state change that would affect the members in that group. Sometimes the state is properly reflected and the group state changes and sometimes it doesn't, leading me to think that there's a race condition.

This leads to a situation where if I review the group in the Z2M UI, I can clearly see that all the included devices are off and are properly reported as such in MQTT and HA, but the state of the group itself in MQTT and HA is wrong and is never corrected.

### What did you expect to happen?

I expect that the state of a group reflects the state of the included devices at all time, and that a state change of one of the included devices causes the group state to be reevaluated based on the *new* state of that device.

### How to reproduce it (minimal and precise)

I can reproduce this with the following:

1. Create a group that contains physical switches that control loads
2. Bind one of the physical switches to the Zigbee group using Z2M
3. Use HA or MQTT to turn that group on
4. Observe that the devices in that group turn on and the group state of `on` is reflected properly reflected in MQTT and HA
5. Turn off the physical switch from #2 and observe that both the switch and the rest of the group devices turn off, as expected
6. Note that the state of the group in MQTT and HA is still `on`, while the states of all the devices included in the group in the Z2M UI, MQTT and HA are all `off`, as expected.  In other words, the group state is inconsistent with the device states.

As per Rene's comments in https://github.com/Koenkk/zigbee2mqtt/issues/25522, I will play around with creating separate groups for manual and remote control, with the former excluding the switches and the latter including them, but it doesn't seem like the user should need to go through that level of complexity.

### Zigbee2MQTT version

[... cut ...]
                    logger_1.default.debug(device.name + " is ON");                                                                
                    return false;                                                                                                    
                }                                                 
            }
        }                                                       
        return true;           
    } 
```

It just outputs whenever a group state is reevaluated due to a device state change and shows the new group state and which device kept it on, if any.

rccoleman, 2025-01-24:
I just created a separate group with two additional lights (`Kit Sink` and `Kit Pendants`) and bound it to the two switches that I do control, named `Kit Light Group Noswitch`.  I kept the original group that includes those two devices, along with the switches I manually control, called `Kit Light Group`.  If I turn on `Kit Light Group`, all the lights turn on and the state is properly reflected everywhere.  If I then turn off one of the switches, all the lights respond properly (they turn off), but the group state is still `on` and I see exactly the same behavior in the log as I described above.  See the log entries starting at
`[2025-01-24 08:24:17] debug:    z2m: Received Zigbee message from 'Kit (FR)', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0`

[log.log](https://github.com/user-attachments/files/18538963/log.log)

rccoleman, 2025-01-24:
Curiouser and Curiouser.  It seems like the device state is maybe being cached and doesn't reflect what's being published to MQTT.  Here's another log with less noise:

```
[2025-01-24 11:35:33] info: 	z2m: Entity update: Kit (FR), State: OFF
[2025-01-24 11:35:33] info: 	z2m: Checking Group: Kit Light Group, Device: Kit (FR)
[2025-01-24 11:35:33] info: 	z2m: Group: Kit Light Group, Device: Kit Sink, Current State: ON
[2025-01-24 11:35:33] info: 	z2m: Kit Sink is ON
[2025-01-24 11:35:33] info: 	z2m: Checking Group: Morning Group, Device: Kit (FR)
[2025-01-24 11:35:33] info: 	z2m: Group: Morning Group, Device: Kit (DR), Current State: ON
[2025-01-24 11:35:33] info: 	z2m: Kit (DR) is ON
[2025-01-24 11:35:33] info: 	z2m: Checking Group: Garage Door Open Light Group, Device: Kit (FR)
[2025-01-24 11:35:33] info: 	z2m: Group: Garage Door Open Light Group, Device: Garage Workshop Lights, Current State: OFF
[2025-01-24 11:35:33] info: 	z2m: Group: Garage Door Open Light Group, Device: DNH (Bar), Current State: OFF
[2025-01-24 11:35:33] info: 	z2m: Group: Garage Door Open Light Group, Device: Kit (DR), Current State: ON
[2025-01-24 11:35:33] info: 	z2m: Kit (DR) is ON
[2025-01-24 11:35:33] info: 	z2m: Checking Group: Kit All Lights Group, Device: Kit (FR)
[2025-01-24 11:35:33] info: 	z2m: Group: Kit All Lights Group, Device: Kit (DR), Current State: ON
[2025-01-24 11:35:33] info: 	z2m: Kit (DR) is ON
[2025-01-24 11:35:33] info: 	z2m: Checking Group: Kit All Lights Group New, Device: Kit (FR)
[2025-01-24 11:35:33] info: 	z2m: Group: Kit All Lights Group New, Device: Kit Desk, Current State: OFF
[2025-01-24 11:35:33] info: 	z2m: Group: Kit All Lights Group New, Device: Kit Pendants, Current State: ON
[2025-01-24 11:35:33] info: 	z2m: Kit Pendants is ON
[2025-01-24 11:35:33] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqttp/Kit (FR)', payload '{"action":null,"activeEnergyReports":0,"activePowerReports":0,"autoTimerOff":null,"auxSwitchUniqueScenes":null,"bindingOffToOnSyncLevel":"Disabled","brightness":254,"brightnessLevelForDoubleTapDown":null,"brightnessLevelForDoubleTapUp":null,"buttonDelay":"0ms","defaultLed1ColorWhenOff":null,"defaultLed1ColorWhenOn":null,"defaultLed1IntensityWhenOff":101,"defaultLed1IntensityWhenOn":101,"defaultLed2ColorWhenOff":null,"defaultLed2ColorWhenOn":null,"defaultLed2IntensityWhenOff":null,"defaultLed2IntensityWhenOn":null,"defaultLed3ColorWhenOff":null,"defaultLed3ColorWhenOn":null,"defaultLed3IntensityWhenOff":null,"defaultLed3IntensityWhenOn":null,"defaultLed4ColorWhenOff":null,"defaultLed4ColorWhenOn":null,"defaultLed4IntensityWhenOff":null,"defaultLed4IntensityWhenOn":null,"defaultLed5ColorWhenOff":null,"defaultLed5ColorWhenOn":null,"defaultLed5IntensityWhenOff":null,"defaultLed5IntensityWhenOn":null,"defaultLed6ColorWhenOff":null,"defaultLed6ColorWhenOn":null,"defaultLed6IntensityWhenOff":null,"defaultLed6IntensityWhenOn":null,"defaultLed7ColorWhenOff":null,"defaultLed7ColorWhenOn":null,"defaultLed7IntensityWhenOff":null,"defaultLed7IntensityWhenOn":null,"defaultLevelLocal":null,"defaultLevelRemote":null,"deviceBindNumber":null,"dimmingSpeedDownLocal":null,"dimmingSpeedDownRemote":null,"dimmingSpeedUpLocal":127,"dimmingSpeedUpRemote":0,"doubleTapClearNotifications":null,"doubleTapDownToParam56":null,"doubleTapUpToParam55":null,"energy":0,"fanControlMode":null,"fanLedLevelType":null,"firmwareUpdateInProgressIndicator":"Disabled","highLevelForFanControlMode":null,"higherOutputInNonNeutral":null,"identify":null,"individual_led_effect":{"duration":"","level":""},"internalTemperature":null,"invertSwitch":null,"ledBarScaling":null,"ledColorForFanControlMode":null,"ledColorWhenOff":null,"ledColorWhenOn":null,"ledIntensityWhenOff":null,"ledIntensityWhenOn":null,"led_effect":null,"linkquality":76,"loadLevelIndicatorTimeout":null,"localProtection":null,"lowLevelForFanControlMode":null,"maximumLevel":null,"mediumLevelForFanControlMode":null,"minimumLevel":null,"onOffLedMode":"All","outputMode":"On/Off","overheat":null,"periodicPowerAndEnergyReports":null,"power":0.2,"powerType":null,"quickStartLevel":null,"quickStartTime":null,"rampRateOffToOnLocal":null,"rampRateOffToOnRemote":null,"rampRateOnToOffLocal":null,"rampRateOnToOffRemote":null,"relayClick":"Enabled (Click Sound Off)","remoteProtection":null,"singleTapBehavior":null,"smartBulbMode":"Disabled","state":"OFF","stateAfterPowerRestored":null,"state_1":"ON","state_2":"OFF","switchType":"Single Pole","update":{"installed_version":16908818,"latest_version":16908818,"state":"idle"},"update_available":false}'
[2025-01-24 11:35:33] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqttp/Kit (FR)', payload '{"action":"","activeEnergyReports":0,"activePowerReports":0,"autoTimerOff":null,"auxSwitchUniqueScenes":null,"bindingOffToOnSyncLevel":"Disabled","brightness":254,"brightnessLevelForDoubleTapDown":null,"brightnessLevelForDoubleTapUp":null,"buttonDelay":"0ms","defaultLed1ColorWhenOff":null,"defaultLed1ColorWhenOn":null,"defaultLed1IntensityWhenOff":101,"defaultLed1IntensityWhenOn":101,"defaultLed2ColorWhenOff":null,"defaultLed2ColorWhenOn":null,"defaultLed2IntensityWhenOff":null,"defaultLed2IntensityWhenOn":null,"defaultLed3ColorWhenOff":null,"defaultLed3ColorWhenOn":null,"defaultLed3IntensityWhenOff":null,"defaultLed3IntensityWhenOn":null,"defaultLed4ColorWhenOff":null,"defaultLed4ColorWhenOn":null,"defaultLed4IntensityWhenOff":null,"defaultLed4IntensityWhenOn":null,"defaultLed5ColorWhenOff":null,"defaultLed5ColorWhenOn":null,"defaultLed [... cut ...]
[2025-01-24 11:35:33] info: 	z2m: Checking Group: Kit Light Group, Device: Kit (DR)
[2025-01-24 11:35:33] info: 	z2m: Group: Kit Light Group, Device: Kit Sink, Current State: ON
[2025-01-24 11:35:33] info: 	z2m: Kit Sink is ON
```

I added some code to ignore the device responsible for the state change because the logic in the group state algorithm in `areAllMembersOffOrClosed` assumes it was changed to OFF, and that gets past the first state change, but it never really changes to OFF and is back ON in the next state change.

mundschenk-at, 2025-01-24:
Might this be related to #24204? I have found groups that contain a dual-endpoint device to reflect the state of the last-updated member endpoint, whereas groups containing only single-endpoint devices appear to work as expected.

rccoleman, 2025-01-24:
Could be, but this is new behavior with Z2M 2.0 and there's been activity around this exact code in the last month.  In my case, all the devices are Inovelli Blue switches and they do each have two endpoints.

mundschenk-at, 2025-01-24:
Now that you mention this, maybe what I've seen with multi-endpoint devices is not the same as #24204 as I've only noticed this behavior on 2.0 (doesn't necessarily mean it wasn't there before, though).

rccoleman, 2025-01-24:
I feel like it's important that it's only an issue when there are state changes coming in from devices that didn't start with a command from Z2M.  If Z2M turns off the group, I see all the state changes come through and everything works fine.  If I turn off the group from a physical switch, but which is bound to the same group, the state changes come into Z2M, but they don't seem to stick for the purposes of the group state evaluation.  In both cases, the state of the individual devices in Z2M is consistent, only the group state is out of sync.  Quite vexing.

mundschenk-at, 2025-01-24:
That's definitely something different from what I've seen (I haven't experimented with direct bindings). But it appears that the 2.0 group state evaluation is somewhat wonky.

rccoleman, 2025-01-26:
So it seems clear that the incoming state hasn't been set yet, or isn't being queried properly, when the group state is evaluated.

```
[2025-01-25 16:06:55] info: 	z2m: Entity update: Kit Sink, State: OFF, endpointName: undefined, Endpoint: {"_events":{},"_eventsCount":0,"deviceID":257,"inputClusters":[0,3,4,5,6,8,1794,2820,2821,64599,64561],"outputClusters":[25],"profileID":260,"ID":1,"clusters":{"genBasic":{"attributes":{"modelId":"VZM31-SN","manufacturerName":"Inovelli","powerSource":1,"zclVersion":8,"appVersion":17,"stackVersion":1,"hwVersion":1,"dateCode":"20230602","swBuildId":"2.18"}},"haElectricalMeasurement":{"attributes":{"acPowerMultiplier":1,"acPowerDivisor":10,"activePower":287}},"seMetering":{"attributes":{"multiplier":1,"divisor":100,"currentSummDelivered":806,"notificationControlFlags":[0,474]}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":254}},"manuSpecificInovelliVZM31SN":{"attributes":{}},"manuSpecificInovelli":{"attributes":{"dimmingSpeedUpRemote":25}}},"deviceIeeeAddress":"0xb43a31fffe2f1f96","deviceNetworkAddress":38516,"_binds":[{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c3d377","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c3d377","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c3d377","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c3d377","endpointID":1}],"_configuredReportings":[{"cluster":8,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":1,"manufacturerCode":null},{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0,"manufacturerCode":null},{"cluster":2820,"attrId":1291,"minRepIntval":15,"maxRepIntval":3600,"repChange":1,"manufacturerCode":null},{"cluster":1794,"attrId":0,"minRepIntval":15,"maxRepIntval":3600,"repChange":0,"manufacturerCode":null},{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":2820,"attrId":1291,"minRepIntval":15,"maxRepIntval":3600,"repChange":1},{"cluster":1794,"attrId":0,"minRepIntval":15,"maxRepIntval":3600,"repChange":0}],"meta":{},"pendingRequests":{"sendInProgress":false,"ID":1,"deviceIeeeAddress":"0xb43a31fffe2f1f96"}}
[2025-01-25 16:06:55] info: 	z2m: Checking Group: Kit Light Group, Device: Kit Sink
[2025-01-25 16:06:55] info: 	z2m: Group: Kit Light Group, Device: Kit Sink, Current State: ON
```
The `onStateChange` update for `Kit Sink` is clearly `OFF`, but when the state is queried down below in `areAllMembersOffOrClosed`, it's still `ON`.  It's like the group has a local, cached version of the state of devices that isn't being updated or is frozen at some point.  `this.state` is for the group is coming from the Extension class, and I don't know when that's being updated yet.

rccoleman, 2025-01-26:
So this seems to work, but I can't test devices that actually do report their state based on endpoint:

```
    areAllMembersOffOrClosed(group, deviceChanged) {
        for (const member of group.zh.members) {
            const device = this.zigbee.resolveEntity(member.getDevice());
            if (this.state.exists(device)) {
                const state = this.state.get(device);
                const endpointNames = device.isDevice() && device.getEndpointNames();
                let stateKey = 'state';
                if (device.definition.meta &&
                    device.definition.meta.multiEndpoint &&
                    (!device.definition.meta.multiEndpointSkip || !device.definition.meta.multiEndpointSkip.includes('state')) &&
                    endpointNames && endpointNames.length >= member.ID) {
                    stateKey = `state_${endpointNames[member.ID - 1]}`;
                }
    
                if (state[stateKey] === 'ON' || state[stateKey] === 'OPEN') { 
                    return false;
                }
            }
        }
        return true;
    }
```

This is from `/app/dist/extension/groups.js` from inside the container, which is how I was testing.  The critical part is this:

```
                let stateKey = 'state';
                if (device.definition.meta &&
                    device.definition.meta.multiEndpoint &&
                    (!device.definition.meta.multiEndpointSkip || !device.definition.meta.multiEndpointSkip.includes('state')) &&
                    endpointNames && endpointNames.length >= member.ID) {
                    stateKey = `state_${endpointNames[member.ID - 1]}`;
                }
```

rccoleman, 2025-01-26:
I opened https://github.com/Koenkk/zigbee2mqtt/pull/26019 for this.  Seems to work for me.

rohankapoorcom, 2025-01-28:
Nice work @rccoleman - Just tested your fix in my installation running develop/edge and it's fixed!

Koenkk, 2025-01-28:
Thanks for the fix @rccoleman !

closed 2025-01-28 by Koenkk