Koenkk/zigbee2mqtt #25980: SLZB-06M configuration or support error

Gronaldo1, 2025-01-24:
### What happened?

Advised configuration of SLZB-06M not supported by Zigbee2MQTT.

2025-01-24 19:02:00.318 WARNING (MainThread) [supervisor.addons.options] Option 'advanced' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)

### What did you expect to happen?

Setting not supported
  advanced:
    transmit_power: 20

### How to reproduce it (minimal and precise)

Advised configuration as mentioned in device webportal and also confimed by SMLight tech support

# Pay attention, if you use the Z2M addon for HA, it is better to edit Z2M addon yaml configuration file directly (Select the 3 dot menu in the upper right corner → "Edit in YAML")
serial:
# Location of SLZB-06M
  port: tcp://192.168.2.2:6638
  baudrate: 115200
  adapter: ember
# Disable green led?
  disable_led: false
# Set output power to max 20
**advanced:
  transmit_power: 20**

### Zigbee2MQTT version

2.0.0-2

### Adapter firmware version

20241127 / v2.7.1

### Adapter

SMLight SLZB-06M

### Setup

HAOS 14.1 / Pi5

roccr, 2025-02-01:
I was able to fix it with newest 2.8.0 release: https://x.com/smlight_tech/status/1885438350825246756

I'm usinga a SLZB-06 (without the M)

Content of the tweet:

✅ Works on: SLZB-06, SLZB-06p7, SLZB-06p10
🚧 Coming soon: SLZB-06M, SLZB-06Mg24

Gronaldo1, 2025-02-06:
2.8.0dev does not resolve issue for me, also updated HA core to 25.2.0

Barml, 2025-03-26:
Same issue for me...  
WARNING (MainThread) [supervisor.addons.options] Option 'advanced' does not exist in the schema for Zigbee2MQTT
Core 2025.2.5, OS15.0

Gronaldo1, 2025-05-27:
Issue still present