Koenkk/zigbee2mqtt #25984: zigbee2mqtt sometimes  not sending ir commands when published from MQTT

guyliorha, 2025-01-24:
### What happened?

zigbee2mqtt sometimes not sending ir commands when published from MQTT.

### What did you expect to happen?

I expected that all commands will be send to the Universal smart IR remote control UFO-R11

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

7.4.5 [GA]

### Adapter

UFRO11

### Setup

add on in home assistance

### Debug log

[2025-01-24 23:57:41] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/IR Remote 1/set' with data '{"ir_code_to_send": "CPoiuhElAqoGQyADAiUCQ2AB4AEHgBeAG+ADC0Af4AMLAmICBuAUOwCAYB/gB1/gAQlAPwBiYB/gBT9AGwCLIAdAG+ALY+ATB4A/4AGf4A+DQC/gA3vgCZ3gDb/gA1/gEYuAjcCL4A8z4Qlb4AUT"}'
[2025-01-24 23:57:41] debug: 	z2m: Publishing 'set' 'ir_code_to_send' to 'IR Remote 1'
[2025-01-24 23:57:41] debug: 	zhc:zosung: Sending IR code: "CPoiuhElAqoGQyADAiUCQ2AB4AEHgBeAG+ADC0Af4AMLAmICBuAUOwCAYB/gB1/gAQlAPwBiYB/gBT9AGwCLIAdAG+ALY+ATB4A/4AGf4A+DQC/gA3vgCZ3gDb/gA1/gEYuAjcCL4A8z4Qlb4AUT"
[2025-01-24 23:57:41] debug: 	zh:controller:endpoint: ZCL command 0xe406bffffef0db2c/1 zosungIRTransmit.zosungSendIRCode00({"seq":0,"length":226,"unk1":0,"unk2":57348,"unk3":1,"cmd":2,"unk4":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false})
[2025-01-24 23:57:41] debug: 	zh:ember: ~~~> [ZCL to=0xe406bffffef0db2c:61701 apsFrame={"profileId":260,"clusterId":60672,"sourceEndpoint":1,"destinationEndpoint":1,"options":4352,"groupId":0,"sequence":0} header={"frameControl":{"reservedBits":0,"frameType":1,"direction":0,"disableDefaultResponse":true,"manufacturerSpecific":false},"transactionSequenceNumber":14,"commandIdentifier":0}]
[2025-01-24 23:57:41] debug: 	zh:ember:ezsp: ===> [FRAME: ID=52:"SEND_UNICAST" Seq=48 Len=40]
[2025-01-24 23:57:41] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=0 frmRx=3](ackRx=0)
[2025-01-24 23:57:41] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-01-24 23:57:41] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1](ackRx=0 frmTx=1)
[2025-01-24 23:57:41] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1 frmNum=3](frmRx=3) Added to rxQueue
[2025-01-24 23:57:41] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=4](ackRx=1)
[2025-01-24 23:57:41] debug: 	zh:ember:ezsp: <=== [FRAME: ID=52:"SEND_UNICAST" Seq=48 Len=7]
[2025-01-24 23:57:41] debug: 	zh:ember:ezsp: ~~~> [SENT type=DIRECT apsSequence=146 messageTag=15 status=OK]
[2025-01-24 23:57:41] debug: 	zhc:zosung: Sending IR code initiated.
[2025-01-24 23:57:41] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/IR Remote 1/set' with data '{"ir_code_to_send": "BxgjnBFiAm0GQAMEgAIGAmIgA0AHBIAC6AGAYBNAAwBPoAfgASeAH0AHQDsAJWA3wDuAQ+AHL4Af4AVfwC/gDz/Ag+API0B/QJvgDRvgAYfgDXsAniCD4AFz4A9f4AVX4AXb4AUj4BU74A+/wQdAx+ALv0Bb"}'
[2025-01-24 23:57:41] debug: 	z2m: Publishing 'set' 'ir_code_to_send' to 'IR Remote 1'
[2025-01-24 23:57:41] debug: 	zhc:zosung: Sending IR code: "BxgjnBFiAm0GQAMEgAIGAmIgA0AHBIAC6AGAYBNAAwBPoAfgASeAH0AHQDsAJWA3wDuAQ+AHL4Af4AVfwC/gDz/Ag+API0B/QJvgDRvgAYfgDXsAniCD4AFz4A9f4AVX4AXb4AUj4BU74A+/wQdAx+ALv0Bb"
[2025-01-24 23:57:41] debug: 	zh:controller:endpoint: ZCL command 0xe406bffffef0db2c/1 zosungIRTransmit.zosungSendIRCode00({"seq":0,"length":234,"unk1":0,"unk2":57348,"unk3":1,"cmd":2,"unk4":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false})
[2025-01-24 23:57:41] debug: 	zh:ember: ~~~> [ZCL to=0xe406bffffef0db2c:61701 apsFrame={"profileId":260,"clusterId":60672,"sourceEndpoint":1,"destinationEndpoint":1,"options":4352,"groupId":0,"sequence":0} header={"frameControl":{"reservedBits":0,"frameType":1,"direction":0,"disableDefaultResponse":true,"manufacturerSpecific":false},"transactionSequenceNumber":15,"commandIdentifier":0}]
[2025-01-24 23:57:41] debug: 	zh:ember:ezsp: ===> [FRAME: ID=52:"SEND_UNICAST" Seq=49 Len=40]
[2025-01-24 23:57:41] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=1 frmRx=4](ackRx=1)
[2025-01-24 23:57:41] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-01-24 23:57:41] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=2](ackRx=1 frmTx=2)
[2025-01-24 23:57:41] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=2 frmNum=4](frmRx=4) Added to rxQueue
[2025-01-24 23:57:41] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=5](ackRx=2)
[2025-01-24 23:57:41] debug: 	zh:ember:ezsp: <=== [FRAME: ID=52:"SEND_UNICAST" Seq=49 Len=7]
[2025-01-24 23:57:41] debug: 	zh:ember:ezsp: ~~~> [SENT type=DIRECT apsSequence=147 messageTag=16 status=OK]
[2025-01-24 23:57:41] debug: 	zhc:zosung: Sending IR code initiated.
[2025-01-24 23:57:42] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-01-24 23:57:42] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=2](ackRx=2 frmTx=2)
[2025-01-24 23:57:42] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=2 frmNum=5](frmRx=5) Added to rxQueue
[2025-01-24 23:57:42] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=6](ackRx=2)
[2025-01-24 23:57:42] debug: 	zh:ember:ezsp: <=== [CBFRAME: ID=63:"MESSAGE_SENT_HANDLER" Seq=49 Len=22]
[2025-01-24 23:57:42] debug: 	zh:ember:ezsp: ezspMessageSentHandler(): callback called with: [status=OK], [type=DIRECT], [indexOrDestination=61701], [apsFrame={"profileId":260,"clusterId":60672,"sourceEndpoint":1,"destinationEndpoint":1,"options":4352,"groupId":0,"sequence":146}], [messageTag=15]
[2025-01-24 23:57:42] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-01-24 23:57:42] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=2](ackRx=2 frmTx=2)
[2025-01-24 23:57:42] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=2 frmNum=6](frmRx=6) Added to rxQueue
[2025-01-24 23:57:42] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=7](ackRx=2)
[2025-01-24 23:57:42] debug: 	zh:ember:ezsp: <=== [CBFRAME: ID=63:"MESSAGE_SENT_HANDLER" Seq=49 Len=22]
[2025-01-24 23:57:42] debug: 	zh:ember:ezsp: ezspMessageSentHandler(): callback called with: [status=OK], [type=DIRECT], [indexOrDestination=61701], [apsFrame={"profileId":260,"clusterId":60672,"sourceEndpoint":1,"destinationEndpoint":1,"options":4352,"groupId":0,"sequence":147}], [messageTag=16]
[2025-01-24 23:57:42] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-01-24 23:57:42] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=2](ackRx=2 frmTx=2)
[2025-01-24 23:57:42] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=2 frmNum=7](frmRx=7) Added to rxQueue
[2025-01-24 23:57:42] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=0](ackRx=2)
[... 30 lines cut ...]
[2025-01-24 23:57:42] debug: 	zhc:zosung: "IR-Message-Code02" received (msg:{"seq":0,"position":0,"maxlen":64})
[2025-01-24 23:57:42] debug: 	zh:controller:endpoint: ZCL command 0xe406bffffef0db2c/1 zosungIRTransmit.zosungSendIRCode03({"zero":0,"seq":0,"position":0,"msgpart":{"type":"Buffer","data":[123,34,107,101,121,95,110,117,109,34,58,49,44,34,100,101,108,97,121,34,58,51,48,48,44,34,107,101,121,49,34,58,123,34,110,117,109,34,58,49,44,34,102,114,101,113,34,58,51,56]},"msgpartcrc":207}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false})
[2025-01-24 23:57:42] debug: 	zh:ember: ~~~> [ZCL to=0xe406bffffef0db2c:61701 apsFrame={"profileId":260,"clusterId":60672,"sourceEndpoint":1,"destinationEndpoint":1,"options":4352,"groupId":0,"sequence":0} header={"frameControl":{"reservedBits":0,"frameType":1,"direction":0,"disableDefaultResponse":true,"manufacturerSpecific":false},"transactionSequenceNumber":16,"commandIdentifier":3}]
[2025-01-24 23:57:42] debug: 	zh:ember:ezsp: ===> [FRAME: ID=52:"SEND_UNICAST" Seq=51 Len=83]
[2025-01-24 23:57:42] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=3 frmRx=3](ackRx=3)
[2025-01-24 23:57:42] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-01-24 23:57:42] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4](ackRx=3 frmTx=4)
[2025-01-24 23:57:42] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4 frmNum=3](frmRx=3) Added to rxQueue
[2025-01-24 23:57:42] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=4](ackRx=4)
[2025-01-24 23:57:42] debug: 	zh:ember:ezsp: <=== [FRAME: ID=52:"SEND_UNICAST" Seq=51 Len=7]
[2025-01-24 23:57:42] debug: 	zh:ember:ezsp: ~~~> [SENT type=DIRECT apsSequence=149 messageTag=18 status=OK]
[... cut ...]
[2025-01-24 23:57:42] debug: 	zh:controller: Received payload: clusterID=60672, address=61701, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=100, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":87,"commandIdentifier":2},"payload":{"seq":0,"position":50,"maxlen":64},"command":{"ID":2,"parameters":[{"name":"seq","type":33},{"name":"position","type":35},{"name":"maxlen","type":32}],"name":"zosungSendIRCode02"}}

lux73, 2025-01-26:
in my Opinion it's not really a z2m issue - more the Problem of the MOES UFO IR Transmitter

i observe the same on my MOES - as a workaround i send some commands twice ^^

closed 2025-04-05 by github-actions[bot]

kingp0dd, 2025-10-02:
I'm experiencing this. Sometimes have to send 5 times