Koenkk/zigbee2mqtt #25987: Suddenly cannot start Zigbee2MQTT cause connection error to SMLight SLZB-06M

Sejk, 2025-01-25:
### What happened?

My HomeAssistant Zigbee2MQTT  stop working from yesterday because is not able to connect properly to SMLight SLZB-06M coordinator. I tried upgrade / downgrade firmware on SMLight SLZB-06M from 2.3.6 to 2.7.1 but without success.  

Error message from debug log "z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR." looks like network problem but I can ping IP of SMLight SLZB-06M and access its web configuration over http, so there is no problem with IP address or networking. 

In SMLight SLZB-06M log I can see, that there is any connection from my Home Assistant:

[07:22:52] taskZB | New client: 192.168.1.104 id: 0
[07:23:04] taskZB | Client disconnected, id: 0
[07:23:07] taskZB | New client: 192.168.1.104 id: 0
[07:23:21] taskZB | Client disconnected, id: 0
[07:23:24] taskZB | New client: 192.168.1.104 id: 0
[07:23:36] taskZB | Client disconnected, id: 0
[07:23:40] taskZB | New client: 192.168.1.104 id: 0

I dont know why, but MDNS stop working in my network so now I cannot use "SLZB-06M.local" and I had to change all configuration to IP:

homeassistant:
  enabled: true
advanced:
  network_key:
    - 115
    - 72
    - 158
    - 127
    - 59
    - 171
    - 128
    - 84
    - 238
    - 94
    - 76
    - 56
    - 29
    - 244
    - 240
    - 204
  pan_id: 50096
  ext_pan_id:
    - 86
    - 233
    - 83
    - 174
    - 108
    - 79
    - 84
    - 231
  ezsp_frame_timeout: 8000
  log_level: debug
  transmit_power: 20
mqtt:
  server: mqtt://core-mosquitto:1883
  user: addons
  password: somehash
frontend:
  enabled: true
  port: 8099
serial:
  port: tcp://192.168.1.103:6638
  adapter: ember
  baudrate: 115200

### Zigbee2MQTT version

Current version: 2.0.0-2 

### Adapter firmware version

v2.7.1
[... 151 lines cut ...]
[2025-01-25 07:24:55] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1100/2500
[2025-01-25 07:24:55] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1200/2500
[2025-01-25 07:24:55] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1300/2500
[2025-01-25 07:24:55] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1400/2500
[2025-01-25 07:24:56] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1500/2500
[2025-01-25 07:24:56] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1600/2500
[2025-01-25 07:24:56] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1700/2500
[2025-01-25 07:24:56] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1800/2500
[2025-01-25 07:24:56] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1900/2500
[2025-01-25 07:24:56] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2000/2500
[2025-01-25 07:24:56] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2100/2500
[2025-01-25 07:24:56] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2200/2500
[2025-01-25 07:24:56] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2300/2500
[2025-01-25 07:24:56] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2400/2500
[2025-01-25 07:24:57] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-25 07:24:57] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-25 07:24:57] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-25 07:24:57] error: 	z2m: Exiting...
[2025-01-25 07:24:57] error: 	z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
    at EmberAdapter.initEzsp (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:670:19)
    at EmberAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1533:24)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)

hugochmel, 2025-01-25:
Is your slzb-06m added as device in HA? i had same error and had to remove the device from HA. then zigbee2mqtt worked

Sejk, 2025-01-27:
Problem solved: The SLZB-06M was removed as a device from HA, the Zigbee2MQTT add-on was removed and then re-added. After that, the SLZB-06M was restarted, and now Zigbee2MQTT is working properly. The bad news is that I need to re-add all Zigbee devices to HA. Maybe it’s better to start with restarting the SLZB-06M first and only remove the add-on as a last resort. :-D

closed 2025-04-05 by github-actions[bot]