Koenkk/zigbee2mqtt #25990: I can't join new devices

qltel1, 2025-01-25:
### What happened?

New devices can't be joined. It worked fine and this morning a Sonoff SNZB-03 lost connection to Z2M. I resettet the device and nothing happened. I tried 3 others and they won't be seen by Z2M either. I tried it with a Lidl HG08673 and this device won't join also. 

All the devices in Z2M (motion detectors, switches and hue lights) works fine. I just can't join new devices or get the failing SNZB-03 working.

The not working SNZB-03 is still in the configuration, but it won't detect movement now. "Permit join (All)"  is on. All the joined devices (except this one) works fine. Joining won't work. 

### What did you expect to happen?

I expect that I can join new devices to Z2M. 

### How to reproduce it (minimal and precise)

Don't know. I did not change anything to my configuration. Joining just stopped working it seems.

### Zigbee2MQTT version

2.0.0-2

### Adapter firmware version

As far as I know I can't see this in HA. 

### Adapter

Raspbee 2

### Setup

Add-on on Home Assistant, running on a Rpi 5 with 8GB memory

### Debug log

I added the  log_level: debug to the Z2M configuration.yaml, but I can't find this data/log and see nothing special in the normal log file.

Strik3R86, 2025-01-25:
It seems that I have some pairing issues also, however i have tricked the sensors to join either by actioning them(buttons) either by removing batteries when they were in the pairing process and they showed up and configuring - during the pairing process i have received and error like - error: interview failed because can not get active endpoints ('0x847127fffe287085')

qltel1, 2025-01-25:
Resetting them or removing and replacing the batteries don't help in my case..  :(

Interview of paired devices goes without problems.

info 2025-01-25 13:35:29z2m: Interviewing 'Lamp in hal'
info 2025-01-25 13:35:30z2m: Successfully interviewed 'Lamp in hal'

salopette, 2025-01-30:
I have the same problem with 2.0

I have 72 devices that were already connected, they work, but new devices are connected and then thrown out again immediately. And this happens in a continuous loop.

`2025-01-30 18:39:43z2m: Failed to configure '0x001788010b003ff1', attempt 1 (Error: Bind 0x001788010b003ff1/1 genPowerCfg from '0x00124b001ca168ea/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Object.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/utils/waitress.ts:59:23) at func (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:423:47) at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20) at ZStackAdapter.sendZdoInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:428:43) at ZStackAdapter.sendZdo (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:305:16) at Endpoint.bind (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:562:30) at Object.bind (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/reporting.ts:42:9) at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/devices/philips.ts:2321:13) at Configure.configure (/app/lib/extension/configure.ts:124:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17))
2025-01-30 18:40:08z2m: Failed to configure '0x001788010b003ff1', attempt 2 (Error: Bind 0x001788010b003ff1/1 genPowerCfg from '0x00124b001ca168ea/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Object.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/utils/waitress.ts:59:23) at func (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:423:47) at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20) at ZStackAdapter.sendZdoInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:428:43) at ZStackAdapter.sendZdo (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:305:16) at Endpoint.bind (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:562:30) at Object.bind (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/reporting.ts:42:9) at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/devices/philips.ts:2321:13) at Configure.configure (/app/lib/extension/configure.ts:124:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17))
`

mithomas, 2025-02-02:
Same here with a [Sonoff SNZB-02D](https://www.zigbee2mqtt.io/devices/SNZB-02D.html#sonoff-snzb-02d) after battery replacement (several others work fine); with the exact same error message.

> Zigbee2MQTT version
>     [2.1.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.1.0) commit: [unknown](https://github.com/Koenkk/zigbee2mqtt/commit/unknown)
> 
> Coordinator type
>     zStack3x0
> 
> Coordinator revision
>     20210708
> 
> Coordinator IEEE Address
>     0x00124b002c3fffb1
> 
> Frontend version
>     0.9.4
> 
> zigbee-herdsman-converters version
>     21.27.1
> 
> zigbee-herdsman version
>     3.2.5

mithomas, 2025-02-04:
Possibly related: #25419, #24695 - let's see if removing/re-adding the device helps here as well.

closed 2025-04-14 by github-actions[bot]