Koenkk/zigbee2mqtt #25993: [Feature request]: get ieeeAddr in MQTT messages

vmath54, 2025-01-25:
### Is your feature request related to a problem? Please describe

When using the friendlyName, it is necessary to enable the `include_device_information` option to obtain the ieeeAddr, which is the only one that can identify a device with certainty.
The problem is that enabling this option results in a very significant increase in the size of MQTT messages sent by z2m; on average, more than double.

### Describe the solution you'd like

an option to add ieeeAddr to every MQTT message.
A bit like the `last_seen` option

### Describe alternatives you've considered

Currently I am using the `include_device_information` option; but it generates large MQTT messages.

### Additional context

example of messages for a NOUS outlet with` include_device_information` option enabled.
This option adds `device` and `update` sections, while we only need ieeeAddr
`
{
	"child_lock": "UNLOCK",
	"current": 0.32,
	"energy": 196.51,
	"indicator_mode": "off/on",
	"last_seen": "2025-01-25T13:36:55.032Z",
	"linkquality": 83,
	"power": 63,
	"power_outage_memory": "restore",
	"state": "ON",
	"voltage": 233
	"device": {
		"applicationVersion": 192,
		"dateCode": "",
		"friendlyName": "prise NOUS_01",
		"hardwareVersion": 1,
		"ieeeAddr": "0xa4c138f687bc8e5c",
		"manufacturerID": 4417,
		"manufacturerName": "_TZ3000_2putqrmw",
		"model": "A1Z",
		"networkAddress": 13650,
		"powerSource": "Mains (single phase)",
		"stackVersion": 0,
		"type": "Router",
		"zclVersion": 3
	},
	"update": {
		"installed_version": 192,
		"latest_version": 192,
		"state": "idle"
	},
}
`

LaurentChardin, 2025-02-03:
I kinda agree with this proposal, given that the IEEEADDR is usually a good primary identifier: i was always a bit puzzled with the friendly name being used as an ID.

@Koenkk what do you think about it ?

Koenkk, 2025-02-03:
>  i was always a bit puzzled with the friendly name being used as an ID.

You mean in the topics for that device?

LaurentChardin, 2025-02-03:
Yes, as the friendly name is defining the topic name when set, i can understand that MQTT consumer systems are losing the proper key information, unless `include_device_information` is activated as @vmath54 reported. 

Which i guess what the initial idea : people had no clues what device was what if all the topics are using the IEEE address. 😄 

Therefore the idea to always include the IEEE address always inside the payload per default.

LaurentChardin, 2025-02-03:
Another idea : customize the properties added to the payload when `include_device_information` is activated. Maybe a default configuration that can be tweaked on demand.

vmath54, 2025-02-04:
I agree with both proposals from @LaurentChardin

Another possibility would be to be able to choose, optionally, the topic used for the MQTT message: the friendly name, or the IEEE address

closed 2025-04-14 by github-actions[bot]