Koenkk/zigbee2mqtt #25994: Solar Rain Sensor RB-SRAIN01 Cleaning reminder always true

nmaster2042, 2025-01-25:
### What happened?

My Solar Rain Sensor RB-SRAIN01, is reporting lux, rain and battery but Cleaning reminder stays always with value true, even when I clean the surface of the sensor. 

### What did you expect to happen?

When I clean the sensor Cleaning reminder should get value of false.

### Zigbee2MQTT version

1.4.2

### Adapter firmware version

20230507

### Adapter

SONOFF Dongle P

### Setup

X86-64

fratichelo, 2025-01-25:
X2

E30M20B20, 2025-02-09:
X3

tobiasg021, 2025-02-11:
same x4

stertzi2012, 2025-02-27:
x5

10der, 2025-03-31:
ping

marcelhoogantink, 2025-04-11:
Does it change when connected to tuya? Maybe it is not a Z2M issue ...?

xlucasfelix, 2025-04-24:
Any solution for this? 👀

TEF2one, 2025-04-25:
same here, still looking for a solution...

jspiros, 2025-04-28:
Same here, just installed one.

jspiros, 2025-04-28:
Same here. I installed a brand new RB-SRAIN01, which was perfectly clean, just yesterday. It's still clean today, and it has not rained. `cleaning_reminder` has been `true` from the moment it was added to my mesh.

Though, unlike the original issue reporter, my `battery` has been consistently `null` from the start as well.

mattmoo, 2025-05-13:
After using a couple of other sensors, I suspect that this is a device issue (e.g., the sensor tracks being jumped somewhere). All the others work fine.

jspiros, 2025-05-14:
@mattmoo Sorry, maybe this is obvious but just making sure, do you mean you have instances of this exact model that didn't have the issue, and it's likely a per-device issue for those of us experiencing it? Or are you saying it's a flaw with this particular model overall?

mattmoo, 2025-05-14:
Yes, I have four. One of them is locked at 3.3V, whereas the other three behave as expected.

jspiros, 2025-05-14:
@mattmoo 3.3V? Do you mean the rain_intensity property, which has mV units? The cleaning_reminder is a boolean. Or do you know what physical feature is being used to test for it, and that's measuring 3.3V with a multimeter or something, like it's shorted?

I had been wondering if it might just be an inversion, since mine is new and it hasn't really gotten dirty, and I don't know what heuristic it's using to issue the reminder. My rain_intensity property seems to work correctly, only the cleaning_reminder is stuck on.

mattmoo, 2025-05-14:
Sorry, I see now that I am looking at a different issue, the "Rain" binary sensor being stuck on due to a short.

All four have the "Cleaning reminder" on at all times!

jonathh21, 2025-05-16:
+1 for the cleaning reminder!

marcelhoogantink, 2025-05-16:
Can someone confirm that it works correct directly with tuya, without Z2M?

jonathh21, 2025-05-17:
I have done this, and I was not presented with a clean sensor.

Hopefully you can see the screenshot of both the tuya sensor and what you then get back in HA (which is very minimal compared to direct via zb2mqtt.

I' also pressed the button on the back once, twice, few times and various combinations and it didn't reset anything.

I will switch it back to direct later but if anyone wants me to check anything before I do let me know.

I'll just disable the clean sensor.

[image]
[image]

jspiros, 2025-05-18:
@jonathh21 When I went looking for screenshots of the Tuya app with this program, I checked the reviews on AliExpress for a few listings, and I found these. I don't know how this screen is accessed, but, it seems like there's a menu somewhere labeled "select function" or similar (as you can see, neither is in English, so I'm guessing).

[image]

[image]

It would seem this is how the "Cleaning reminder" is accessed? If you still have it set up with the Tuya app, maybe you can try to find this?

jonathh21, 2025-08-03:
Ok, ok I have readded this unit from zb2mqtt to Tuya.

Immediately I can see the battery level works for tuya.

[image]
[image]

Anything to look for whilst it is on tuya?

fbloemhof, 2025-08-03:
> Ok, ok I have readded this unit from zb2mqtt to Tuya.
> 
> Immediately I can see the battery level works for tuya.
> 
> <img alt="Image" width="864" height="1920" src="https://private-user-images.githubusercontent.com/13969744/473803580-3cc683a1-97e4-4253-a893-5854e990f934.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTQyNDI4OTEsIm5iZiI6MTc1NDI0MjU5MSwicGF0aCI6Ii8xMzk2OTc0NC80NzM4MDM1ODAtM2NjNjgzYTEtOTdlNC00MjUzLWE4OTMtNTg1NGU5OTBmOTM0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA4MDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwODAzVDE3MzYzMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWRjYTQyZTdhYmZiYTJhYzI3ZjM2MmQ3ZWZlMmNmYTliNDg5ZTEyYTA4Yzg5OGJmYzVhMDg2Mjk5YjYzYTk5NzImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.ch5aENmZmnn84VG0SJJa_740gX2-tMSaf26r2QN-rWM"> <img alt="Image" width="864" height="1920" src="https://private-user-images.githubusercontent.com/13969744/473803581-0d8efddf-65c8-4d97-ab57-ff4a49e46999.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTQyNDI4OTEsIm5iZiI6MTc1NDI0MjU5MSwicGF0aCI6Ii8xMzk2OTc0NC80NzM4MDM1ODEtMGQ4ZWZkZGYtNjVjOC00ZDk3LWFiNTctZmY0YTQ5ZTQ2OTk5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA4MDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwODAzVDE3MzYzMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTY4ZWJiN2EzMTQ4YjkxZjM4NDhkYzBjZjhlNjY3ZTVlNzBiNWI2ODQ0Y2FmNjUxMjNlYWZmODhlYTNkOTQ2ZjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.Hn0Rv_rayBenMK-vI0qnQqTGeKuDmx0PSGPUpr6UZY0">
> Anything to look for whilst it is on tuya?

In Z2M in the 'Exposes' tab there is a button at the battery item you can click on, after that it started reporting the battery state for mee.

jonathh21, 2025-08-03:
> I've tapped everywhere I can, just can't see any hidden menu
>
> Is it possibly in the edit menu? (The edit icon in the upper right?)

Afraid not..

There is one or two more options just out of sight but just device
replacement/update.

Anything else to try?

It isn't that hard to switch, I'm just lazy.

-----------------------
N: Jon Hardcastle
E: ***@***.***
-----------------------

closed 2026-10-02 by github-actions[bot]

milandzuris, 2026-10-02:
.