Koenkk/zigbee2mqtt #25995: [Wrong device]: Zigbee Water Sensor SNZB-05 recognized as SNZB-03

TRIGGERhappyTV, 2025-01-25:
### Link

https://de.aliexpress.com/w/wholesale-Ewelink-Zigbee3.0-Water.html?spm=a2g0o.home.search.0

### Model

SNZB-05

### Description

Water leakage detector

### Vendor

eWeLink

### Picture (link)

https://www.zigbee2mqtt.io/devices/SNZB-05.html

### Database entry

{"id":10,"type":"EndDevice","ieeeAddr":"0xa4c138bbfaf04aba","nwkAddr":16277,"manufId":4742,"manufName":"eWeLink","powerSource":"Battery","modelId":"SNZB-03","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1,1280,32],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"hwVersion":0}},"ssIasZone":{"attributes":{"iasCieAddr":"0x6c5cb1fffe298423","zoneState":1}},"genPollCtrl":{"attributes":{"checkinInterval":14400}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":98,"batteryVoltage":25}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x6c5cb1fffe298423","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x6c5cb1fffe298423","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":7200,"repChange":2},{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":7200,"repChange":100}],"meta":{}}},"appVersion":113,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1737817812219,"checkinInterval":3600}

Koenkk, 2025-01-27:
This device seems to have the wrong modelID flashed on it, stop z2m, change `SNZB-03` to `SNZB-05` in the data/database.db and starting z2m should fix the issue.

TRIGGERhappyTV, 2025-01-27:
Many thanks! That did the trick.