Koenkk/zigbee2mqtt #25996: [Wrong device]: LoraTap SC500ZB detected as Tuya TS130F

idjul, 2025-01-25:
### Link

https://www.zigbee2mqtt.io/devices/SC500ZB.html

### Model

SC500ZB

### Description

Roller Shutter Blind Module

### Vendor

LoraTap

### Picture (link)

https://www.zigbee2mqtt.io/images/devices/SC500ZB.png

### Database entry

{"id":20,"type":"Router","ieeeAddr":"0xa4c138eeafc36bd3","nwkAddr":11138,"manufId":4417,"manufName":"_TZ3000_femsaaua","powerSource":"Mains (single phase)","modelId":"TS130F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":514,"inClusterList":[4,5,6,258,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS130F","manufacturerName":"_TZ3000_femsaaua","powerSource":1,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"tuyaBacklightSwitch":0,"tuyaBacklightMode":0}},"closuresWindowCovering":{"attributes":{"32768":0,"currentPositionLiftPercentage":0,"tuyaMovingState":1,"tuyaCalibration":1,"tuyaMotorReversal":1,"moesCalibrationTime":181}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1737822684195} {"id":21,"type":"Router","ieeeAddr":"0xa4c1386643a15261","nwkAddr":31765,"manufId":4417,"manufName":"_TZ3000_femsaaua","powerSource":"Mains (single phase)","modelId":"TS130F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":514,"inClusterList":[4,5,6,258,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS130F","manufacturerName":"_TZ3000_femsaaua","powerSource":1,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"tuyaBacklightSwitch":0,"tuyaBacklightMode":0}},"closuresWindowCovering":{"attributes":{"32768":0,"currentPositionLiftPercentage":0,"tuyaMovingState":1,"tuyaCalibration":1,"tuyaMotorReversal":1,"moesCalibrationTime":227}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1737822688692} {"id":22,"type":"Router","ieeeAddr":"0xa4c1389db7724da4","nwkAddr":20946,"manufId":4417,"manufName":"_TZ3000_femsaaua","powerSource":"Mains (single phase)","modelId":"TS130F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":514,"inClusterList":[4,5,6,258,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS130F","manufacturerName":"_TZ3000_femsaaua","powerSource":1,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"tuyaBacklightSwitch":0,"tuyaBacklightMode":0}},"closuresWindowCovering":{"attributes":{"32768":0,"currentPositionLiftPercentage":0,"tuyaMovingState":1,"tuyaCalibration":1,"tuyaMotorReversal":1,"moesCalibrationTime":237}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1737822689706}

### Notes

Working almost well, but not recognized as expected.

alekslegkih, 2025-01-26:
I join the question.  The LoraTap SC500ZB device is recognized as a Tuya TS130F.
Result: the curtains don't work.

idjul, 2025-01-31:
Any thoughts?
btw, i can't add a label on the issue

ardoviniandrea, 2025-02-07:
I have the same issue, but other than the fact the % open doens't work, everything else does (opening, closing, stop).

I'd like to have the calibration and % of blinds working, but not a big deal for now

kolmberger, 2025-02-12:
Same here. I just installed 3 of these modules. Pairing, renaming and calibration worked so far. On my end, I can control the motor though.

dmbuil, 2025-02-17:
Same here with the SC500ZB v4. Pairing and renaming work OK, but when it comes to calibration... I had to come with [this](https://github.com/Koenkk/zigbee2mqtt/issues/17250#issuecomment-2006703054) workaround.

pmfnarra, 2025-02-18:
I also have this problem with v2 and v4, but v4 apparently has 2 calibration times, one for closing and one for opening. Currently only the opening works correctly, the other I’m not able to calibrate. @dmbuil , could you explain the workaround for a beginner in zigbee and home assistant? Thanks

kolmberger, 2025-02-22:
Hey guys,

I could need some help here. I have these aliexpress Loratap modules installed and they are making me crazy: https://de.aliexpress.com/item/1005007099671541.html?spm=a2g0o.pdp_unavailable_rcmd
I don't know which version this is.
Detected in HA as TS310F, I am using Zigbee2MQTT. I can't get this thing to operate with the some standard hardware switch anymore, because at the moment the blinds are stopping in the middle of the window / door.

- How do I recognize the version of the module?
- How do I manually correct the calibration so that the blinds open and close fully at least by the switch?

Thanks for your help

midiwidi, 2025-04-06:
Same here. The SC500ZB (the darker orange version - Style B) is recognised as Tuya TS130F. Thanks to the comments here I was able to set the calibration times for up and down. Thanks guys! Would be nice to have it showing up as SC500ZB and have both calibrations available.

Does anyone know if the SC500ZB could be operated with momentary switches (push buttons)? I was hoping I could press the up button and the shutter would keep opening until it reaches the programmed time. But unfortunately that doesn't wok. Does the module maybe have an endpoint to configure the button type?

kolmberger, 2025-04-06:
Hey @midiwidi,
not sure, if I got your question right or if my answer will help in any way. In my use case, I have Gira switches (Schalter) installed instead of push buttons (Taster) on the walls right beside windows and terrace door. I noticed, that if I switch a direction twice (2x) within a second, the blinds go that direction until the endpoint of the motor is reached respectively until the calibrated time has expired.
Secondly, I have some Ikea Tradfri E1743 switches and Tradfri E1812 push buttons installed which work as controls to open/close all blinds of a story (Stockwerk) or all blinds of the house. This is of course done in an automation. For me, it was quite fiddling around, but it finally works just fine and the family loves using it :-)

midiwidi, 2025-04-06:
@kolmberger, thank you so much (tausend Dank) for sharing this observation. This does exactly what I want. The key here is timing. If the push button (Taster) is held in the ON position for too long then the motor stops when the button is released. This would be useful if one wants to open the shutter to a certain point. If the button is pushed only briefly then the shutter opens/closes all the way until the programmed time is reached. While the shutter is moving (due to the previous brief button press) it can be stopped with another quick button press in the same direction as it is travelling.
Thanks again, that really helped me.

raoulbhatia, 2025-04-19:
@Koenkk what can we, as a community, do to improve this device configuration via z2m?

Like others
1. My device is reported as Zigbee Model `TS130F`, Zigbee Manufacturer `_TZ3000_5iixzdo7`
2. I also had manually configure `endpoint 2`
3. I believe I see some drift, but perhaps my calibration was not 100% accurate

Thanks for your feedback!

Koenkk, 2025-04-21:
@raoulbhatia for point 2, what did you have to configure exactly, only the `moesCalibrationTime`?

raoulbhatia, 2025-05-18:
> [@raoulbhatia](https://github.com/raoulbhatia) for point 2, what did you have to configure exactly, only the `moesCalibrationTime`?

@Koenkk sorry for the late reply. Yes, I configured `moesCalibrationTime` for Endpoint 2
`[2025-03-30 18:03:45] info: 	zhc:tz: Wrote '{"moesCalibrationTime":462}' to 'closuresWindowCovering'`

dmbuil, 2025-06-02:
Is there any way that the separate calibration Times can be added to the LoraTap SC500ZB settings, @Koenkk?
How can we help here?

mt-mrx, 2025-06-08:
Hi,

my device is discovered as:

Zigbee Model: TS130F
Zigbee Manufacturer: _TZ3000_5iixzdo7

to add to what @raoulbhatia said he set on the device, my settings are the following. My module was set for only momentary switch action, which was pretty annoying so I played around with endpoint=1, cluster=manuSpecificTuya_3, attribute=SwitchType.
However I'm not 100% certain that the settings I "figured out" for the SwitchType are really behaving like that. There is sometimes a bit of weirdness going on, maybe there is some additional debounce behaviour encoded in the different settings.
Maybe @midiwidi can test this as well since he was looking into "push buttons".

Also I don't think the **Indicator Mode** nor **Backlight Mode** are relevant for this device since it's missing those LEDs.

# Up calibration
Select endpoint=1, cluster=Closures, attribute=moesCalibrationTime 
-> e.g. 200 = 20.5 seconds

# Down calibration
Select endpoint=2, cluster=Closures, attribute=moesCalibrationTime attribute
-> e.g. 100 = 10.0 seconds

# SwitchType change
Select endpoint=1, cluster=manuSpecificTuya_3, attribute=SwitchType
Value:
0 - latching on very short press (<500ms ?) / momentary on longer press
1 - latching on very short press / momentary on longer press
2 - always latching, action on release
3 - always latching, action on press

weichenb, 2025-06-25:
Today I performed tests with some changes. Here the result for proper visualisation and separate up and down-time calibration:

[image]

[image]

@idjul and @mt-mrx 
Is that what you expect?

weichenb, 2025-06-25:
> I only meant to literally replace the words. Since the value of "State" are "Open|Close|Stop" wouldn't it match better to replace "up" with "open" and "down" with "close".
Thanks, I see. This is common terminology in all adapters. However it would be worth discussion that with Koen.

A follow-up on "SwitchType".
Yes there is a basic definition existing in the code. However that mapping is not included anywhere (at least I could not discover).
Where have you that mapping from?
>0 - latching on very short press (<500ms ?) / momentary on longer press
>1 - latching on very short press / momentary on longer press
>2 - always latching, action on release
>3 - always latching, action on press

mt-mrx, 2025-06-25:
> Thanks, I see. This is common terminology exists for all adapters. However it would be worth discussion that with Koen.

Oh ok, if that's common for all adapters then leaving it like that makes sense as well.
 
> A follow-up on "SwitchType". Yes there is a basic definition existing in the code. However that mapping is not included anywhere (at least I could not discover). Where have you that mapping from?

I "just" tested the values and that's what I think the differences are.
I started at 0, because that was the initial reading and tried up to 5 (I think), couldn't see a change in behaviour for 4 and 5.
For 0 and 1 I couldn't really figure out a difference and I'm not even sure if the description of the behaviour is fully correct or complete for any of the values. So it probably is better to not include that in any form.

weichenb, 2025-06-26:
proof of concept works for button configuration ;-)

[image]

However, I am still not statisfied with coding. Since switchType has actually another preset, I had to use a coding that I don't like.
Therefore the fine-coding will require some more time.

weichenb, 2025-06-28:
Development is completed. I am going to submit code changes next week.
Beside **LoraTap SC500ZBv2** I implemented features also for **Nous B4Z** (calibration here valid for both directions).
Generally the changes can be activated easily for other devices. Since I have others not at hand, I restricted changes #1, #3 and #4 only to these 2 devices so far.

### Changes:
1. **Dual direction calibration**
For devices, which differentiate calibration for separate directions, configuration shows separate set for "close" and "open" directions.
2. **Calibration quick and extended**
On one hand calibration can be performed by starting "Calibration"-mode, moving shutter from one end to the other and ending "Calibration"-mode. Here resulting time is transferred automatically into "Calibration time (to open/to close)" setting(s). This was the standard procedure so far.
On the other hand from now on time(s) with 1 decimal precision (e.g. 35,1 "s") can be directly entered into field. Values are directly transferred into device registers.
3. **Configuration for physically connected electrical switch**
With the modification you can easily select the button mode that is required for your scenario.
There are typcially 2 kind of shutter switches (both have for each moving direction either a single button or at least a separate position):
**Button-switch**: Such kind of switch only transmits a signal as long as button is actually pressed. After the button-push the button returns into neutral position.
However there are 2 ways to press the button, either with a short or longer (typically >1s) push. In dependency of the push the modul generates a different behavior.
**Flip-switch**: For such kind of switch the button remains after the push in its position and does not return into neutral position.
**Sync-switch**: On Nous B4Z switch this mode allows a more precise angle control of jalousies. The remaining behavior I would rate as identical with "Flip-switch"-mode.
LoraTap SC500ZBv2 seems not to show a difference compared to "Button-switch"-mode.
**Button2-switch**: On LoraTap SC500ZBv2 a long push seems to show a slightly different behavior.
Nous B4Z seems not to show a difference compared to "Button-switch"-mode.
I contacted both manufactures to provide a bit more information about the supported features. I am going to add more information later in the manual.
4. **Backlight and Indicator LED options **
Both devices do not support that feature. Therefore I deactivated the option for both models.
5. **Code homogenization and simplification in devices/tuya.ts for generic model "TS130F"**

Open issue:
One topic that was mentioned in this thread is not solved yet, it is the leaking graphical position update. I observed that it typically works reliably. However especially after configuration changes (mostly calibrations), the prosition update freezes. I think that his issue is not related to the device support module, but to some subordinate code. When refreshing the page, updates continue.
Therefore I have not spent more energy in analysing the issue.

**Nous B4Z**
[image]

**LoraTap SC500ZBv2**
[image]

mt-mrx, 2025-06-28:
Looks very good! Thanks a lot for putting in the effort!

weichenb, 2025-06-28:
> Looks very good! Thanks a lot for putting in the effort!

Thanks for your honors, @mt-mrx . 
Btw. my favorite switch will be the Nous B4Z. 3 reasons: 
- LoraTap does not fit in most of my switch-boxes (if too many cables the switch-box, the LoraTap cannot be moved in, although switch-box has sufficient depth. 
- The B4Z is much smaller and there is no problem in any of my switch boxes to integrate it.
- B4Z switches more accurate, especially for my jalousies.
- From my feeling radio transmission of B4Z seems to be stronger. For LoraTap I had during tests at the same position often connections losses. I had to move it much closer to the coordinator.

weichenb, 2025-06-30:
Hi All and @kolmberger,
changes have been transferred into master. With next docker image update, you should be able to use changes directly. Options should be also directly available in Homeasissistant. 

@idjul: I think, you can close this issue, right?

weichenb, 2025-07-02:
> With next docker image update, you should be able to use changes directly.

Since yesterday new docker release with referred update is available. Happy testing and using!

kolmberger, 2025-07-04:
I do confirm, after the Z2M update, the new configuration options are there. Very nice, thanks again for the effort.

Is there a change to exchange the device picture in Z2M, as I still have something like this:

[image]

weichenb, 2025-07-04:
> Is that the recommended way to do it?

Actually not.
In my docker installation new picture appeared automatically after update.
Could you try to remove and add the device again?
It would be interesting to see, what happens then.

idjul, 2025-07-04:
Issue closed, thank you everyone :)