Koenkk/zigbee2mqtt #26001: [New device support]: Tuya 8 gang smart switch (NovaDigital with Tuya Whitelabel)

eliezerfb, 2025-01-25:
### Link

https://www.novadigitalsmart.com.br/produtos/interruptor-touch-zigbee-8-botoes-ws-us8-w-b

### Database entry

{"id":30,"type":"Router","ieeeAddr":"0xa4c13852fe150f67","nwkAddr":12487,"manufId":4417,"manufName":"_TZE204_nvxorhcj","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZE204_nvxorhcj","zclVersion":3,"appVersion":74,"modelId":"TS0601","powerSource":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":217189513},"lastSeen":1737832349181}

### Zigbee2MQTT version

2.0.0

### Comments

const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE204_nvxorhcj']),
    model: 'TS0601_switch_8',
    vendor: 'Tuya',
    description: 'ZYXH 8 gang switch',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
        tuya.exposes.switch().withEndpoint('l1'),
        tuya.exposes.switch().withEndpoint('l2'),
        tuya.exposes.switch().withEndpoint('l3'),
        tuya.exposes.switch().withEndpoint('l4'),
        tuya.exposes.switch().withEndpoint('l5'),
        tuya.exposes.switch().withEndpoint('l6'),
        tuya.exposes.switch().withEndpoint('l7'),
        tuya.exposes.switch().withEndpoint('l8'),
    ],
    endpoint: (device) => {
        return {l1: 1, l2: 1, l3: 1, l4: 1, l5: 1, l6: 1, l7: 1, l8: 1};
    },
    meta: {
        multiEndpoint: true,
        tuyaDatapoints: [
            [1, 'state_l1', tuya.valueConverter.onOff],
            [2, 'state_l2', tuya.valueConverter.onOff],
            [3, 'state_l3', tuya.valueConverter.onOff],
            [4, 'state_l4', tuya.valueConverter.onOff],
            [5, 'state_l5', tuya.valueConverter.onOff],
            [6, 'state_l6', tuya.valueConverter.onOff],
            [0x65, 'state_l7', tuya.valueConverter.onOff],
            [0x66, 'state_l8', tuya.valueConverter.onOff],
        ],
    }, 
};

module.exports = definition;

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_nvxorhcj',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

I tried to configure this device with a similar device.
Partly it works, but I need send 2x the command to turn on or turn off.

gui28347, 2025-02-17:
@eliezerfb I have same device and was able to use the following template, try and see if it works for you.

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
//const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_nvxorhcj',
        },
    ],
    model: 'TS0601_8_way_switch',
    vendor: 'TuYa',
    description: '8 way switch',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        // Here you should put all functionality that your device exposes
        ...Array.from({length: 8}, (_, i) => tuya.exposes.switch().withEndpoint(`l${i + 1}`))
    ],
    endpoint: (device) => {
        return {'l1': 1, 'l2': 1, 'l3': 1, 'l4': 1, 'l5': 1, 'l6': 1, 'l7': 1, 'l8': 1};
    },
    meta: {
        // All datapoints go in here
        multiEndpoint: true,
        tuyaDatapoints: [
            [1, 'state_l1', tuya.valueConverter.onOff],
            [2, 'state_l2', tuya.valueConverter.onOff],
            [3, 'state_l3', tuya.valueConverter.onOff],
            [4, 'state_l4', tuya.valueConverter.onOff],
            [5, 'state_l5', tuya.valueConverter.onOff],
            [6, 'state_l6', tuya.valueConverter.onOff],
            [0x65, 'state_l7', tuya.valueConverter.onOff],
            [0x66, 'state_l8', tuya.valueConverter.onOff],
            [0x67, 'null', null],
            [0x68, 'null', null],
            [0x69, 'null', null],
            [0x6A, 'null', null],
            [0x6B, 'null', null],
            [0x6C, 'null', null],
            [0x6D, 'null', null],
           [0x6E, 'null', null],
            [0xB6, 'null', null],           //Datapoint 182?
            [0xB7, 'null', null],           //Datapoint 183?
            [0xB8, 'null', null],           //Datapoint 184?
            [0xB9, 'null', null],           //Datapoint 185?
            [0xBA, 'null', null],           //Datapoint 186?
            [0xBB, 'null', null],           //Datapoint 187?
            [0xBC, 'null', null],           //Datapoint 188?
            [0xBD, 'null', null],           //Datapoint 189?
        ],
    },
};

module.exports = definition;

eduseiden, 2025-03-28:
HI all.

I am facing a problem with this device. Even in the latest Z2M version, it is not recognized. Using external converter, it works but I need to control backlight. And I can't manage it.
As an alternative, I adopted through zigbee gateway in tuya and i could turn backlight off.
Does anyone know how to add this control in external converter?

Thanks !

closed 2025-06-09 by github-actions[bot]