Koenkk/zigbee2mqtt #26002: Watchdog unable to restart Z2M tries  to fast?

jeroenvdd, 2025-01-25:
### What happened?

In my logs I see that my adapter is ofline:
```
[2025-01-25 17:48:53] error: 	zh:zstack:znp: Socket error Error: read ETIMEDOUT
[2025-01-25 17:48:53] info: 	zh:zstack:znp: Port closed
[2025-01-25 17:48:53] info: 	zh:zstack:znp: closing
[2025-01-25 17:48:53] error: 	z2m: Adapter disconnected, stopping

```
(don't know if the first notice is also about the adapter)
2 seconds later IC this ;
```
[2025-01-25 17:48:53] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2025-01-25 17:48:53] info: 	z2m: Disconnecting from MQTT server
[2025-01-25 17:48:53] info: 	z2m: Stopping zigbee-herdsman...
[2025-01-25 17:48:53] info: 	z2m: Stopped zigbee-herdsman
[2025-01-25 17:48:53] info: 	z2m: Stopped Zigbee2MQTT
[17:48:53] INFO: Preparing to start...
[17:48:53] INFO: Socat not enabled
[17:48:54] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-25 17:48:55] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-25 17:48:55] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-25 17:48:55] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-25 17:48:55] info: 	zh:zstack:znp: Opening TCP socket with slzb-06.local:6638
[2025-01-25 17:48:55] error: 	zh:zstack:znp: Socket error AggregateError
[2025-01-25 17:48:55] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-25 17:48:55] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-25 17:48:55] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-25 17:48:55] error: 	z2m: Exiting...
[2025-01-25 17:48:55] error: 	z2m: Error: Error while opening socket
    at Socket.<anonymous> (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:158:24)
    at Socket.emit (node:events:530:35)
    at emitErrorNT (node:internal/streams/destroy:170:8)
    at emitErrorCloseNT (node:internal/streams/destroy:129:3)
    at processTicksAndRejections (node:internal/process/task_queues:90:21)
/app/node_modules/.pnpm/readable-stream@3.6.2/node_modules/readable-stream/lib/_stream_writable.js:264
  var er = new ERR_STREAM_WRITE_AFTER_END();
           ^
Error: write after end
    at writeAfterEnd (/app/node_modules/.pnpm/readable-stream@3.6.2/node_modules/readable-stream/lib/_stream_writable.js:264:12)
    at DerivedLogger.Writable.write (/app/node_modules/.pnpm/readable-stream@3.6.2/node_modules/readable-stream/lib/_stream_writable.js:300:21)
    at DerivedLogger.log (/app/node_modules/.pnpm/winston@3.17.0/node_modules/winston/lib/winston/logger.js:231:12)
    at Logger.log (/app/lib/util/logger.ts:198:25)
    at Logger.info (/app/lib/util/logger.ts:211:14)
    at Znp.onPortClose (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:88:16)
    at Object.onceWrapper (node:events:633:26)
    at Socket.emit (node:events:518:28)
    at TCP.<anonymous> (node:net:343:12)
```

### What did you expect to happen?

Z2M to just restart when it has lost connection?

or is their any way to make Z2m restart, is the functionality I am describing by design or is something (other than the dc) going wrong?

if it's by design that Z2M just goes down and watchdog doesn't intervene, I should probably make an automation / script that checks for HB every 10/15 mins?

### How to reproduce it (minimal and precise)

happens randomly 

### Zigbee2MQTT version

2.0.0-2

### Adapter firmware version

20240710

### Adapter

slzb 06m

### Setup

add-on in HA

Koenkk, 2025-02-02:
This should be fixed with z2m 2.1.0

socram70, 2025-02-05:
I'm using 2.1.0-1 and the issue still persists. It doesn't work after the first retry, but only after the second one (1 minute). I also noticed that the built-in watchdog of the addon interferes with the Home Assistant one and leads to a reboot loop, so I had to turn the Home Assistant one off. My device is a SLZB-06 (via LAN cable, firmware 2.7.1). I checked the uptime of the device and there hasn't been a reboot. Here's the adapter settings and the log:

serial:
  port: tcp://192.168.25.163:6638
  baudrate: 115200
  adapter: zstack
  disable_led: false
advanced:
  transmit_power: 20`

`[2025-02-05 06:31:32] error: 	zh:zstack:znp: Socket error Error: read ETIMEDOUT
[2025-02-05 06:31:32] info: 	zh:zstack:znp: Port closed
[2025-02-05 06:31:32] info: 	zh:zstack:znp: closing
[2025-02-05 06:31:32] error: 	z2m: Adapter disconnected, stopping
[2025-02-05 06:31:32] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2025-02-05 06:31:32] info: 	z2m: Disconnecting from MQTT server
[2025-02-05 06:31:32] info: 	z2m: Stopping zigbee-herdsman...
[2025-02-05 06:31:32] info: 	z2m: Stopped zigbee-herdsman
[2025-02-05 06:31:32] info: 	z2m: Stopped Zigbee2MQTT
WATCHDOG: Waiting 0.03333333333333333min before next start try.
Starting Zigbee2MQTT with watchdog (2000,60000,300000,900000,1800000,3600000).
[2025-02-05 06:31:34] info: 	z2m: Logging to console, file (filename: log.log)
[2025-02-05 06:31:34] info: 	z2m: Starting Zigbee2MQTT version 2.1.0 (commit #unknown)
[2025-02-05 06:31:34] info: 	z2m: Starting zigbee-herdsman (3.2.5)
[2025-02-05 06:31:34] info: 	zh:zstack:znp: Opening TCP socket with 192.168.25.163:6638
[2025-02-05 06:31:36] info: 	zh:zstack:znp: Socket connected
[2025-02-05 06:31:36] info: 	zh:zstack:znp: Socket ready
[2025-02-05 06:31:36] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2025-02-05 06:31:36] error: 	zh:zstack:znp: Socket error Error: read ECONNRESET
[2025-02-05 06:31:36] error: 	z2m: Error while starting zigbee-herdsman
[2025-02-05 06:31:36] error: 	z2m: Failed to start zigbee-herdsman
[2025-02-05 06:31:36] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-02-05 06:31:36] error: 	z2m: Exiting...
[2025-02-05 06:31:36] error: 	z2m: Error: Error while opening socket
    at Socket.<anonymous> (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:158:24)
    at Socket.emit (node:events:530:35)
    at emitErrorNT (node:internal/streams/destroy:170:8)
    at emitErrorCloseNT (node:internal/streams/destroy:129:3)
    at processTicksAndRejections (node:internal/process/task_queues:90:21)
[2025-02-05 06:31:36] info: 	zh:zstack:znp: Port closed
WATCHDOG: Waiting 1min before next start try.
[2025-02-05 06:31:37] info: 	zh:zstack:znp: Skip`

bouwew, 2025-02-05:
I'm seeing something similar now:
```
[2025-02-05 12:30:12] error: 	zh:zstack:znp: Socket error Error: read ECONNRESET
[2025-02-05 12:30:12] info: 	zh:zstack:znp: Port closed
[2025-02-05 12:30:12] info: 	zh:zstack:znp: closing
[2025-02-05 12:30:12] error: 	z2m: Adapter disconnected, stopping
[2025-02-05 12:30:12] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2025-02-05 12:30:12] info: 	z2m: Disconnecting from MQTT server
[2025-02-05 12:30:12] info: 	z2m: Stopping zigbee-herdsman...
[2025-02-05 12:30:12] info: 	z2m: Stopped zigbee-herdsman
[2025-02-05 12:30:12] info: 	z2m: Stopped Zigbee2MQTT
```

socram70, 2025-02-06:
@bouwew I assume that you switched watchdog in the settings off, but configured watchdog: default (or some custom values) in the settings? If not, then this looks like your watchdog didn't start. If it's set to default, you should find the following in your logs:

`Starting Zigbee2MQTT with watchdog (2000,60000,300000,900000,1800000,3600000).`

bouwew, 2025-02-06:
@socram70 I was not aware of this, now I am, thanks!

closed 2025-04-15 by github-actions[bot]