Koenkk/zigbee2mqtt #26006: After update from 1.42 to 2.0, I lose "ACTION" entity of my Aqara/Xiamoi button

nappyjim, 2025-01-26:
### What happened?

After updating from 1.42 to 2.0, I lose my action entity of ALL (qty 5) Aqara buttons

### What did you expect to happen?

I have the same entities as before update

### How to reproduce it (minimal and precise)

Update from 1.42 to 2.0 and then check your entities.  

Below is same search results when running 1.42 vs 2.0.  Notice my action entity is gone in 2nd.

[image]

[image]

### Zigbee2MQTT version

2.0.0-2

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0

### Setup

Add on Home Assistant running on a Dell PC

nappyjim, 2025-01-26:
I tried doing the recommended changes to my configuration.yaml specified in the 2.0.0-1 release, it did not make a difference.

erdeidominik99, 2025-01-26:
I also don't have the action on any tuya buttons.

lux73, 2025-01-26:
https://github.com/Koenkk/zigbee2mqtt/discussions/24198

nappyjim, 2025-01-26:
I just read the 2.0.0-1 release closer and I believe my issue is in there, so I guess my question now is WHY?  The main purpose of a button is to perform an action.  Why do I have to do a work around for that now?  Why is the battery level the only entity exposed to HA?  I guess i just don't understand the logic behind this.

[image]

manuelkue, 2025-01-26:
> I tried doing the recommended changes to my configuration.yaml specified in the 2.0.0-1 release, it did not make a difference.

Have you added it to the correct configuration.yaml in the zigbee2mqtt sub folder? (not homeassistant's configuration.yaml)

nappyjim, 2025-01-26:
breaking change

sand3rk, 2025-01-27:
> Totally agree - what is a logic of (by default) removing action entities from BUTTONS / LIGHT SWITCHES ?!
> 
> Last Edit; same happens to “TRADFRI shortcut button (E1812)“ which anymore have action to press - anything else but not this even with mentioned above change in configuration.yaml inside zigbee2mqtt folder
Agree! I had this happen to me aswell on TRADFRI buttons and Wireless mini switch (WXKG11LM) by Aqara!

closed 2025-01-28 by Koenkk