Koenkk/zigbee2mqtt #26007: Devices randomly become unsupported by Z2M

DaveTSG, 2025-01-26:
### What happened?

Devices randomly becoming unsupported in Zigbee2MQTT Home Assistant add-on.

[image]

The first two items in this list, I'm guessing, are my [Mercator Ikuu Zigbee double two way wall switches](https://www.ikuu.com.au/product/double-two-way-switch-2/) that controls my garage lighting, which I have suddenly lost control of in Home Assistant.

I have had other devices do this same thing where they'll work fine for months, then suddenly become unsupported for no apparent reason. Unfortunately, when this happens, as you can see, the name I previously assigned the device gets replaced with its IEEE address, making it impossible to definitively identify which device it is when the error occurs.

I first noticed this problem with cheapy Tuya Zigbee devices (temperature sensors, motion sensors, window sensors) about six months ago. I assumed it was just because it was cheap crap. Now it has happened today with my pricey Mercator wall switches too, so I'm wondering if maybe it's Z2M?

I think I have had about a dozen devices do this over the last six months.

I have solved the issue previously by force-removing a device and then re-pairing it with Z2M.

Is there a way of preventing this from happening in future?

Thanks in advance.
Dave
P.S.: [this](https://tubeszb.com/product/cc2652p2-based-zigbee-to-poe-coordinator-2023/) is my Zigbee coordinator.

### What did you expect to happen?

For devices to continue operating for years.

### How to reproduce it (minimal and precise)

Pair and configure device as normal. Wait several months.

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20221226

### Adapter

TubesZB CC2652P2 Based Zigbee to PoE Coordinator 2023

### Setup

Home Assistant add-on, running on a Home Assistant Yellow.

### Debug log

I had debug logging enabled all along. For some reason, though, only logs for some days exist, as per the below screenshot. 

[image]

In any case, here are the three log files from today, 26th January 2025:

[log.log](https://github.com/user-attachments/files/18548892/log.log)
[log1.log](https://github.com/user-attachments/files/18548893/log1.log)
[log2.log](https://github.com/user-attachments/files/18548894/log2.log)

xs400dohc, 2025-02-03:
I have the same / similar problem:
I have a Din rail switch with power monitoring working fine for a long time. Now I recognized, it's no longer supported. It shows the original device-name:
[image]
The device was working, so I integrated in some helper and everthing was working. But now the data from device are not available anymore so the helper switched to unavailable.

xs400dohc, 2025-02-03:
My FAULT!!!!
Should have read the breaking changes for update to 2.0.0.
My device was defined as external converter and the external converters now are stored in /root/homeassistant/zigbee2mqtt/external_converters/
After moving my js-file to the new location it works again.

Koenkk, 2025-02-03:
@xs400dohc please consider making a pr for out-of-the-box support: Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

xs400dohc, 2025-02-04:
@Koenkk thanks for asking. Thought about. The sensors are working, but the thresholds aren't. The device is inproductive use, so I can't investigate anymore (my son will kill me, if I cut off the power to his room and he can't play anymore). I got the code for the external converter from issue #16118. Will publicate my (so far) working external converter there (have a look there). As far as I can see, they are still working on a solution.

Koenkk, 2025-02-04:
@xs400dohc a partial implementation is better than none :)

DaveTSG, 2025-04-06:
Bump. Still having this issue.

closed 2025-08-15 by github-actions[bot]