Koenkk/zigbee2mqtt #26008: ECONNRESET error with SLZB-06M

michelcourbon, 2025-01-26:
### What happened?

hello, 
**context**
I have a SLZB-06M and zigbee2mqtt installed in home assistant : version Core 2025.1.4 / Supervisor 2024.12.3 / Operating System 14.1
the SLZB-06M web interface is recognized in home-Assistant and work perfectly, version installed is for radio 20231030	and core v2.7.1
**error for zigbee2mqtt module**
- I try the configuration proposed by the webpage of the SZLB-06M and the eszb adapter and obtain in zigbee2mqtt journal the error
  Starting Zigbee2MQTT without watchdog.
[2025-01-26 09:06:24] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-26 09:06:24] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-26 09:06:24] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-26 09:06:25] warning: 	zh:ezsp: 'ezsp' driver is deprecated and will only remain to provide support for older firmware (pre 7.4.x). Migration to 'ember' is recommended. If using Zigbee2MQTT see https://github.com/Koenkk/zigbee2mqtt/discussions/21462
[2025-01-26 09:06:25] error: 	zh:ezsp:uart: Port error: Error: read ECONNRESET
Error: Failure to connect
    at SerialDriver.resetForReconnect (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:341:19)
    at SerialDriver.emit (node:events:518:28)
    at SerialDriver.onPortClose (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:385:18)
    at Object.onceWrapper (node:events:633:26)
    at Socket.emit (node:events:518:28)
    at TCP.<anonymous> (node:net:343:12)
- I try with the proposed adapter "ember" and I have another error 
[09:07:22] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-26 09:07:26] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-26 09:07:26] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-26 09:07:26] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-26 09:07:26] info: 	zh:ember: Using default stack config.
[2025-01-26 09:07:26] info: 	zh:ember: ======== Ember Adapter Starting ========
[2025-01-26 09:07:26] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2025-01-26 09:07:26] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-26 09:07:26] info: 	zh:ember:uart:ash: Socket ready
[2025-01-26 09:07:26] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-01-26 09:07:27] error: 	zh:ember:uart:ash: Port Error: read ECONNRESET
[2025-01-26 09:07:27] info: 	zh:ember:uart:ash: Port closed.
[2025-01-26 09:07:29] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-26 09:07:29] info: 	zh:ember:uart:ash: Socket ready
[2025-01-26 09:07:29] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-01-26 09:07:30] error: 	zh:ember:uart:ash: Port Error: read ECONNRESET
[2025-01-26 09:07:30] info: 	zh:ember:uart:ash: Port closed.
[2025-01-26 09:07:31] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-26 09:07:31] info: 	zh:ember:uart:ash: Socket ready
[2025-01-26 09:07:31] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-01-26 09:07:33] error: 	zh:ember:uart:ash: Port Error: read ECONNRESET
[2025-01-26 09:07:33] info: 	zh:ember:uart:ash: Port closed.
[2025-01-26 09:07:34] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-26 09:07:34] info: 	zh:ember:uart:ash: Socket ready
[2025-01-26 09:07:34] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-01-26 09:07:34] error: 	zh:ember:uart:ash: Port Error: read ECONNRESET
[2025-01-26 09:07:34] info: 	zh:ember:uart:ash: Port closed.
[2025-01-26 09:07:36] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-26 09:07:36] info: 	zh:ember:uart:ash: Socket ready
[2025-01-26 09:07:36] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-01-26 09:07:37] error: 	zh:ember:uart:ash: Port Error: read ECONNRESET
[2025-01-26 09:07:37] info: 	zh:ember:uart:ash: Port closed.
[2025-01-26 09:07:39] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-26 09:07:39] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-26 09:07:39] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-26 09:07:39] error: 	z2m: Exiting...
[2025-01-26 09:07:39] error: 	z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
    at EmberAdapter.initEzsp (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:670:19)
    at EmberAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1533:24)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)

thanks a lot to explain this errors !!  

### Zigbee2MQTT version

2.0.0-2

### Adapter firmware version

20231030

### Adapter

ember

### Setup

NAS Synology, VM home-assistant in same network 192.168.1.xxx, zigbee2mqtt add-on ver 2.0.0-2

### Debug log

more information in context that in log files

TBergner, 2025-01-26:
Hello, I have the same error.

[18:57:22] INFO: Preparing to start...
[18:57:22] INFO: Socat not enabled
[18:57:23] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-26 18:57:24] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-26 18:57:24] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-26 18:57:24] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-26 18:57:24] info: 	zh:ember: Using default stack config.
[2025-01-26 18:57:24] info: 	zh:ember: ======== Ember Adapter Starting ========
[2025-01-26 18:57:24] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2025-01-26 18:57:24] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-26 18:57:24] info: 	zh:ember:uart:ash: Socket ready
[2025-01-26 18:57:24] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-01-26 18:57:26] error: 	zh:ember:uart:ash: Port Error: read ECONNRESET
[2025-01-26 18:57:26] info: 	zh:ember:uart:ash: Port closed.
[2025-01-26 18:57:27] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-26 18:57:27] info: 	zh:ember:uart:ash: Socket ready
[2025-01-26 18:57:27] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-01-26 18:57:27] error: 	zh:ember:uart:ash: Port Error: read ECONNRESET
[2025-01-26 18:57:27] info: 	zh:ember:uart:ash: Port closed.
[2025-01-26 18:57:29] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-26 18:57:29] info: 	zh:ember:uart:ash: Socket ready
[2025-01-26 18:57:29] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-01-26 18:57:30] error: 	zh:ember:uart:ash: Port Error: read ECONNRESET
[2025-01-26 18:57:30] info: 	zh:ember:uart:ash: Port closed.
[2025-01-26 18:57:32] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-26 18:57:32] info: 	zh:ember:uart:ash: Socket ready
[2025-01-26 18:57:32] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-01-26 18:57:33] error: 	zh:ember:uart:ash: Port Error: read ECONNRESET
[2025-01-26 18:57:33] info: 	zh:ember:uart:ash: Port closed.
[2025-01-26 18:57:34] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-26 18:57:34] info: 	zh:ember:uart:ash: Socket ready
[2025-01-26 18:57:34] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-01-26 18:57:35] error: 	zh:ember:uart:ash: Port Error: read ECONNRESET
[2025-01-26 18:57:35] info: 	zh:ember:uart:ash: Port closed.
[2025-01-26 18:57:37] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-26 18:57:37] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-26 18:57:37] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-26 18:57:37] error: 	z2m: Exiting...
[2025-01-26 18:57:37] error: 	z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
    at EmberAdapter.initEzsp (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:670:19)
    at EmberAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1533:24)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)

??????????????????

nmavor, 2025-01-27:
same 

I have a SLZB-06M (POE) and zigbee2mqtt installed in home assistant : version Core 2025.1.4 / Supervisor 2024.12.3 / 
the SLZB-06M web interface is recognized in home-Assistant and work perfectly, version installed is for radio 20231030 and core v2.7.1

Starting Zigbee2MQTT without watchdog.
[2025-01-21 18:11:42] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-21 18:11:42] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-21 18:11:42] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-21 18:11:42] info: 	zh:ember: Using default stack config.
[2025-01-21 18:11:42] info: 	zh:ember: ======== Ember Adapter Starting ========
[2025-01-21 18:11:42] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2025-01-21 18:11:42] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-21 18:11:42] info: 	zh:ember:uart:ash: Socket ready
[2025-01-21 18:11:42] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-01-21 18:11:45] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-21 18:11:45] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-01-21 18:11:47] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-21 18:11:47] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-01-21 18:11:50] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-21 18:11:50] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-01-21 18:11:52] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-21 18:11:52] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-01-21 18:11:55] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-21 18:11:55] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-21 18:11:55] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-21 18:11:55] error: 	z2m: Exiting...
[2025-01-21 18:11:55] error: 	z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
    at EmberAdapter.initEzsp (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:670:19)
    at EmberAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1533:24)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)
[18:16:57] INFO: Preparing to start...
[18:16:57] INFO: Socat not enabled
[18:16:57] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.

TBergner, 2025-01-27:
The problem is solved for me.

I still had Zigbee activated, but I have now deactivated it in the home assistant. It has been working since then.

cyayon, 2025-03-04:
Hi,
Same issue here. I had to restart periodically Z2M in HA (latest version 2.3.x included).

```
[2025-03-03 23:49:04] [31merror[39m: 	zh:ember:uart:ash: Port Error: read ECONNRESET
[2025-03-03 23:49:04] [31merror[39m: 	zh:ember: Adapter fatal error: ERROR_SERIAL_INIT
[2025-03-03 23:49:04] [32minfo[39m: 	zh:ember:uart:ash: ASH COUNTERS since last clear:
[2025-03-03 23:49:04] [32minfo[39m: 	zh:ember:uart:ash:   Total frames: RX=1556, TX=1896
[2025-03-03 23:49:04] [32minfo[39m: 	zh:ember:uart:ash:   Cancelled   : RX=0, TX=0
[2025-03-03 23:49:04] [32minfo[39m: 	zh:ember:uart:ash:   DATA frames : RX=1556, TX=340
[2025-03-03 23:49:04] [32minfo[39m: 	zh:ember:uart:ash:   DATA bytes  : RX=35711, TX=8903
[2025-03-03 23:49:04] [32minfo[39m: 	zh:ember:uart:ash:   Retry frames: RX=0, TX=0
[2025-03-03 23:49:04] [32minfo[39m: 	zh:ember:uart:ash:   ACK frames  : RX=0, TX=1556
[2025-03-03 23:49:04] [32minfo[39m: 	zh:ember:uart:ash:   NAK frames  : RX=0, TX=0
[2025-03-03 23:49:04] [32minfo[39m: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2025-03-03 23:49:04] [32minfo[39m: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2025-03-03 23:49:04] [32minfo[39m: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2025-03-03 23:49:04] [32minfo[39m: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2025-03-03 23:49:04] [32minfo[39m: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2025-03-03 23:49:04] [32minfo[39m: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2025-03-03 23:49:04] [32minfo[39m: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2025-03-03 23:49:04] [32minfo[39m: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2025-03-03 23:49:04] [32minfo[39m: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2025-03-03 23:49:04] [32minfo[39m: 	zh:ember:uart:ash:   Retry dupes     : RX=0
[2025-03-03 23:49:04] [32minfo[39m: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2025-03-03 23:49:04] [32minfo[39m: 	zh:ember:uart:ash:   ACK timeouts    : RX=0
[2025-03-03 23:49:04] [32minfo[39m: 	zh:ember:uart:ash: ======== ASH stopped ========
[2025-03-03 23:49:04] [32minfo[39m: 	zh:ember:ezsp: ======== EZSP stopped ========
[2025-03-03 23:49:04] [32minfo[39m: 	zh:ember: ======== Ember Adapter Stopped ========
[2025-03-03 23:49:04] [31merror[39m: 	z2m: Adapter disconnected, stopping
[2025-03-03 23:49:04] [32minfo[39m: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2025-03-03 23:49:04] [32minfo[39m: 	z2m: Disconnecting from MQTT server
[2025-03-03 23:49:04] [32minfo[39m: 	zh:ember:uart:ash: Port closed.
[2025-03-03 23:49:04] [32minfo[39m: 	z2m: Stopping zigbee-herdsman...
[2025-03-03 23:49:04] [32minfo[39m: 	z2m: Stopped zigbee-herdsman
[2025-03-03 23:49:04] [32minfo[39m: 	z2m: Stopped Zigbee2MQTT
```

akira215, 2025-03-31:
Hi, 

I faced the same issue, I pointed out that it appear when Home Assistant supervisor is updating, just like if when supervisor restart, it does not properly close the connection to the SLZB-06M:
```
Mar 27 15:56:00 homeassistant hassio_supervisor[636]: 2025-03-27 16:56:00.299 INFO (MainThread) [supervisor.supervisor] jou 2025.03.4.dev2703
Mar 27 15:56:00 homeassistant hassio_supervisor[636]: 2025-03-27 16:56:00.300 INFO (MainThread) [supervisor.docker.interface] Downloading docker image ghcr.io/home-assistant/aarch64-hassio-supervisor with tag 2025.03.4.dev2703.
Mar 27 15:56:07 homeassistant hassio_supervisor[636]: 2025-03-27 16:56:07.812 INFO (MainThread) [supervisor.misc.scheduler] Shutting down scheduled tasks
Mar 27 15:56:07 homeassistant hassio_supervisor[636]: 2025-03-27 16:56:07.813 INFO (MainThread) [supervisor.docker.monitor] Stopped docker events monitor
Mar 27 15:56:17 homeassistant hassio_supervisor[636]: 2025-03-27 16:56:17.814 WARNING (MainThread) [supervisor.core] Stage 1: Force Shutdown!
Mar 27 15:56:17 homeassistant hassio_supervisor[636]: 2025-03-27 16:56:17.815 INFO (MainThread) [supervisor.hardware.monitor] Stopped Supervisor hardware monitor
Mar 27 15:56:17 homeassistant hassio_supervisor[636]: 2025-03-27 16:56:17.819 INFO (MainThread) [supervisor.dbus.manager] Closed conection to system D-Bus.
Mar 27 15:56:17 homeassistant hassio_supervisor[636]: 2025-03-27 16:56:17.820 WARNING (MainThread) [supervisor.homeassistant.websocket] Connection is closed
Mar 27 15:56:17 homeassistant hassio_supervisor[636]: 2025-03-27 16:56:17.821 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API for 491eb00d_hamh closed
Mar 27 15:56:17 homeassistant hassio_supervisor[636]: 2025-03-27 16:56:17.822 INFO (MainThread) [supervisor.api] Stopping API on 172.30.32.2
Mar 27 15:56:17 homeassistant hassio_supervisor[636]: 2025-03-27 16:56:17.834 INFO (MainThread) [supervisor.core] Supervisor is down - 0
Mar 27 15:56:17 homeassistant hassio_supervisor[636]: 2025-03-27 16:56:17.835 INFO (MainThread) [__main__] Closing Supervisor
Mar 27 15:56:18 homeassistant hassio_supervisor[636]: [15:56:18] WARNING: Halt Supervisor
Mar 27 15:56:18 homeassistant hassio_supervisor[636]: [15:56:18] INFO: Supervisor restart after closing
Mar 27 15:56:18 homeassistant hassio_supervisor[636]: s6-rc: info: service legacy-services: stopping
Mar 27 15:56:18 homeassistant hassio_supervisor[636]: [15:56:18] INFO: Watchdog restart after closing
Mar 27 15:56:18 homeassistant hassio_supervisor[636]: s6-rc: info: service legacy-services successfully stopped
Mar 27 15:56:18 homeassistant hassio_supervisor[636]: s6-rc: info: service legacy-cont-init: stopping
Mar 27 15:56:18 homeassistant hassio_supervisor[636]: s6-rc: info: service legacy-cont-init successfully stopped
Mar 27 15:56:18 homeassistant hassio_supervisor[636]: s6-rc: info: service fix-attrs: stopping
Mar 27 15:56:18 homeassistant hassio_supervisor[636]: s6-rc: info: service fix-attrs successfully stopped
Mar 27 15:56:18 homeassistant hassio_supervisor[636]: s6-rc: info: service s6rc-oneshot-runner: stopping
Mar 27 15:56:18 homeassistant hassio_supervisor[636]: s6-rc: info: service s6rc-oneshot-runner successfully stopped
Mar 27 15:56:21 homeassistant systemd[1]: docker-8e16a9f7fe3e1c278777c449065990a58c564beaae07da3efbff44b938fbd5d2.scope: Deactivated successfully.
Mar 27 15:56:21 homeassistant systemd[1]: docker-8e16a9f7fe3e1c278777c449065990a58c564beaae07da3efbff44b938fbd5d2.scope: Consumed 37.703s CPU time.
.....
.....
.....
Mar 27 15:56:30 homeassistant hassio_supervisor[636]: 2025-03-27 16:56:30.105 INFO (MainThread) [supervisor.host.services] Updating service information
Mar 27 15:56:30 homeassistant hassio_supervisor[636]: 2025-03-27 16:56:30.194 INFO (MainThread) [supervisor.host.network] Updating local network information
Mar 27 15:56:30 homeassistant hassio_supervisor[636]: 2025-03-27 16:56:30.294 INFO (MainThread) [supervisor.host.manager] Host information reload completed
Mar 27 15:56:30 homeassistant hassio_supervisor[636]: 2025-03-27 16:56:30.320 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_failed/dns_server
Mar 27 15:56:30 homeassistant hassio_supervisor[636]: 2025-03-27 16:56:30.358 INFO (MainThread) [supervisor.resolution.checks.base] Run check for security/core
Mar 27 15:56:30 homeassistant hassio_supervisor[636]: 2025-03-27 16:56:30.359 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_ipv6_error/dns_server
Mar 27 15:56:30 homeassistant hassio_supervisor[636]: 2025-03-27 16:56:30.380 INFO (MainThread) [supervisor.resolution.checks.base] Run check for disabled_data_disk/system
Mar 27 15:56:30 homeassistant hassio_supervisor[636]: 2025-03-27 16:56:30.380 INFO (MainThread) [supervisor.resolution.checks.base] Run check for no_current_backup/system
Mar 27 15:56:30 homeassistant hassio_supervisor[636]: 2025-03-27 16:56:30.380 INFO (MainThread) [supervisor.resolution.module] Create new suggestion create_full_backup - system / None
Mar 27 15:56:30 homeassistant hassio_supervisor[636]: 2025-03-27 16:56:30.380 INFO (MainThread) [supervisor.resolution.module] Create new issue no_current_backup - system / None
Mar 27 15:56:30 homeassistant hassio_supervisor[636]: 2025-03-27 16:56:30.381 INFO (MainThread) [supervisor.resolution.checks.base] Run check for free_space/system
[... cut ...] ddr=? terminal=? res=succe
Mar 27 15:56:42 homeassistant addon_45df7312_zigbee2mqtt_edge[636]: [2025-03-27 16:56:42] error:         zh:ember:uart:ash: Port Error: read ECONNRESET
Mar 27 15:56:42 homeassistant addon_45df7312_zigbee2mqtt_edge[636]: [2025-03-27 16:56:42] error:         zh:ember: Adapter fatal error: ERROR_SERIAL_INIT
```

For now I leave the Home assistant dev channel so that supervisor is not updating so frequently, but I assume it is linked,  does anybody note the same behavior ?

closed 2025-06-08 by github-actions[bot]