Koenkk/zigbee2mqtt #26009: [Wrong device]: Miboxer FUT035Z+ still recognized as TS0502B

mreilaender, 2025-01-26:
### Link

https://miboxer.com/product/2-in-1-led-controller-zigbee-3-0-2-4g

### Model

FUT035Z+

### Description

LED Controller

### Vendor

Miboxer

### Picture (link)

https://miboxer.com/wp-content/uploads/a1-51-300x300.jpg

### Database entry

{"id":19,"type":"Router","ieeeAddr":"0x8c65a3fffe55bc1a","nwkAddr":21752,"manufId":4098,"manufName":"_TZB210_ue01a0s2","powerSource":"Mains (single phase)","modelId":"TS0502B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,4096,8,768,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000\u0012","65506":31,"65508":0,"65534":0,"modelId":"TS0502B","manufacturerName":"_TZB210_ue01a0s2","stackVersion":0,"dateCode":"","zclVersion":3,"appVersion":71,"powerSource":1,"hwVersion":1}},"genOnOff":{"attributes":{"onOff":0}},"lightingColorCtrl":{"attributes":{"57344":530,"57600":4243493224448,"61453":1,"tuyaRgbMode":0,"colorCapabilities":16,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}},"genLevelCtrl":{"attributes":{"61440":980,"currentLevel":251}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00259b696c","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b00259b696c","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0,"manufacturerCode":null},{"cluster":8,"attrId":0,"minRepIntval":1,"maxRepIntval":3600,"repChange":1,"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":71,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1737886953386}

### Notes

So initially I paired the device with z2m 1.35.0. I then saw the github issues and updated to z2m 2.0 and reinterviewed the device via the GUI. It is recognized as this device: https://www.zigbee2mqtt.io/devices/TS0502B.html#tuya-ts0502b and works fine for brightness and on/off control.

But color temperature setting is not working and produces the following exception in the logs:

```
z2m: Received Zigbee message from 'Kitchen under cabinet lights', type 'raw', cluster 'lightingColorCtrl', data '{"data":[8,64,10,2,241,72,12,1,0,0,0,232,3,220,3,220,3,234,1],"type":"Buffer"}' from endpoint 1 with groupID 0
[2025-01-26 11:42:00] debug: 	z2m: No converter available for 'TS0502B' with cluster 'lightingColorCtrl' and type 'raw' and data '{"data":[8,64,10,2,241,72,12,1,0,0,0,232,3,220,3,220,3,234,1],"type":"Buffer"}'
```

Found those related issues:

- https://github.com/Koenkk/zigbee2mqtt/issues/16768
- https://github.com/Koenkk/zigbee2mqtt/issues/18810

which seem to suggest this was fixed but indeed it hasn't.

jobe451, 2025-03-06:
I do have also a problem with FUT035Z+ devices. Specifically brightness_move does not properly work. I originally opened an issue in a different project. Looks like that is a converters issue. FUT035Z+ is identified as TS0502B in mercator.ts. 

miboxer.ts is missing a definition for FUT035Z+ 

my iobroker adapter uses zigbee-herdsman-converters in version 23.1.1.

My original issue:
https://github.com/ioBroker/ioBroker.zigbee/issues/2406

Kapteenivamma, 2025-04-10:
I have the same problem. https://github.com/Koenkk/zigbee2mqtt/issues/26588¨
The problem for me is also that the light transition is not working. Originally they where identified as FUT035Z+.

FilippoS1973, 2025-06-28:
Producer code: _TZB210_ue01a0s2
I also bought about 10 of them for the new house, but they’re not working properly. 
@Koenkk Any chance I could get your valuable support?
Super thank you for this project

mreilaender, 2025-07-01:
I'm sorry I think I should've been clearer in the issue description: Color temperature setting is not working at all, it's not a cosmetic issue.

I don't have an official hub so I can't verify if color temperature is supported but it says so on the official website (see my link). Also given the logs that I provided I don't think this is a device issue ... @Koenkk

FilippoS1973, 2025-07-01:
> I fixed the detection but this is just a cosmetic change.
> 
> > Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.
> 
> [@FilippoS1973](https://github.com/FilippoS1973) what is not working properly with z2m 2.4.0?

The state synchronization wasn't working when I used the push switch. I deleted the device and paired it again, and now it works. I have no other issues to report.

Leny1996, 2025-07-07:
I have the same device, but using it in single color mode. It still shows temperature control and I couldn't find any option to disable it. If I can help somehow with detecting this mode, let me know.

mreilaender, 2025-07-08:
@Koenkk I already did ... do you need more than this:

```
z2m: Received Zigbee message from 'Kitchen under cabinet lights', type 'raw', cluster 'lightingColorCtrl', data '{"data":[8,64,10,2,241,72,12,1,0,0,0,232,3,220,3,220,3,234,1],"type":"Buffer"}' from endpoint 1 with groupID 0
[2025-01-26 11:42:00] debug: 	z2m: No converter available for 'TS0502B' with cluster 'lightingColorCtrl' and type 'raw' and data '{"data":[8,64,10,2,241,72,12,1,0,0,0,232,3,220,3,220,3,234,1],"type":"Buffer"}'
```

Koenkk, 2025-07-08:
@mreilaender this is the logging after setting the temperature, I need a log which starts before setting it (until 10 seconds after)

mreilaender, 2025-07-16:
There you go

[zigbee2mqtt.log](https://github.com/user-attachments/files/21262887/zigbee2mqtt.log)

Koenkk, 2025-07-17:
Regardless of the log message below, it should still set the color temperature (as this is the response send from the device, it should already have set the color before that):

```
[2025-07-16 19:19:16] debug: 	z2m: No converter available for 'TS0502B' with cluster 'lightingColorCtrl' and type 'raw' and data '{"data":[8,55,10,2,241,72,12,1,0,0,0,232,3,208,2,208,2,182,1],"type":"Buffer"}'
```

Somehow it doesn't work with the standard color temp change command. I'm afraid that the only way to figure this out is by sniffing what the original hub sends to change the color temperature (https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html)