Koenkk/zigbee2mqtt #26011: [New device support]: MOES Tuya ZigBee Smart Curtain Switch

MrLeHavre89, 2025-01-26:
### Link

https://a.aliexpress.com/_EHmGfgU

### Database entry

Not sure where to find this in ha

### Zigbee2MQTT version

2.0.0-2

### Comments

Hey all. 
I need some help getting a moes curtain switch into home assistant via zigbee2mqtt. 
I have a moes ZTS-EUR-C which is supported according to the website. 

https://www.zigbee2mqtt.io/devices/ZTS-EUR-C.html

However when I added it, it said it is unsupported. Hence went looking and know that i need to add support (which i find strange as it is mentioned as supported. 

I am not sure how to continue but it can be that i am too stupid of finding it out. 

I am running zigbee2mqtt as an addon in home assistant. 
I read that i needed to check the external definition in z2m which is:

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_5slehgeo',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;

I see TS0601 zts-eur-c in the list of the moes file though vendor is different. 

https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/moes.

Does someone by any chance know how to fix my problem. 

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_5slehgeo',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};
```

### What does/doesn't work with the external definition?

It is connected but completely unsupported. Not sure how to get it working.. can someone help me withh this?

Koenkk, 2025-01-27:
Could you try to:

- Stop Z2M
- Open `data/database.db` with a text editor and change all occurrences of `_TZE204_5slehgeo` to `_TZE200_nhyj64w2`
- Start Z2M and see if the device now works

closed 2025-01-28 by MrLeHavre89

Koenkk, 2025-01-28:
Added it, you can change it back after the next release.