Koenkk/zigbee2mqtt #26013: [EMBER] Random ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY for all devices routed through HEIMAN HS2WD-E

Ten0, 2025-01-26:
### What happened?

When devices decide to connect through [HEIMAN HS2WD-E](https://www.zigbee2mqtt.io/devices/HS2WD-E.html#heiman-hs2wd-e), which presents itself as a router and AFAIK I can't prevent devices on the network from using, I'm getting `ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY` errors, which suggests that the devices are not communicating properly. (Also since I added it to the network, sometimes packets seem to get lost on devices connected through it, so I assume this is linked.)

I'm not completely sure if that's an ember issue of if it's the device itself that's broken, but I've seen that #22492 that showcases the same error has been handled as an issue here, so I'm opening this to get feedback.

Please let me know if this makes sense, and/or if I can help diagnose.

Unless it looks like I'm mistaken I'll probably just be returning this device in the coming days, since it has other issues (#14424).

### What did you expect to happen?

No error, devices work properly

### How to reproduce it (minimal and precise)

Get a HEIMAN HS2WD-E, an  zigbee dongle, and e.g. a SONOFF SNZB-03P (in my case), get a network where the SONOFF is connected through the HEIMAN, go to the SONOFF device page and click the "refresh 🔄" buttons in front of the properties in the "Exposes" tab of the device. This yields `ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY`  errors at the bottom right of the page and in the logs.

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

7.4.1 [GA]

### Adapter

SMLIGHT SLZB-07

### Setup

HAOS on rockpi 4

gvillo, 2025-02-10:
I am running 2.1.1 and both sirens are offline. I'll troubleshoot and see what the hell is going on

Ten0, 2025-02-11:
I had the issues from a fresh pairing just after buying it. I sent it back.

closed 2025-04-21 by github-actions[bot]