Koenkk/zigbee2mqtt #26031: Network Collapse Endpoint Error

TheDescender, 2025-01-27:
### What happened?

Been having a lot of issues with Z2M for several months. Been using a Sonoff Dongle Plus, an SLZB-06 and a UZG-01. Each failing after between 2 days to 2 weeks. Often on the earlier side of the spectrum. Since updating to Z2M v2.0, and sticking to a small network of only 7 devices, it's been stable for roughly several weeks. Just now I got hit with multiple devices suddenly not working that were working fine yesterday. I've been observing the logs over the past couple months and got rid of what I assumed might've been a spammy device. But now nothing is left in my network other than Tradfri and Phillips Hue products, which to my understanding have some of the best Z2M compatibility.

[image]

I've tried running Z2M on my Raspberry on HAOS, and seperately as a docker container on my server, both suffer network collapse after several days, no matter what controller I use.

I have genuinely no clue where to look any more. This error in the logs is now, first time i've seen it, no clue what it means. Any help is much appreciated.

Snippit of logs below. I can post more expansive debug logs on request. Please help me.

### Zigbee2MQTT version

 2.0.0

### Adapter firmware version

20240519

### Adapter

UZG-01

### Setup

HAOS

### Debug log

```[2025-01-27 18:35:21] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/Dinner Light/set' with data '{"state":"ON"}'
[2025-01-27 18:35:21] debug: 	z2m: Publishing 'set' 'state' to 'Dinner Light'
[2025-01-27 18:35:21] debug: 	zh:controller:endpoint: ZCL command 0x00178801030302a0/11 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false})
[2025-01-27 18:35:21] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0x00178801030302a0:31431/11 (0,0,1)
[2025-01-27 18:35:21] debug: 	zh:zstack:znp: --> SREQ: AF - dataRequest - {"dstaddr":31431,"destendpoint":11,"srcendpoint":1,"clusterid":6,"transid":107,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,116,1]}}
[2025-01-27 18:35:21] debug: 	zh:zstack:unpi:writer: --> frame [254,13,36,1,199,122,11,1,6,0,107,0,30,3,1,116,1,155]
[2025-01-27 18:35:21] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,2,102]
[2025-01-27 18:35:21] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,2,102]
[2025-01-27 18:35:21] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [2] - 102
[2025-01-27 18:35:21] debug: 	zh:zstack:znp: <-- SRSP: AF - dataRequest - {"status":2}
[2025-01-27 18:35:21] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-01-27 18:35:21] debug: 	zh:controller:endpoint: Error: ZCL command 0x00178801030302a0/11 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (--> 'SREQ: AF - dataRequest - {"dstaddr":31431,"destendpoint":11,"srcendpoint":1,"clusterid":6,"transid":107,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,116,1]}}' failed with status '(0x02: INVALID_PARAM)' (expected '(0x00: SUCCESS)'))
    at /app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:258:27
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZStackAdapter.dataRequest (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:1046:9)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:509:35)
    at /app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:443:20
    at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZStackAdapter.sendZclFrameToEndpoint (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:441:16)
    at Request.send (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20)
    at Endpoint.sendRequest (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:324:20)```

TheDescender, 2025-02-01:
Ok I have ordered a CC2531 and the necessary hardware to to turn it into a sniffer. Will report back. Hope I have some engagement at that point, since I notice other people having similar problems.

simowilso, 2025-02-13:
Yep I'm getting the same problem. It was better with a Sonoff Zigbee-P dongle, but I cannot get SLZB-06 working at all. I'm going to have to go back to the Sonoff.

closed 2025-04-23 by github-actions[bot]