Koenkk/zigbee2mqtt #26033: [New device support]: TS0001 / _TZ3000_rxlv8md7

ciux23, 2025-01-27:
### Link

https://www.temu.com/it/smart-diy-switch-module-with-wifi-zigbee--power-monitor-remote-operation-via--smart--indoor-home-decor-electric-wall-switch-box-installation-supports-dual-way--mini-16a-compatible-with--google-home-voice-control-no-battery-or-laser-programmable-requires-assembly-plastic-white-g-601099661224536.html?_oak_mp_inf=ENjUoN6m1ogBGiBkODI2NzU4ODdlNzc0NWFkYjdiMzM5OThlNjVlNmMxNiDI4fDLyjI%3D&top_gallery_url=https%3A%2F%2Fimg.kwcdn.com%2Fproduct%2Ffancy%2Fb3f9484e-e0d6-4209-b830-bfcb43a78372.jpg&spec_gallery_id=601099661224536&refer_page_sn=10009&refer_source=0&freesia_scene=2&_oak_freesia_scene=2&_oak_rec_ext_1=NTAz&_oak_gallery_order=417132599%2C1699091722%2C2061058884%2C428400551%2C1797910367&search_key=smart%20switch%20Zigbee&refer_page_el_sn=200049&_x_sessn_id=4ky4g3y808&refer_page_name=search_result&refer_page_id=10009_1738010277041_mqvou0y2c9

### Database entry

{"id":28,"type":"Router","ieeeAddr":"0xa4c138d2a706b41e","nwkAddr":44556,"manufId":4417,"manufName":"_TZ3000_rxlv8md7","powerSource":"Mains (single phase)","modelId":"TS0001","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"57344":{"attributes":{"53251":"AAAA"}},"genBasic":{"attributes":{"65503":"S�*/\u0012","65506":56,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3000_rxlv8md7","zclVersion":3,"appVersion":74,"modelId":"TS0001","powerSource":1,"hwVersion":1}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"moesStartUpOnOff":0}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"switchType":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff0671e1","endpointID":1},{"cluster":57345,"type":"endpoint","deviceIeeeAddress":"0x00212effff0671e1","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00212effff0671e1","endpointID":1},{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x00212effff0671e1","endpointID":1}],"configuredReportings":[{"cluster":57345,"attrId":53296,"minRepIntval":60,"maxRepIntval":3600,"repChange":0},{"cluster":1794,"attrId":512,"minRepIntval":60,"maxRepIntval":3600,"repChange":0}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1738009727771}

### Zigbee2MQTT version

2.0.0

### Comments

 With and without external definition the device is shown as ts0001: https://www.zigbee2mqtt.io/devices/TS0001.html#_tz3000_rxlv8md7-ts0001

### External definition

```shell
const {onOff, electricityMeter} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS0001'],
    model: 'TS0001',
    vendor: '_TZ3000_rxlv8md7',
    description: 'Automatically generated definition',
    extend: [onOff({"powerOnBehavior":false}), electricityMeter()],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Works:
- state
- linkquality

doesn't works:
- Power
- Voltage
- Current
- Energy.

ciux23, 2025-01-28:
Hi, now I can see Power, Voltage, Current, Energy, but they always show 0.
Plus now I have Switch Type (I don't know if it works) and Power outage memory.
The image is not the right one.

[image]

[image]

mmorfey, 2025-01-29:
I have a similar device with a similar issue. TS0001_power
Tried the option with database substitution – didn’t help.

`const {onOff, electricityMeter} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS0001'],
    model: 'TS0001',
    vendor: '_TZ3000_xkap8wtb',
    description: 'Automatically generated definition',
    extend: [onOff({"powerOnBehavior":false}), electricityMeter()],
    meta: {},
};

module.exports = definition;`

Koenkk, 2025-01-29:
@ciux23 what happens when you attach a load and read the active power via the frontend?

[image]

ciux23, 2025-01-29:
The current load is a little lamp of 4W let me know if I have to try with an higher one, the switch is set with _TZ3000_g92baclx and this is the result:

[image]

Koenkk, 2025-01-30:
I would try with a higher one, the device returns `0` load with this (as you can see in the right bottom

mmorfey, 2025-01-30:
Hi again!
When I selected these three attributes and pressed the button, almost the correct parameters appeared, except for the amperage.
What do I need to make them read constantly?
[image]
[image]

ciux23, 2025-01-31:
Hi Koenkk, I made a test with a 60W classic lamp, the result is the same.

Koenkk, 2025-02-01:
@ciux23 if it also returns `0` when attaching a load, I guess the device broken

ciux23, 2025-02-01:
Perhaps it is broker, but I tested my second one and the result is the same. 
Have I 2 broken device or the don't have this functionality?

[image]

Koenkk, 2025-02-02:
This error is slightly different, `UNSUPPORTED_ATTRIBUTE` means that the device doesn't support it at all (other device returned 0)

ciux23, 2025-02-03:
They are very chip devices, no problem. I'll use them like router and simple switch on/off with a low load.

Is there a way to change the device image? they are shown like https://www.zigbee2mqtt.io/devices/TS0001.html#_tz3000_rxlv8md7-ts0001.

I don't know if it could help, but I made a test with another instance but with zha to see what it exposed. Same result.

[image]

closed 2025-04-12 by github-actions[bot]