Koenkk/zigbee2mqtt #26037: TS0044 and TS004F are not working after update to 2.0.0-2

pkmaxl, 2025-01-28:
### What happened?

I have these scene ( TS0044 and TS004F) buttons and whenever i update to 2.0.0-2 they stop working. When i go back to 1.42.0-2 the are working great.

### What did you expect to happen?

I expected them to work properly even after the update.

### Zigbee2MQTT version

1.42.0-2

### Adapter firmware version

 20230507

### Adapter

SLZB-06P7 / Current core firmware version: v2.7.1

### Setup

HA Zigbee2MQTT addon

manuelkue, 2025-01-28:
How do you react to the button clicks?
Have you read the [breaking changes](https://github.com/Koenkk/zigbee2mqtt/discussions/24198) from the release notes?

pkmaxl, 2025-01-28:
Hello manuelkue,
thank you very much for your answer. I'm unfortunately not a pro, and my skills are limited.
I have read the breaking changes, but I'm having a hard time understanding and implementing them.
My buttons trigger different scripts depending on how they're pressed. As far as I can remember, the buttons rely on "tap" actions.
It's been a long time since I set everything up, and it has been working perfectly until now.
Thanks for your help in advance.
Peter

Murada99, 2025-01-28:
I had the same problem, I did a repairing of my switches and now it's working again.

Murada99, 2025-01-28:
I think this must also be done :
All action sensors are now disabled by default (sensor.*_action entities). It's recommended to use the [MQTT device trigger](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended) instead. In case you really need the action sensors, add the following to your configuration.yaml :
**homeassistant:
  legacy_action_sensor: true**

GigiPompieru, 2025-01-28:
this will get it 'working' from a Z2M perspective, but when HA drops support for the legacy way of triggering, you'll be back here.

pkmaxl, 2025-01-28:
Thanks to everyone who wants to help me. Do I have to add:

`homeassistant:
  legacy_action_sensor: true
`

to /homeassistant/zigbee2mqtt/configuration.yaml or /homeassistant/configuration.yaml?

GigiPompieru, 2025-01-28:
/zigbee2mqtt/configuration.yaml  but again, read up. this is only a temp fix, you should change your automation to the new way of working as i mentioned https://github.com/Koenkk/zigbee2mqtt/issues/26037#issuecomment-2619330493

closed 2025-04-06 by github-actions[bot]