Koenkk/zigbee2mqtt #26038: Z2M fails to start: z2m: Error: target adapter aps link key data table size insufficient (size=3)

Vitek-1977, 2025-01-28:
### What happened?

Z2M fails to start, actually without any particular reason, in the morning just stopped working

### What did you expect to happen?

Start normally

### How to reproduce it (minimal and precise)

Network stopped working, after restart Z2M fails to start

### Zigbee2MQTT version

latest, latest-dev

### Adapter firmware version

20240710

### Adapter

SLZB-06

### Setup

Docker container on Proxmox VM

### Debug log

[2025-01-28 08:31:00] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-28 08:31:00] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-28 08:31:00] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-28 08:31:00] error: 	z2m: Exiting...
[2025-01-28 08:31:00] error: 	z2m: Error: target adapter aps link key data table size insufficient (size=3)
    at AdapterBackup.restoreBackup (/app/node_modules/.pnpm/zigbee-herdsman@3.2.4/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/adapter-backup.ts:366:31)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at ZnpAdapterManager.beginRestore (/app/node_modules/.pnpm/zigbee-herdsman@3.2.4/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:326:9)
    at ZnpAdapterManager.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.4/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:79:21)
    at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.4/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:158:16)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.4/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)

closed 2025-01-29 by Vitek-1977

Bodya4Git, 2025-03-24:
Hi!
Have same issue. Was it fixed?

Vitek-1977, 2025-03-24:
Nope, had to reinstall and re-pair all devices.