Koenkk/zigbee2mqtt #26044: Schneider W599001 smokealarm shows as mains powered

KimCarlsen, 2025-01-28:
### What happened?

Schneider electric smokealarm W599001 shows as main powered on the devices page. even that it is battery power and battery percentages is exposed. 
Everything works as expected.

[image]

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20240710

### Adapter

zStack3x0

### Setup

proxmox lxc

KCarlsen, 2025-03-30:
still an issue

KCarlsen, 2025-05-30:
error still persists

KCarlsen, 2025-07-31:
still an error

KCarlsen, 2025-10-01:
still an problem

KimCarlsen, 2026-02-06:
still an issue