Koenkk/zigbee2mqtt #26046: Unable to start zigbee2mqtt after switching from Raspberry PI3 to Raspberry PI5

Tos26, 2025-01-28:
### What happened?

Hello,

I have a setup since almost one year based on docker-compose with HA, mosquitto and zigbee2mqtt on a raspberry pi 3 and I have no problem with it.

I'm currently switching this system from the raspberry pi 3 to a raspberry pi 5.

All the config files have been transfered from my previous setup and now, I have an issue during zigbee2mqtt starting:

> Using '/app/data' as data directory
Starting Zigbee2MQTT without watchdog.
[2025-01-28 21:06:14] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-28 21:06:14] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #060ae99cff715cc9ef6a0cfa7cae0cdcb17f0c8a)
[2025-01-28 21:06:14] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-28 21:06:14] info: 	zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyUSB0","manufacturer":"ITead","serialNumber":"7e381d2e6dd8ed11ab326f6162c613ac","pnpId":"usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_7e381d2e6dd8ed11ab326f6162c613ac-if00-port0","vendorId":"10c4","productId":"ea60"} => zstack: 4
[2025-01-28 21:06:14] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2025-01-28 21:06:14] info: 	zh:zstack:znp: Serialport opened
[2025-01-28 21:07:19] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-28 21:07:19] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-28 21:07:19] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-28 21:07:19] error: 	z2m: Exiting...
[2025-01-28 21:07:19] error: 	z2m: Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby (Error: AREQ - ZDO - stateChangeInd after 60000ms
.......

It seems to be an issue about panId. I'd like to point out that all my zigbee devices and routers from my previous network are still in place, power up and running while I try to start the PI 5. Is this the cause of the issue ? is it necessary to stop or reset all the device and re appair them after ?

So, I would like to know what is the correct process to:

- keep the existing zigbee network
- Initialize zigbee2mqtt on the new system with the same dongle previously used

For information, Home assistant and mosquitto work well.

Thank you in advance.

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20210708

### Adapter

Sonoff Dongle Plus P

### Setup

Docker on Raspberry PI 5

### Debug log

[log.log](https://github.com/user-attachments/files/18579653/log.log)

chris-1243, 2025-01-29:
You may find some usefull information in this [thread](https://github.com/Koenkk/zigbee2mqtt/issues/23730#issuecomment-2507769568).

Before editing your files, make a backup of it first...

Tos26, 2025-01-30:
Thank you @chris-1243 

I have to study this thread and others about the parameter panID = GENERATE

But I'm very surprised. I thought that a simple migration from PI3 to PI5 should be easy, it was just a matter of copy/paste the config files from the old to the new one.
But it seems not the good way for migration.

Is there a solution to recover the previous network in that case of migration ?

chris-1243, 2025-01-30:
If you use `GENERATE`, you will need to pair your all devices.

You may try to shutdown all your routers (cuting the power). Start Z2M and slowly bring all the routers back

closed 2025-02-15 by Tos26