Koenkk/zigbee2mqtt #26047: SNZB-01P lost the action entity after 2.0 upgrade

grfernandes54, 2025-01-28:
### What happened?

After I upgraded to 2.0 version the button SNZB-01P lost the action entity 

### What did you expect to happen?

Still having the action entity after the upgrade 

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20240716

### Adapter

Slzb-06p7

### Setup

Addon at home assistant with raspberry pi 5

waldmeier, 2025-01-31:
I have the exact same issue an had to restore to 1.42.0-2 to have it back

pauldcomanici, 2025-03-31:
Thanks @waldmeier , the solution worked for me, an example trigger:
```
trigger: device
domain: mqtt
device_id: <<FIND_IT_IN_THE_BROWSER_URL>>
type: action
subtype: single
id: single
```

closed 2025-06-07 by github-actions[bot]