Koenkk/zigbee2mqtt #26051: Network unresponsive - ZCL command failed with status (0x02: INVALID_PARAM)

lightheaded, 2025-01-29:
### What happened?

After updating SLZB from 2.6.8.dev26 to 2.7.1 I can't control any of the devices, can't enter pairing more nor interact with the network in any way. Some messages seem to be coming through, but for all intents and purposes the Zigbee network is down. Everything used to work well befre the upgrade of SLZB.

What I've tried so far:
- Updating HA and HAOS to latest version
- Downgrading SLZB-06 back to 2.6.8.dev26
- Installing latest dev zigbee coordinator firmware (20240710)
- Restarting the addon, HA, SLZB multiple times
- Re-pairing a device – can't "permit join". Not sure if related, but my HA instance seems to become unresponsive after I try to do "Permit join"

My setup:
**Hardware:** 
- SLZB-06 
- Core firmware version: v2.7.1
- ZIgbee firmware revision: 20240710
- Zigbee2mqtt: 2.0.0-2

zigbee2mqtt configuration:

```yaml
data_path: /config/zigbee2mqtt_eth
socat:
  enabled: true
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  server: mqtt://xxxxxxx
  user: hass
  password: XXXXXXX
  base_topic: z2m_eth
serial:
  port: tcp://192.168.1.70:6638
  baudrate: 115200
  adapter: zstack
  disable_led: true
availability: true
advanced:
  transmit_power: 20
  pan_id: GENERATE
  ext_pan_id: GENERATE
  network_key: GENERATE
  homeassistant_legacy_entity_attributes: false
  homeassistant_legacy_triggers: false
  legacy_api: false
  legacy_availability_payload: false
device_options:
  legacy: false
```

### Zigbee2MQTT version

2.0.0-2

### Adapter firmware version

20240710

### Adapter

SLZB-06 

### Setup

Home Assistant Yellow

### Debug log

```
[2025-01-29 09:33:39] info: 	z2m:mqtt: MQTT publish: topic 'z2m_eth/A1 Mini Power', payload '{"child_lock":"UNLOCK","countdown":0,"current":0.69,"energy":8.11,"indicator_mode":"off/on","linkquality":123,"power":79,"power_outage_memory":"restore","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":222}'
[2025-01-29 09:33:39] info: 	z2m:mqtt: MQTT publish: topic 'z2m_eth/Christmas Lights', payload '{"child_lock":"UNLOCK","countdown":0,"current":0,"energy":39.44,"indicator_mode":"off/on","linkquality":76,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":223}'
[2025-01-29 09:33:39] info: 	z2m:mqtt: MQTT publish: topic 'z2m_eth/Techroom Leak Sensor', payload '{"battery":100,"battery_low":false,"device_temperature":30,"linkquality":102,"power_outage_count":7,"tamper":false,"trigger_count":0,"voltage":3045,"water_leak":false}'
[2025-01-29 09:33:47] error: 	z2m: Publish 'set' 'state' to 'Living room light' failed: 'Error: ZCL command 0x60a423fffef8d202/1 manuSpecificTuya.dataRequest({"seq":1,"dpValues":[{"dp":1,"datatype":1,"data":[1]}]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (--> 'SREQ: AF - dataRequest - {"dstaddr":36290,"destendpoint":1,"srcendpoint":1,"clusterid":61184,"transid":15,"options":0,"radius":30,"len":10,"data":{"type":"Buffer","data":[17,9,0,1,0,1,1,0,1,1]}}' failed with status '(0x02: INVALID_PARAM)' (expected '(0x00: SUCCESS)'))'
[2025-01-29 09:59:31] error: 	z2m: Publish 'set' 'state' to 'Kitchen Light' failed: 'Error: ZCL command 0xcc86ecfffef3bcb5/1 manuSpecificTuya.dataRequest({"seq":1,"dpValues":[{"dp":7,"datatype":1,"data":[0]}]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (--> 'SREQ: AF - dataRequest - {"dstaddr":2129,"destendpoint":1,"srcendpoint":1,"clusterid":61184,"transid":16,"options":0,"radius":30,"len":10,"data":{"type":"Buffer","data":[17,10,0,1,0,7,1,0,1,0]}}' failed with status '(0x02: INVALID_PARAM)' (expected '(0x00: SUCCESS)'))'
[2025-01-29 10:03:54] error: 	z2m: Publish 'set' 'state' to 'A1 Mini Power' failed: 'Error: ZCL command 0xa4c138a601cca322/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (--> 'SREQ: AF - dataRequest - {"dstaddr":2479,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":17,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,11,0]}}' failed with status '(0x02: INVALID_PARAM)' (expected '(0x00: SUCCESS)'))'
[2025-01-29 10:13:49] error: 	z2m: Publish 'set' 'state' to 'Nursery Window Powerstrip' failed: 'Error: ZCL command 0x842e14fffefcdc0e/4 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (--> 'SREQ: AF - dataRequest - {"dstaddr":1523,"destendpoint":4,"srcendpoint":1,"clusterid":6,"transid":18,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,12,0]}}' failed with status '(0x02: INVALID_PARAM)' (expected '(0x00: SUCCESS)'))'
[2025-01-29 10:13:49] error: 	z2m: Publish 'set' 'state' to 'Nursery light' failed: 'Error: ZCL command 0x60a423fffea643d7/1 manuSpecificTuya.dataRequest({"seq":1,"dpValues":[{"dp":1,"datatype":1,"data":[0]}]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (--> 'SREQ: AF - dataRequest - {"dstaddr":18129,"destendpoint":1,"srcendpoint":1,"clusterid":61184,"transid":19,"options":0,"radius":30,"len":10,"data":{"type":"Buffer","data":[17,13,0,1,0,1,1,0,1,0]}}' failed with status '(0x02: INVALID_PARAM)' (expected '(0x00: SUCCESS)'))'
[2025-01-29 10:36:18] error: 	z2m: Publish 'set' 'state' to 'Nursery Window Powerstrip' failed: 'Error: ZCL command 0x842e14fffefcdc0e/4 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (--> 'SREQ: AF - dataRequest - {"dstaddr":1523,"destendpoint":4,"srcendpoint":1,"clusterid":6,"transid":20,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,14,0]}}' failed with status '(0x02: INVALID_PARAM)' (expected '(0x00: SUCCESS)'))'
[2025-01-29 10:36:18] error: 	z2m: Publish 'set' 'state' to 'Nursery light' failed: 'Error: ZCL command 0x60a423fffea643d7/1 manuSpecificTuya.dataRequest({"seq":1,"dpValues":[{"dp":1,"datatype":1,"data":[0]}]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (--> 'SREQ: AF - dataRequest - {"dstaddr":18129,"destendpoint":1,"srcendpoint":1,"clusterid":61184,"transid":21,"options":0,"radius":30,"len":10,"data":{"type":"Buffer","data":[17,15,0,1,0,1,1,0,1,0]}}' failed with status '(0x02: INVALID_PARAM)' (expected '(0x00: SUCCESS)'))'
[2025-01-29 10:41:30] error: 	z2m: Publish 'set' 'state' to 'Kitchen Light' failed: 'Error: ZCL command 0xcc86ecfffef3bcb5/1 manuSpecificTuya.dataRequest({"seq":1,"dpValues":[{"dp":7,"datatype":1,"data":[0]}]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (--> 'SREQ: AF - dataRequest - {"dstaddr":2129,"destendpoint":1,"srcendpoint":1,"clusterid":61184,"transid":22,"options":0,"radius":30,"len":10,"data":{"type":"Buffer","data":[17,16,0,1,0,7,1,0,1,0]}}' failed with status '(0x02: INVALID_PARAM)' (expected '(0x00: SUCCESS)'))'
```

fuzzysb, 2025-03-25:
This issue happens to me on a weekly basis and i need to reboot the SLZB to fix, it drives me mad especially when i am away from home and cannot fix easily. i have just found a reboot daily script, lets see if that helps prevent the issue

lightheaded, 2025-03-26:
> i have just found a reboot daily script, lets see if that helps prevent the issue

For me, it doesn't