Koenkk/zigbee2mqtt #26053: Upgrade to 2.0.0

ricain59, 2025-01-29:
### What happened?

Hi,

I've read all things about breaking changes and apply the necessary.
On starting the addon on HA , i'm get errors on starting addon.

My configuration files:
```
external_converters:
  - silabs.js
devices:
  - devices.yaml
groups:
  - groups.yaml
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://x.x.x.x
  user: user
  password: password
serial:
  port: /dev/ttyUSB0
  disable_led: true
  adapter: ember
advanced:
  homeassistant_legacy_entity_attributes: false
  homeassistant_legacy_triggers: false
  legacy_api: false
  legacy_availability_payload: false
  log_level: error
  pan_id: 6754
  channel: 11
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
  availability_blocklist: []
  availability_passlist: []
  last_seen: ISO_8601
  elapsed: true
device_options:
  legacy: false
blocklist:
  - '0x00158d00020ea5af'
  - '0x00158d0002d4771d'
  - '0x00124b00251ee623'
passlist: []
queue: {}
frontend:
  port: 8099
experimental: {}
availability: true
```

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20210708

### Adapter

Sonoff ZBDongle-E 

### Setup

Add-on on HA

### Debug log

[13:49:38] INFO: Preparing to start...
[13:49:38] INFO: Socat not enabled
[13:49:38] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-29 13:49:52] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-29 13:49:52] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-29 13:49:52] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-29 13:49:52] error: 	z2m: Exiting...
[2025-01-29 13:49:52] error: 	z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
    at EmberAdapter.initEzsp (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:670:19)
    at EmberAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1533:24)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)

senna1992, 2025-01-29:
Did you try upgrading your adapter firmware to 8.0.2?

ricain59, 2025-01-29:
No, where i can find this version?
I find this page: https://sonoff.tech/wp-content/uploads/2023/02/SONOFF-Zigbee-3.0-USB-dongle-plus-firmware-flashing.pdf but not this version?

Tha link have paste, it's correct for flash?
I've buy the adapter flashed.

thank you.

senna1992, 2025-01-29:
Please post pictures of your adapter for identification.

Also make sure to read the guidelines on zigbee2mqtt webpage. YouTube got great tutorials on how to update each kind of adapter.

chris-1243, 2025-01-30:
@ricain59 

Which Sonoff Dongle do you have? P: `zstack`
E: `ezsp` (deprecated), `'ember` (firmware 7.4 or higher)

You filled Dongle-E. According to the firmware version you posted, it seems you have the Dongle-P (20210708).

Could you provide the output of `ls -l /dev/serial/by-id`

ricain59, 2025-02-01:
Hi,

Sorry for the late respond and thank you for your help.
My dongle is zbdongle-p

After that, i've changed my configuration to zstack (TY @chris-1243 ) and apparently it's working fine.

Thanks for your time.

closed 2025-02-01 by ricain59