Koenkk/zigbee2mqtt #26054: [New device support]: ORVIBO DSZ12060 Spot Light S10

Wikeolf, 2025-01-29:
### Link

https://tamyeez.odoo.com/shop/dsz12060-orvibo-soprosmart-wall-spot-light-s10-12w-36-10284 or https://www.orvibo.com/cn/product/s_install.html

### Database entry

{"id":38,"type":"Router","ieeeAddr":"0x3120000021042288","nwkAddr":19007,"manufId":4633,"manufName":"TELINK","powerSource":"Mains (single phase)","modelId":"9ff5a780c5a4470d9087175c71d50f92","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":258,"inClusterList":[0,3,4,5,6,8,768,65280],"outClusterList":[25],"clusters":{"65280":{"attributes":{"0":"9273756719110946816"}},"genBasic":{"attributes":{"modelId":"9ff5a780c5a4470d9087175c71d50f92","manufacturerName":"TELINK","powerSource":1,"zclVersion":3,"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"5.7.2"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"5.7.2","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1738163151121}

### Zigbee2MQTT version

2.0.0

### Comments

I own a DSZ12060, which is a 12W light with a 36° illumination angle. This S10 series seems to have many models
I listed them here so that later people can retrieve this post and it is convenient for them to initiate requests to add similar devices
DSZ07061 DSZ07057 DSZ07058 (7w 15/24/36°) DSZ12062 DSZ12059 DSZ12060 (12w 15/24/36°)

[image](https://tamyeez.odoo.com/web/image/product.product/10543/image_1024/%5BDSZ12060%5D%20ORVIBO-SoproSmart%20Wall%20Spot%20light-S10%20%2812W%2036%5E%29?unique=084d435)from internet

As well as the external converters I can use are as follows

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;
const {identify, light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
        zigbeeModel: ['9ff5a780c5a4470d9087175c71d50f92'],
        model: 'DSZ12060',
        vendor: 'ORVIBO',
        description: 'ORVIBO Spot Light S10',
        extend: [identify(),light({effect: false, colorTemp: {range: [166,370]}})]
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

everything is work fine for me

Koenkk, 2025-01-29:
Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/orvibo.ts)?

Wikeolf, 2025-01-30:
> Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/orvibo.ts)?

Sure, I've submitted a pull request [#8699](https://github.com/Koenkk/zigbee-herdsman-converters/pull/8699) for your review. Thanks.

closed 2025-01-30 by Koenkk