Koenkk/zigbee2mqtt #26061: UNREPORTABLE_ATTRIBUTE error for supported device (Tuya Switch with power meter)

snaiperis, 2025-01-30:
### What happened?

Hello,

I've got this `UNREPORTABLE_ATTRIBUTE` error trying to join a new device for the first time (see debug below). The subsequent attempts to rejoin the device does not emit this error, but power monitoring is not available (only ON/OFF toggle).

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20221226

### Adapter

CC2652P USB Dongle Stick

### Setup

Plain on Raspberry 3B

### Debug log

[2025-01-30 01:21:12] info: 	z2m: Zigbee: allowing new devices to join.
[2025-01-30 01:21:12] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254},"status":"ok","transaction":"c2ozy-8"}'
[2025-01-30 01:21:19] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x70b3d52b600fb71f', payload '{"last_seen":"2025-01-30T01:21:19+02:00"}'
[2025-01-30 01:21:19] info: 	zh:controller: Interview for '0x70b3d52b600fb71f' started
[2025-01-30 01:21:19] info: 	z2m: Device '0x70b3d52b600fb71f' joined
[2025-01-30 01:21:19] info: 	z2m: Starting interview of '0x70b3d52b600fb71f'
[2025-01-30 01:21:19] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x70b3d52b600fb71f","ieee_address":"0x70b3d52b600fb71f"},"type":"device_joined"}'
[2025-01-30 01:21:19] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x70b3d52b600fb71f","ieee_address":"0x70b3d52b600fb71f","status":"started"},"type":"device_interview"}'
[2025-01-30 01:21:19] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x70b3d52b600fb71f', payload '{"last_seen":"2025-01-30T01:21:19+02:00"}'
[2025-01-30 01:21:19] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c1383d6ffb7991', payload '{"humidity":53.46,"last_seen":"2025-01-30T01:21:19+02:00","linkquality":36,"temperature":22.9}'
[2025-01-30 01:21:20] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x70b3d52b600fb71f","ieee_address":"0x70b3d52b600fb71f"},"type":"device_announce"}'
[2025-01-30 01:21:20] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c1383d6ffb7991', payload '{"humidity":56.69,"last_seen":"2025-01-30T01:21:20+02:00","linkquality":21,"temperature":22.9}'
[2025-01-30 01:21:20] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x70b3d52b600fb71f', payload '{"last_seen":"2025-01-30T01:21:20+02:00","linkquality":32}'
[2025-01-30 01:21:20] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x70b3d52b600fb71f', payload '{"last_seen":"2025-01-30T01:21:20+02:00","linkquality":32}'
[2025-01-30 01:21:20] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x70b3d52b600fb71f', payload '{"last_seen":"2025-01-30T01:21:20+02:00","linkquality":54}'
[2025-01-30 01:21:20] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x70b3d52b600fb71f', payload '{"last_seen":"2025-01-30T01:21:20+02:00","linkquality":47}'
[2025-01-30 01:21:21] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x70b3d52b600fb71f', payload '{"last_seen":"2025-01-30T01:21:21+02:00","linkquality":51}'
[2025-01-30 01:21:21] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x70b3d52b600fb71f', payload '{"last_seen":"2025-01-30T01:21:21+02:00","linkquality":54}'
[2025-01-30 01:21:21] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x70b3d52b600fb71f', payload '{"last_seen":"2025-01-30T01:21:21+02:00","linkquality":47}'
[2025-01-30 01:21:21] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x70b3d52b600fb71f', payload '{"last_seen":"2025-01-30T01:21:21+02:00","linkquality":51}'
[2025-01-30 01:21:21] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x70b3d52b600fb71f', payload '{"last_seen":"2025-01-30T01:21:21+02:00","linkquality":54}'
[2025-01-30 01:21:21] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x70b3d52b600fb71f', payload '{"last_seen":"2025-01-30T01:21:21+02:00","linkquality":54}'
[2025-01-30 01:21:21] info: 	zh:controller: Succesfully interviewed '0x70b3d52b600fb71f'
[2025-01-30 01:21:21] info: 	z2m: Successfully interviewed '0x70b3d52b600fb71f', device has successfully been paired
[2025-01-30 01:21:21] info: 	z2m: Device '0x70b3d52b600fb71f' is supported, identified as: Tuya Switch with power monitoring (TS0001_power)
[2025-01-30 01:21:21] info: 	z2m: Configuring '0x70b3d52b600fb71f'
[2025-01-30 01:21:21] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Switch with power monitoring","exposes":[{"features":[{"access":7,"description":"On/off state of the switch","label":"State","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"category":"diagnostic","description":"Instantaneous measured power","label":"Power","name":"power","property":"power","type":"numeric","unit":"W"},{"access":1,"category":"diagnostic","description":"Instantaneous measured electrical current","label":"Current","name":"current","property":"current","type":"numeric","unit":"A"},{"access":1,"category":"diagnostic","description":"Measured electrical potential value","label":"Voltage","name":"voltage","property":"voltage","type":"numeric","unit":"V"},{"access":1,"description":"Sum of consumed energy","label":"Energy","name":"energy","property":"energy","type":"numeric","unit":"kWh"},{"access":7,"description":"Type of the switch","label":"Switch type","name":"switch_type","property":"switch_type","type":"enum","values":["toggle","state","momentary"]},{"access":7,"description":"Recover state after power outage","label":"Power outage memory","name":"power_outage_memory","property":"power_outage_memory","type":"enum","values":["on","off","restore"]},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS0001_power","options":[{"access":2,"description":"Calibrates the power value (percentual offset), takes into effect on next report of device.","label":"Power calibration","name":"power_calibration","property":"power_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for power, takes into effect on next report of device. This option can only decrease the precision, not increase it.","label":"Power precision","name":"power_precision","property":"power_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the current value (percentual offset), takes into effect on next report of device.","label":"Current calibration","name":"current_calibration","property":"current_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for current, takes into effect on next report of device. This option can only decrease the precision, not increase it.","label":"Current precision","name":"current_precision","property":"current_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the voltage value (percentual offset), takes into effect on next report of device.","label":"Voltage calibration","name":"voltage_calibration","property":"voltage_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for voltage, takes into effect on next report of device. This option can only decrease the precision, not increase it.","label":"Voltage precision","name":"voltage_precision","property":"voltage_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the energy value (percentual offset), takes into effect on next report of device.","label":"Energy calibration","name":"energy_calibration","property":"energy_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for energy, takes into effect on next report of device. This option can only decrease the precision, not increase it.","label":"Energy precision","name":"energy_precision","property":"energy_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"State actions will also be published as 'action' when true (default false).","label":"State action","name":"state_action","property":"state_action","type":"binary","value_off":false,"value_on":true}],"supports_ota":false,"vendor":"Tuya"},"friendly_name":"0x70b3d52b600fb71f","ieee_address":"0x70b3d52b600fb71f","status":"successful","supported":true},"type":"device_interview"}'
[... cut ...] "}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0029e87769","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0029e87769","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0029e87769","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":100,"stackVersion":1,"hwVersion":2,"dateCode":"20210625","swBuildId":"500+TZSKT31Z3101","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1738192881513}

istenik, 2025-02-03:
I have the same problem with this device

deusrulezz, 2025-03-24:
same probleme here 

TS0001

_TZ3000_xkap8wtb

20210625 - 500+TZSKT31Z3101

Tuya - [TS0001_power]

igorermolenko, 2025-03-26:
Same problem with TS0001
zigbee2mqtt version 2.1.3
Sonoff ZBDongle P  firmware  20240710

closed 2025-06-02 by github-actions[bot]