Koenkk/zigbee2mqtt #26070: [New device support]: Adurosmart Eria 81813-V2

StarkeContrast, 2025-01-30:
### Link

https://www.amazon.com/ERIA-Equivalent-Dimmable-Philips-Assistant/dp/B07ZDHNN5M

### Database entry

{"id":15,"type":"Router","ieeeAddr":"0x00158d008b1f91b7","nwkAddr":17414,"manufId":4151,"manufName":"ERIA","powerSource":"Mains (single phase)","modelId":"AD-BR3RGBW3001","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,4,3,6,8,5,768,4096,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"AD-BR3RGBW3001","manufacturerName":"ERIA","powerSource":1,"zclVersion":8,"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20240718","swBuildId":"02000015"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500,"colorMode":1,"currentX":26148,"currentY":25034,"enhancedCurrentHue":0,"currentSaturation":0,"colorTemperature":357}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20240718","swBuildId":"02000015","zclVersion":8,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1738236504917}

### Zigbee2MQTT version

2.0.0

### Comments

[image]

While I am relatively new to automation, Z2M, embedded programming, and this codebase, I have quite a few years of development experience under my belt. I am getting lost because the file names and directories in the guide do not match what I am seeing in the file system (over SAMBA).  There are some things that I have been able to find, like the Database entry above, but it is unclear how and where an external definition can be placed, as the "external_converters" folder does not exist in my instance.  Were there structural / architectural changes across the major version bump, 1.x => 2.0 that rendered the guide out of date, or is this one of those "not created by default, you create it you need it" sort of scenarios?

I searched the repo and found the existing 81809/81813 v1 converter in ardurosmart.ts, and it seems that another zigbeeModel needs to be added to the array.  But again, I am unsure where to make that change to see it realized.

### External definition

```shell
const {light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['AD-BR3RGBW3001'],
    model: 'AD-BR3RGBW3001',
    vendor: 'ERIA',
    description: 'Automatically generated definition',
    extend: [light({"colorTemp":{"range":[153,500]},"color":{"modes":["xy","hs"],"enhancedHue":true}})],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Unsure of where to put it.  As described above, the guide seems to be for an older 1.x version..?

Koenkk, 2025-02-01:
The generated external converter is automatically used, can you control everything from the exposes tab?

StarkeContrast, 2025-02-01:
I did not test everything when tried a few days ago.  Simple stuff like Color (X/Y) and Color temp seemed to work alright from the Exposes tab.  However, nothing other than State and Power-on behavior were flowing through and controllable in HA.

The State tab contains the following JSON, so it does seem to be reading the data from the bulb:
```
{
    "brightness": 254,
    "color": {
        "h": 36,
        "hue": 36,
        "s": 58,
        "saturation": 58,
        "x": 0.399,
        "y": 0.382
    },
    "color_mode": "xy",
    "color_temp": 280,
    "linkquality": 87,
    "state": "ON",
    "effect": null,
    "power_on_behavior": "previous"
}
```

I will go back and test everything on the Exposes tab and report back shortly.

Koenkk, 2025-02-02:
If you click on the light in HA you will get all the controls (this is how HA works). I added the device!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

Koenkk, 2025-02-02:
Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `81813-V2.png`
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

closed 2025-02-02 by Koenkk