Koenkk/zigbee2mqtt #26080: Ikea JETSTROM 6060 Color temperature after update no longer working

Nobody931, 2025-01-31:
### What happened?

After updating the Jetstrom 6060 lamp to Firmware version 2.4.8 I can no longer set the color temperature, getting the following message:
z2m: Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'x')}.
Everything else works, setting the color temperature was working before the firmware update.

### What did you expect to happen?

Being able to change the color temperature

### How to reproduce it (minimal and precise)

update the jetstrom firmware to the newest version?

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20221226

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P (CC2652P)

### Setup

Add-on in Home Assistant on a Raspberry Pi 4

### Debug log

[2025-01-31 13:42:06] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/DeckenlichtHO/set' with data '{"color_temp":346}'
[2025-01-31 13:42:06] debug: 	z2m: Skipping state because of Home Assistant
[2025-01-31 13:42:06] debug: 	z2m: Publishing 'set' 'color_temp' to 'DeckenlichtHO'
[2025-01-31 13:42:06] debug: 	zhc:light: Requested color_temp 346 is lower than minimum supported 56864, using minimum!
[2025-01-31 13:42:06] debug: 	zh:controller:endpoint: ZCL command 0x7cc6b6fffe66852c/1 lightingColorCtrl.moveToColorTemp({"colortemp":56864,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false})
[2025-01-31 13:42:06] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0x7cc6b6fffe66852c:44402/1 (0,0,1)
[2025-01-31 13:42:06] debug: 	zh:zstack:znp: --> SREQ: AF - dataRequest - {"dstaddr":44402,"destendpoint":1,"srcendpoint":1,"clusterid":768,"transid":4,"options":0,"radius":30,"len":7,"data":{"type":"Buffer","data":[1,5,10,32,222,0,0]}}
[2025-01-31 13:42:06] debug: 	zh:zstack:unpi:writer: --> frame [254,17,36,1,114,173,1,1,0,3,4,0,30,7,1,5,10,32,222,0,0,5]
[2025-01-31 13:42:06] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2025-01-31 13:42:06] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2025-01-31 13:42:06] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2025-01-31 13:42:06] debug: 	zh:zstack:znp: <-- SRSP: AF - dataRequest - {"status":0}
[2025-01-31 13:42:06] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-01-31 13:42:06] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,4,194]
[2025-01-31 13:42:06] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,4,194]
[2025-01-31 13:42:06] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,4] - 194
[2025-01-31 13:42:06] debug: 	zh:zstack:znp: <-- AREQ: AF - dataConfirm - {"status":0,"endpoint":1,"transid":4}
[2025-01-31 13:42:06] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-01-31 13:42:06] debug: 	zh:zstack:unpi:parser: <-- [254,25,68,129,0,0,0,3,114,173,1,1,0,142,0,219,119,213,0,0,5,8,5,11,10,0,114,173,29,60]
[2025-01-31 13:42:06] debug: 	zh:zstack:unpi:parser: --- parseNext [254,25,68,129,0,0,0,3,114,173,1,1,0,142,0,219,119,213,0,0,5,8,5,11,10,0,114,173,29,60]
[2025-01-31 13:42:06] debug: 	zh:zstack:unpi:parser: --> parsed 25 - 2 - 4 - 129 - [0,0,0,3,114,173,1,1,0,142,0,219,119,213,0,0,5,8,5,11,10,0,114,173,29] - 60
[2025-01-31 13:42:06] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":768,"srcaddr":44402,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":142,"securityuse":0,"timestamp":13989851,"transseqnumber":0,"len":5,"data":{"type":"Buffer","data":[8,5,11,10,0]}}
[2025-01-31 13:42:06] debug: 	zh:controller: Received payload: clusterID=768, address=44402, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=142, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":5,"commandIdentifier":11},"payload":{"cmdId":10,"statusCode":0},"command":{"ID":11,"name":"defaultRsp","parameters":[{"name":"cmdId","type":32},{"name":"statusCode","type":32}]}}
[2025-01-31 13:42:06] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-01-31 13:42:06] error: 	z2m: Publish 'set' 'color_temp' to 'DeckenlichtHO' failed: 'TypeError: Cannot read properties of undefined (reading 'x')'
[2025-01-31 13:42:06] debug: 	z2m: TypeError: Cannot read properties of undefined (reading 'x')
    at Function.fromObject (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/color.ts:222:16)
    at Function.fromMireds (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/color.ts:235:24)
    at Object.syncColorState (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/color.ts:779:36)
    at Object.convertSet (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/converters/toZigbee.ts:181:33)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:250:36)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17)

Koenkk, 2025-02-02:
Could you try to reconfigure the device by using the yellow refresh button on the frontend and provide the debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

Nobody931, 2025-02-02:
Thank you very much for your response, I hope I have done this correctly

[debug.log](https://github.com/user-attachments/files/18633978/debug.log)

IkaSGC, 2025-02-03:
Same problem here, Z2M see the update, but is not possible to flash him...

Nobody931, 2025-03-01:
Any news regarding this issue? Or is there some known work-around? Currently both lamps only shine in a warm color... I also tried to rebind them, but still no success, binding a lamp directly to a Styrbar remote control however lets me at least change the color to a cool white, but not back to a warm white...

IkaSGC, 2025-03-01:
Nothing new, unfortunately. It still doesn't work, nobody knows what to do about it.

Nobody931, 2025-03-17:
Update: I realized I could simply downgrade the firmware to test if it would resolve the issue, so I followed the steps outlined at [Zigbee2MQTT OTA Updates](https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#downgrade-1). After successfully downgrading both lamps, they are functioning as before, allowing me to adjust the color temperature again. The Jetstrom lamps are now running firmware version "2.4.5" with a build date of "20230807".

closed 2025-03-17 by Nobody931

DP365, 2025-03-23:
@Nobody931 -- I don't manage to downgrade -- 
`zigbee2mqtt/bridge/request/device/ota_update/check/downgrade` on WIC-E (2.4.8, 0x8c65a3fffef4e6e3) gives me:
`Error 2025-03-23 11:35:20 Update of 'WIC-E' failed (No new image available)`
While on WIC-W (2.4.5) gives me:
`Info 2025-03-23 11:37:05 No update available for 'WIC-W'`
Check all via OTA in the GUI returns:
`Warning 2025-03-23 11:38:43 Firmware on '0x8c65a3fffef4e6e3' (JETSTROM 6060) is newer than latest firmware online.`

How were you able to downgrade? It seems that 2.4.8 was pulled as I don't see it as an update option for WIC-W (2.4.5)

DP365, 2025-03-23:
ok never mind, I updated my Z2M from pre v2 to v2.1.3 and was able to downgrade as described in [Zigbee2MQTT OTA Updates](https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#downgrade-1)

`zhc:ota: [0x8c65a3fffef4e6e3 | JETSTROM 6060] Getting previous image, meta: {"fileName":"10115366-zingo_lds_bulb_jetstrom_ws-0x1105-2.4.5-prod.ota.ota.signed","fileVersion":33816581,"fileSize":249490,"url":"https://raw.githubusercontent.com/Koenkk/zigbee-OTA/master/images1/IKEA/10115366-zingo_lds_bulb_jetstrom_ws-0x1105-2.4.5-prod.ota.ota.signed","imageType":4357,"manufacturerCode":4476,"sha512":"2d903d8b28202b9e4ab32c9cf63b9fd25966edc3a21baea45a62854035c021081a55758ad85958fb8830478177af6e2d14d9fe0e296721b42d7cb44c2db84fb5","otaHeaderString":"GBL zingo_lds_bulb_jetstrom_ws\u0000\u0000","originalUrl":"http://fw.ota.homesmart.ikea.net/global/GW1.0/01.21.031/bin/10115366-zingo_lds_bulb_jetstrom_ws-0x1105-2.4.5-prod.ota.ota.signed","releaseNotes":"https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html"}`