Koenkk/zigbee2mqtt #26081: SMLIGHT SLZB-06P10 error zigbee-herdsman

wassenm, 2025-01-31:
### What happened?

I try to connect my SMLIGHT SLZB-06P10 witch zigbee2mqtt. The task is not starting. I receive a error. I had contact with SMLIGHT and they said that the controller is working ok based on the debug log. 

### What did you expect to happen?

The zigbee2mqqt task starts.

Info log:
[07:30:02] INFO: Preparing to start...
[07:30:02] INFO: Socat not enabled
[07:30:03] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-30 07:30:05] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-30 07:30:05] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-30 07:30:05] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-30 07:30:05] info: 	zh:zstack:znp: Opening TCP socket with 192.168.50.211:6638
[2025-01-30 07:30:05] info: 	zh:zstack:znp: Socket connected
[2025-01-30 07:30:05] info: 	zh:zstack:znp: Socket ready
[2025-01-30 07:30:05] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2025-01-30 07:30:06] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2025-01-30 07:30:40] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-30 07:30:40] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-30 07:30:40] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-30 07:30:40] error: 	z2m: Exiting...
[2025-01-30 07:30:41] error: 	z2m: Error: All table entries need to be the same length
    at Table.build (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/structs/table.ts:171:23)
    at apsTcLinkKeyTable (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/structs/entries/aps-tc-link-key-table.ts:21:85)
    at AdapterNvMemory.readTable (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/adapter-nv-memory.ts:375:27)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at AdapterBackup.getTclkTable (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/adapter-backup.ts:497:20)
    at AdapterBackup.createBackup (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/adapter-backup.ts:119:27)
    at ZStackAdapter.backup (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:900:16)
    at Controller.backup (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:376:28)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:185:13)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)

### How to reproduce it (minimal and precise)

I use the config:
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: '-d -d'
  log: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://core-mosquitto:1883
  user: mqtt_user
  password: 
serial:
# Location of SLZB-06P10
  port: tcp://192.168.50.211:6638
  baudrate: 115200
  adapter: zstack
# Disable green led?
  disable_led: false
# Set output power to max 20
advanced:
  transmit_power: 20
  log_level: debug
frontend:
  enabled: true
  port: 8099
version: 4

### Zigbee2MQTT version

2.0.0-2

### Adapter firmware version

Radio 20240716 core: v2.7.1

### Adapter

SMLIGHT SLZB-06P10 

### Setup

Home assistant on Intel PC

### Debug log

[debuglogging_20250131.txt](https://github.com/user-attachments/files/18618491/debuglogging_20250131.txt)

42erMonty, 2025-02-21:
Hello, I have exactly the same problem

cserem, 2025-03-07:
For me the fix was to reset the Zigbee core and then reset the System core. (Perhaps a power-cycle of the device would have done the same, but I didn't have physical access.)

closed 2025-05-15 by github-actions[bot]