Koenkk/zigbee2mqtt #26088: [New device support]: TS0601 Manufacturer _TZE284_xnbkhhdr

gpaolo969, 2025-01-31:
### Link

https://a.aliexpress.com/_EvidC6d

### Database entry

{"id":39,"type":"EndDevice","ieeeAddr":"0xa4c138f82c8e3288","nwkAddr":42919,"manufId":4417,"manufName":"_TZE284_xnbkhhdr","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0,60672],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65487":14400,"65506":56,"65508":0,"appVersion":77,"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1738340758816}

### Zigbee2MQTT version

2.0.0-2

### Comments

Good evening,
This device is not supported, but the model is in the list of compatibility.

The model is ZWT198.

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE284_xnbkhhdr',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;

The device appears as compatible but when adding it it says not supported

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE284_xnbkhhdr',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

not supported

Westcott1, 2025-02-03:
Seems it is now -

> _TZE284_xnbkhhdr is supported in z2m 2.1.0

gpaolo969, 2025-02-04:
Thanks for your work 😊

Cheers,

g

closed 2025-04-13 by github-actions[bot]